linux-nvme/nvme-cli #2209: nvme set-feature quietly fails to set the power mode

eosolid, 2024-02-06:
In the past I was able to use nvme-cli to set the different power modes allowed by the nvme device. Latest version of nvme-cli seems to be working fine and commands are running successfully, but they don't take effect. Last tested working version was on 2.2.1.
I tried this same procedure on version 2.6 and on master ( as today) and still fails to update the power mode of the nvme device.

Working version 2.2.1
[image]
[image]

Failing to set the correct power mode on 2.6
[image]

Failing to set the correct power mode on master ( tested today as 2/6/24 )
[image]
[image]

keithbusch, 2024-02-07:
Because some of the short parameters got hijacked for other purposes... The '-v' parameter now means "verbose" instead of "value". If you want it to work again, you'll need to replace your '-v' option with '--value' or use capital '-V'.

Yes, I know these types of changes are annoying and harm confidence in consistency. I cautioned against changing args because of the destruction to backward compatibility, and @igaw agreed in a different thread that we won't be doing such changes anymore.

keithbusch, 2024-02-07:
When in doubt, run --help:

```
$ ./.build/nvme set-feature --help
Usage: nvme set-feature <device> [OPTIONS]

Options:
  [  --verbose, -v ]                    --- Increase output verbosity
...
  [  --value=<IONUM>, -V <IONUM> ]      --- new value of feature (required)
```

closed 2024-02-07 by keithbusch