linux-nvme/nvme-cli #2221: nvme wdc cap-diag /dev/nvme0 - Decode

dastasho, 2024-02-22:
Hello everyone, I wanted to check if there is a way to decode the file that is gathered through the command

`nvme wdc cap-diag /dev/nvme0`

It generates a ".bin" file and opening it, gives a part human readable information and part gibberish, which I suspect it is either encrypted or requires some sort of the tool to read it.

igaw, 2024-02-22:
As this a plugin maintained by WDC I can't really say much about this command. Maybe @jeff-lien-wdc knows more?

jeff-lien-wdc, 2024-02-22:
@dastasho The cap-diag bin file contains proprietary debug data.  We do have internal tools that can be used to decode the file but we don't release them to the public.

closed 2024-02-23 by igaw

dastasho, 2024-02-23:
Thank you for the responses!