linux-nvme/nvme-cli #2241: Commands fail on nvme-cli 2.x when nvme_core module isn't loaded, regression from nvme-cli 1.x

calebsander, 2024-03-14:
On `nvme-cli` 1.x, `nvme list`, `nvme disconnect-all`, etc. are no-ops when the `nvme_core` module isn't loaded:
```
[root@init100-17-1 ~]# lsmod | grep nvme
[root@init100-17-1 ~]# nvme version
nvme version 1.16
[root@init100-17-1 ~]# nvme list
Node                  SN                   Model                                    Namespace Usage                      Format           FW Rev
--------------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
[root@init100-17-1 ~]# nvme disconnect-all
```
But on `nvme-cli` 2.x, they fail when trying to open the nonexistent `/sys/class/nvme`. It would be nice for the commands to be no-ops again.
```
[root@init100-17 ~]# lsmod | grep nvme
[root@init100-17 ~]# nvme list
Failed to scan topology: No such file or directory
[root@init100-17 ~]# nvme disconnect-all
Failed to scan topology: No such file or directory
```

igaw, 2024-03-14:
Is still the case with v2.8? I thought it should not complain anymore.

calebsander, 2024-03-14:
Yes, I just tried it with 2.8:
```
[root@init100-17 nvme-cli]# nvme version
nvme version 2.8 (git 2.8-38-g2af3218)
libnvme version 1.8 (git 1.8-22-gd96e92a)
[root@init100-17 nvme-cli]# modprobe -r nvme_core
[root@init100-17 nvme-cli]# nvme disconnect-all
Failed to scan topology: No such file or directory
```

PR #2250 (2024-03-19): nvme: do not issue warning when nvme_core module is not loaded
nvme 1.x did not issue an warning when the nvme_core module was not loaded. Make 2.x behave in the same way.

Fixes: #2241
closed 2024-03-19 by igaw