linux-nvme/nvme-cli #2254: nvme format fails

eosolid, 2024-03-19:
I have a drive where the format command is failing, the odd part here is that is only being observed on a dell platform. Same drive is formatting like a champ on a supermicro server. I installed the nvme-cli using the same steps on both platform and version. 

this was tested on nvme 2.2.x and on nvme 2.8

CMD: nvme format --lbaf 5 --pi 0 --ms 1 --pil 0 --ses=1 /dev/nvme0n1 -t 999999999

nvme 2.2.x
[image]

nvme 2.8
[image]

Does " (0x210a)  " this error code means anything to anyone??

Would it be possible that the dell platform has some sort of hardware drive lock that I should look for?

keithbusch, 2024-03-19:
The "0x210a" is the status code the device responded with for the command. The human readable text for that status code precedes it in the output: "Invalid Format: The LBA Format specified is not supported".

As to why the device responds that way, I can't tell. With the last 3 commands succeeding, and the id-ns output snippet you're showing, it sure looks like the previous ones should have also succeeded (though I can't tell about the "ms=1" case, some devices don't support that). Sounds like a firmware or quirky device issue.

eosolid, 2024-03-19:
@keithbusch so as far as you are aware it... do you know if OEMs implement platform restrictions that can keep me from giving the format I wish to the drive?

The reason of the odd question is because a solution to something similar here:
https://superuser.com/questions/1574424/format-ssd-with-nvme-lba-format-specified-is-not-supported

and like I said, is very odd that is only on my dell platform that the drive fails to format, same OS and kernel on both systems.

keithbusch, 2024-03-19:
The error status is coming from the device, so if the platform really is doing something to block this, then I assume the device cooperating with whatever mechanism the platform uses to achieve it. There are out-of-band management mechanisms for a platform to do low level device communications and configuration, so I guess it's possible even if I've never heard of any using it this way.

eosolid, 2024-03-19:
Yeah I need to look somewhere else... I know for a fact that is not the nvme tool ... is might as well be something in the firmware as you mention...

Thanks for your thoughts

closed 2024-03-20 by igaw

aggieNick02, 2024-06-21:
It seems like there is a decent chance this is related to some of the other issues filed against `nvme-cli` for similar error messages, although I'll admit this one is a bit different (especially surprising/different is that it works just fine on a supermicro board).

Anyways, I recently figured out more details about what was going on in those related issues. In case it may help with this one, I wanted to mention the update I posted [here](https://github.com/linux-nvme/nvme-cli/issues/84#issuecomment-2183511051).

eosolid, 2024-06-21:
@aggieNick02 @keithbusch 

Yes for me this issue was firmware related. I finally was able to complete a fw update on the device and I notice that updating the FW fix that issue.

elukey, 2025-08-27:
Adding a comment since I got into the same issue and I didn't resolve it with a firmware upgrade. I used `nvme id-ns /dev/nvme0n1` to see the LBA formats available, and I noticed that I was trying to use `nvme format -l X` with a X value that wasn't supported (in my case I tried `2` while only `0` and `1` were supported by the hardware).