linux-nvme/nvme-cli #2280: rpmb_program_auth_key does not request the result of key programming

yang8621, 2024-04-01:
Refer to figure 465 of nvme spec: https://nvmexpress.org/wp-content/uploads/NVMe-NVM-Express-2.0a-2021.07.26-Ratified.pdf.

https://github.com/linux-nvme/nvme-cli/blob/master/nvme-rpmb.c#L613 before this line, RPMB_REQ_READ_RESULT does not send.

Please refer to auth_data_write_chunk: https://github.com/linux-nvme/nvme-cli/blob/master/nvme-rpmb.c#L684

igaw, 2024-04-02:
I am not really familiar with this part of the spec. Maybe @hanumanthuh has an idea?

hanumanthuh, 2024-04-04:
The additional RPMB_REQ_READ_RESULT was omitted intentionally, as the explicit request for result was presumed to be programming aid for operations that controller takes some time. Controller should still respond to Key programming result to sec_recv command from host. Got any controller that results in error or takes longer time than expected to respond to sec_recv without sec_send for result ? Please go ahead and add RPMB_REQ_READ_RESULT as done in other data write (to target) operations (like the one you indicated).

closed 2024-07-11 by igaw