linux-nvme/nvme-cli #2328: 2.9: build fails

kloczek, 2024-05-03:
```console
[tkloczko@pers-jacek x86_64-redhat-linux-gnu]$ ninja
[8/212] Compiling C object nvme.p/plugins_spdk_spdk-nvme.c.o
FAILED: nvme.p/plugins_spdk_spdk-nvme.c.o
/usr/bin/gcc -Invme.p -I. -I.. -Iccan -I../ccan -I/usr/include/json-c -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=gnu99 -fomit-frame-pointer -D_GNU_SOURCE -include config.h -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -O2 -g -grecord-gcc-switches -pipe -mtls-dialect=gnu2 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -fdata-sections -ffunction-sections -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -flto=auto -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -Wall -Werror=format-security -Os -MD -MQ nvme.p/plugins_spdk_spdk-nvme.c.o -MF nvme.p/plugins_spdk_spdk-nvme.c.o.d -o nvme.p/plugins_spdk_spdk-nvme.c.o -c ../plugins/spdk/spdk-nvme.c
../plugins/spdk/spdk-nvme.c:22:10: fatal error: nvme/private.h: No such file or directory
   22 | #include "nvme/private.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
[57/212] Generating Documentation/nvme_fdp_update_man with a custom command
ninja: build stopped: subcommand failed.
```

igaw, 2024-05-03:
Alright, I am going to drop the spdk plugin. Clearly someone didn't get the memo what private means.

igaw, 2024-05-03:
Thanks for the report. Happy packaging!

kloczek, 2024-05-03:
Don't worry.
Bug happens .. sometimes.
I'm happy to see another project which maintainer who reacts in matter of minutes 👍 
Remember that "we are living it the World where all software is not-enough-tested and what i more important is how quickly found bugs are fixed"  😋

Thank you for your time 😄

kloczek, 2024-05-03:
Just tested and 2.9.1 is OK 👍

igaw, 2024-05-03:
Thanks for your kind words. Highly appreciated.

I agree, bugs happen but this is one is not really a bug. Including private headers from libraries is an absolute no go. I should have spotted it but I relied on our CI builds to catch things like this. I make sure this doesn't happen again.

closed 2024-05-07 by igaw