linux-nvme/nvme-cli #2359: cannot create new ns after removing old one

nkukard, 2024-06-04:
Hi there, maybe someone can share some light on what I'm doing wrong.

I removed /dev/nvme0n1 which had a 512 block size and wanted to create the namespace with a 4096 block size.

Sadly after removing it, I was unable to create another one.

Here is the output I'm stuck on...
```
root@localhost ~ # nvme create-ns /dev/nvme0
FLBAS corresponding to block size 0 not found
Please correct block size, or specify FLBAS directly
root@localhost ~ # nvme create-ns /dev/nvme0 --block-size 512
NVMe status: Invalid Field in Command: A reserved coded value or an unsupported value in a defined field(0x2002)
root@localhost ~ # nvme create-ns /dev/nvme0 --block-size 4096
NVMe status: Invalid Field in Command: A reserved coded value or an unsupported value in a defined field(0x2002)
```

I'm using Arch..
```
root@localhost ~ # nvme --version
nvme version 2.9.1 (git 2.9.1)
libnvme version 1.9 (git 1.9)
root@localhost ~ # uname -a
Linux localhost.localdomain 6.9.2-arch1-1 #1 SMP PREEMPT_DYNAMIC Sun, 26 May 2024 01:30:29 +0000 x86_64 GNU/Linux
```

Here is the output of id-ctrl....
```
root@localhost ~ # nvme id-ctrl /dev/nvme0
NVME Identify Controller:
vid       : 0x144d
ssvid     : 0x144d
sn        : S3SZNX0M208847
mn        : MZQLW1T9HMJP-00005
fr        : CXV8202Q
rab       : 2
ieee      : 002538
cmic      : 0
mdts      : 6
cntlid    : 0x2
ver       : 0x10200
rtd3r     : 0x7a1200
rtd3e     : 0x7a1200
oaes      : 0
ctratt    : 0
rrls      : 0
cntrltype : 0
fguid     : 00000000-0000-0000-0000-000000000000
crdt1     : 0
crdt2     : 0
crdt3     : 0
nvmsr     : 0
vwci      : 0
mec       : 0
oacs      : 0xf
acl       : 7
aerl      : 3
frmw      : 0x17
lpa       : 0x3
elpe      : 63
npss      : 0
avscc     : 0x1
apsta     : 0
wctemp    : 357
cctemp    : 358
mtfa      : 0
hmpre     : 0
hmmin     : 0
tnvmcap   : 1920383410176
unvmcap   : 0
rpmbs     : 0
edstt     : 0
dsto      : 0
fwug      : 0
kas       : 0
hctma     : 0
mntmt     : 0
mxtmt     : 0
sanicap   : 0
hmminds   : 0
hmmaxd    : 0
nsetidmax : 0
endgidmax : 0
anatt     : 0
anacap    : 0
anagrpmax : 0
nanagrpid : 0
pels      : 0
domainid  : 0
megcap    : 0
sqes      : 0x66
cqes      : 0x44
maxcmd    : 0
nn        : 1
oncs      : 0x1f
fuses     : 0
fna       : 0x4
vwc       : 0
awun      : 0
awupf     : 0
icsvscc   : 1
nwpc      : 0
acwu      : 0
ocfs      : 0
sgls      : 0
mnan      : 0
maxdna    : 0
maxcna    : 0
oaqd      : 0
subnqn    :
ioccsz    : 0
iorcsz    : 0
icdoff    : 0
fcatt     : 0
msdbd     : 0
ofcs      : 0
ps      0 : mp:9.00W operational enlat:5 exlat:5 rrt:0 rrl:0
            rwt:0 rwl:0 idle_power:- active_power:-
            active_power_workload:-
```

I have 5 of these disks and I only tried this on the one, so I have others I can check...

Here is what another disk shows....
```
root@localhost ~ # nvme id-ns /dev/nvme1n1
NVME Identify Namespace 1:
nsze    : 0xdf8fe2b0
ncap    : 0xdf8fe2b0
nuse    : 0xdf8fe2b0
nsfeat  : 0x2
nlbaf   : 1
flbas   : 0
mc      : 0
dpc     : 0
dps     : 0
nmic    : 0
rescap  : 0
fpi     : 0x80
dlfeat  : 0
nawun   : 7
nawupf  : 7
nacwu   : 0
nabsn   : 7
nabo    : 0
nabspf  : 7
noiob   : 0
nvmcap  : 1920383410176
mssrl   : 0
mcl     : 0
msrc    : 0
nulbaf  : 0
anagrpid: 0
nsattr  : 0
nvmsetid: 0
endgid  : 0
nguid   : 33535a304d2035810025385800000001
eui64   : 0000000000000000
lbaf  0 : ms:0   lbads:9  rp:0 (in use)
lbaf  1 : ms:0   lbads:12 rp:0
```

HaroPanosyan, 2024-06-04:
Not sure if this will help, but in case you did delete namespace without first detaching could be the issue. I have noticed similar things in past. Simple controller reset helps in this case.

keithbusch, 2024-06-04:
Don't you need to specify the size and capacity? Giving just the block size isn't enough; you have to say how many blocks you want. The parameters are "`--nsze`" and "`--ncap`".  Newer versions have support for "`-si`" versions for SI units too.

Note, the parameters are aligned to the hardware protocol specification, which are not necessarily intuitive, so check the help or manpage for a description of the units.

nkukard, 2024-06-04:
> Don't you need to specify the size and capacity? Giving just the block size isn't enough; you have to say how many blocks you want. The parameters are "`--nsze`" and "`--ncap`". Newer versions have [...]

Thanks for the info :)

closed 2024-06-04 by nkukard

keithbusch, 2024-06-04:
> Note, the parameters are aligned to the hardware protocol specification, which are not necessarily intuitive, so check the help or manpage for a description of the units.

Ha, that was a bit optimistic: none the documentation is very clear on how these fields work. Anyway, you generally want "ncap" and "nsze" to be the same for most devices unless you're doing thin provisioning. They are in units of the "--block-size" value (so nsze=100 with 4096 block size is not the same as nsze=100 with 512 block size), and the number represents the largest block address (0 is a valid address), and some controllers want the value to be aligned on a particular granularity or require a minimum size or they won't succeed with the command.