linux-nvme/nvme-cli #2367: nvme 2.8 id-ctrl produces invalid JSON output

jvilk-stripe, 2024-06-13:
On Ubuntu Noble:

```bash
$ nvme --version
nvme version 2.8 (git 2.8)
libnvme version 1.8 (git 1.8)
```

Apologies if this is fixed already; I searched the issues tracker and did not see it reported.

`nvme id-ctrl`'s JSON output option does not output valid JSON. The JSON is preceeded by non-JSON output.

Observed behavior:

```bash
$ sudo nvme id-ctrl -o json -v /dev/nvme0n1
opcode       : 06
flags        : 00
rsvd1        : 0000
nsid         : 00000000
cdw2         : 00000000
cdw3         : 00000000
data_len     : 00001000
metadata_len : 00000000
addr         : 63dc0a881000
metadata     : 0
cdw10        : 00000001
cdw11        : 00000000
cdw12        : 00000000
cdw13        : 00000000
cdw14        : 00000000
cdw15        : 00000000
timeout_ms   : 00000000
result       : 00000000
err          : 0
latency      : 58 us
{
  "vid":7439,
  "ssvid":7439,
  "sn":"(redacted)",
  "mn":"Amazon Elastic Block Store              ",
  "fr":"1.0     ",
  "rab":32,
  "ieee":10486492,
  "cmic":0,
  "mdts":6,
  "cntlid":0,
  "ver":65536,
  "rtd3r":0,
  "rtd3e":0,
  "oaes":256,
  "ctratt":0,
  "rrls":0,
  "cntrltype":0,
  "fguid":"00000000-0000-0000-0000-000000000000",
  "crdt1":0,
  "crdt2":0,
  "crdt3":0,
  "nvmsr":0,
  "vwci":0,
  "mec":0,
  "oacs":0,
  "acl":4,
  "aerl":0,
  "frmw":3,
  "lpa":0,
  "elpe":63,
  "npss":0,
  "avscc":1,
  "apsta":0,
  "wctemp":343,
  "cctemp":0,
  "mtfa":0,
  "hmpre":0,
  "hmmin":0,
  "tnvmcap":0,
  "unvmcap":0,
  "rpmbs":0,
  "edstt":0,
  "dsto":0,
  "fwug":0,
  "kas":0,
  "hctma":0,
  "mntmt":0,
  "mxtmt":0,
  "sanicap":0,
  "hmminds":0,
  "hmmaxd":0,
  "nsetidmax":0,
  "endgidmax":0,
  "anatt":0,
  "anacap":0,
  "anagrpmax":0,
  "nanagrpid":0,
  "pels":0,
  "domainid":0,
  "megcap":0,
  "sqes":102,
  "cqes":68,
  "maxcmd":0,
  "nn":1,
  "oncs":0,
  "fuses":0,
  "fna":0,
  "vwc":0,
  "awun":0,
  "awupf":0,
  "icsvscc":0,
  "nwpc":0,
  "acwu":0,
  "ocfs":0,
  "sgls":0,
  "mnan":0,
  "maxdna":0,
  "maxcna":0,
  "oaqd":0,
  "ioccsz":0,
  "iorcsz":0,
  "icdoff":0,
  "fcatt":0,
  "msdbd":0,
  "ofcs":0,
  "psds":[
    {
      "max_power":1,
      "max_power_scale":0,
      "non-operational_state":0,
      "entry_lat":1000000,
      "exit_lat":1000000,
      "read_tput":0,
      "read_lat":0,
      "write_tput":0,
      "write_lat":0,
      "idle_power":0,
      "idle_scale":0,
      "active_power":0,
      "active_power_work":0,
      "active_scale":0
    }
  ]
}
```

Expected behavior: The output does not include the non-JSON output and begins at the first {.

This is not an issue on Focal with nvme version 1.9.

igaw, 2024-06-14:
The non JSON output is due to the `-v/--verbose` flag, which is used for debugging/diagnose purposes. Just leave it away and the rest should validate.

closed 2024-06-14 by igaw

jvilk-stripe, 2024-06-14:
🤦 Sorry for taking your time! You are indeed right. Thank you for the help.

igaw, 2024-06-14:
No worries, all good. I was pondering if we should just disable the verbose option for the JSON output format. Maybe we should extend the help text?