linux-nvme/nvme-cli #2391: "nvme list" command not listing the NVMe device information

ganou2007, 2024-07-03:
I have a system on Oracle Linux 9.X customized system unit ("uname -a" output:Linux AP6400-P1.0B-000387 5.15.0 #15 SMP PREEMPT Wed Jun 26 18:07:02 UTC 2024 x86_64 GNU/Linux), When commanded "nvme list"("nvme version" output: nvme version 2.4 (git 2.4),libnvme version 1.4 (git 1.4)), the output is blank as below:

[image]

But if the same NVMe SSD installed in another Oracle Linux 8.X desktop system (not customized, installed Linux OS from downloaded image in official website), the output was full of content for each item.

What would be missed in the first unit ?
Thanks.

ganou2007, 2024-07-03:
"nvme list -v" command output:

[image]

ganou2007, 2024-07-03:
Command "nvme get-log /dev/nvme0 -l 1024" output as below:

[image]

igaw, 2024-07-03:
> Command "nvme get-log /dev/nvme0 -l 1024" output as below:

This is a low level function for retrieving log pages.  Looks like your device doesn't support  'Supported Log Pages (0x00)`. You might need to select a different log pages, e.g 0x01 (Error Information).

Can you post the kernel messages from your Oracle Linux 9.X system. Does the kernel actually discover any nvme device?

ganou2007, 2024-07-04:
~ # dmesg
[    0.000000] Linux version 5.15.0 (gaou@cgbu-phx-533) (x86_64-redhat-linux-gcc.exe (GCC) 11.4.1 20230605 (Red Hat 11.4.1-2.1.0.1), GNU ld version 2.35.2-42.0.1.el9) #15 SMP PREEMPT Wed Jun 26 18:07:02 UTC 2024
[    0.000000] Command line: console=ttyS0,115200,n8 flags=0x40000011 hostname=AP6400-P1.0B-000387 wancom=10.196.35.11:255.255.240.0:10.196.32.1: loglevel=3   quiet crashkernel=256M  default_hugepagesz=1G hugepagesz=1G hugepages=64 ip6wancom=% mitigations=off initramfs_async=0 intel_iommu=on iommu=pt i2c-i801.disable_features=0x10  
[    0.000000] x86/split lock detection: #AC: crashing the kernel on kernel split_locks and warning on user-space split_locks
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000400-0x0000000000099fff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009a000-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000063eeffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000063ef0000-0x0000000065feffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000065ff0000-0x0000000066134fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x0000000066135000-0x0000000068374fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000068375000-0x000000006cffefff] reserved
[    0.000000] BIOS-e820: [mem 0x000000006cfff000-0x000000006f7fffff] usable
[    0.000000] BIOS-e820: [mem 0x000000006f800000-0x000000008fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed20000-0x00000000fed44fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000407fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0x00100000-0x0010002f] usable ==> usable
[    0.000000] e820: update [mem 0x00100000-0x0010002f] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000400-0x0000000000099fff] usable
[    0.000000] reserve setup_data: [mem 0x000000000009a000-0x000000000009ffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000000e0000-0x00000000000fffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x000000000010002f] usable
[    0.000000] reserve setup_data: [mem 0x0000000000100030-0x0000000063eeffff] usable
[    0.000000] reserve setup_data: [mem 0x0000000063ef0000-0x0000000065feffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000065ff0000-0x0000000066134fff] ACPI data
[    0.000000] reserve setup_data: [mem 0x0000000066135000-0x0000000068374fff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x0000000068375000-0x000000006cffefff] reserved
[    0.000000] reserve setup_data: [mem 0x000000006cfff000-0x000000006f7fffff] usable
[    0.000000] reserve setup_data: [mem 0x000000006f800000-0x000000008fffffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fed20000-0x00000000fed44fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000100000000-0x000000407fffffff] usable
[    0.000000] SMBIOS 3.5.0 present.
[    0.000000] DMI: Oracle Corp, Inc AP6400/AP6400, BIOS 5.28 02/23/2024
[    0.000000] tsc: Detected 2100.000 MHz processor
[    0.000006] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000008] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000014] last_pfn = 0x4080000 max_arch_pfn = 0x400000000
[    0.000166] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.002571] e820: update [mem 0x7f000000-0xffffffff] usable ==> reserved
[    0.002575] last_pfn = 0x6f800 max_arch_pfn = 0x400000000
[    0.014978] found SMP MP-table at [mem 0x000fd020-0x000fd02f]
[    0.014986] Using GB pages for direct mapping
[    0.015269] ACPI: Early table checksum verification disabled
[    0.015271] ACPI: RSDP 0x00000000000F05B0 000024 (v02 ALASKA)
[    0.015275] ACPI: XSDT 0x0000000068370728 0000EC (v01 ALASKA A M I    01072009 AMI  01000013)
[    0.015280] ACPI: FACP 0x0000000066132000 000114 (v06 ALASKA A M I    01072009 INTL 20091013)
[    0.015286] ACPI: DSDT 0x000000006611B000 016E8B (v02 ALASKA A M I    01072009 INTL 20091013)
[    0.015289] ACPI: FACS 0x0000000068373000 000040
[    0.015292] ACPI: FIDT 0x000000006611A000 00009C (v01 ALASKA A M I    01072009 AMI  00010013)
[    0.015295] ACPI: SSDT 0x0000000066134000 000409 (v02 INTEL  xh_nccrb 00000000 INTL 20200430)
[    0.015298] ACPI: SSDT 0x0000000066133000 000748 (v02 INTEL  RAS_ACPI 00000001 INTL 20200430)
[    0.015300] ACPI: ERST 0x0000000066119000 000230 (v01 ALASKA A M I    00000001 INTL 00000001)
[    0.015303] ACPI: BERT 0x0000000066118000 000030 (v01 ALASKA A M I    00000001 INTL 00000001)
[    0.015306] ACPI: SSDT 0x0000000066117000 000745 (v02 INTEL  ADDRXLAT 00000001 INTL 20200430)
[    0.015309] ACPI: MCFG 0x0000000066116000 00003C (v01 ALASKA A M I    01072009 MSFT 00000097)
[    0.015312] ACPI: BDAT 0x0000000066115000 000030 (v01 ALASKA A M I    00000000 INTL 20091013)
[    0.015315] ACPI: HMAT 0x0000000066114000 000D08 (v01 ALASKA A M I    00000001 INTL 20091013)
[    0.015317] ACPI: HPET 0x0000000066113000 000038 (v01 ALASKA A M I    00000001 INTL 20091013)
[    0.015320] ACPI: WDDT 0x0000000066112000 000040 (v01 ALASKA A M I    00000000 INTL 20091013)
[... cut ...]
[  193.778310] device wancom1 entered promiscuous mode
[  193.810311] device wancom2 entered promiscuous mode
[  193.965496] 8021q: adding VLAN 0 to HW filter on device wancom1
[  194.015932] 8021q: adding VLAN 0 to HW filter on device wancom2
[  592.914211] ice 0000:89:00.1: Exiting bypass mode on PHY port 1
[374521.780369] halDataWrite() I2C Ready error after write
[374521.780389] axiRegisterWrite() halDataWrite error: -16  offset: 6
[374521.780391] jup_attr_write error -16 writing to register 6
~ #

ganou2007, 2024-07-04:
~ # nvme error-log /dev/nvme0
Error Log Entries for device:nvme0 entries:64
.................
 Entry[ 0]   
.................
error_count     : 0
sqid            : 0
cmdid           : 0
status_field    : 0(Successful Completion: The command completed without error)
phase_tag       : 0
parm_err_loc    : 0
lba             : 0
nsid            : 0
vs              : 0
trtype          : The transport type is not indicated or the error is not transport related.
csi             : 0
opcode          : 0
cs              : 0
trtype_spec_info: 0
log_page_version: 0
.................
 Entry[ 1]   
.................
error_count     : 0
sqid            : 0
cmdid           : 0
status_field    : 0(Successful Completion: The command completed without error)
phase_tag       : 0
parm_err_loc    : 0
lba             : 0
nsid            : 0
vs              : 0
trtype          : The transport type is not indicated or the error is not transport related.
csi             : 0
opcode          : 0
cs              : 0
trtype_spec_info: 0
log_page_version: 0
.................
 Entry[ 2]   
.................
error_count     : 0
sqid            : 0
cmdid           : 0
status_field    : 0(Successful Completion: The command completed without error)
phase_tag       : 0
parm_err_loc    : 0
lba             : 0
nsid            : 0
vs              : 0
trtype          : The transport type is not indicated or the error is not transport related.
csi             : 0
opcode          : 0
cs              : 0
trtype_spec_info: 0
log_page_version: 0
.................
 Entry[ 3]   
.................
error_count     : 0
sqid            : 0
cmdid           : 0
status_field    : 0(Successful Completion: The command completed without error)
phase_tag       : 0
parm_err_loc    : 0
lba             : 0
nsid            : 0
vs              : 0
trtype          : The transport type is not indicated or the error is not transport related.
csi             : 0
[... 1061 lines cut ...]
csi             : 0
opcode          : 0
cs              : 0
trtype_spec_info: 0
log_page_version: 0
.................
 Entry[63]   
.................
error_count     : 0
sqid            : 0
cmdid           : 0
status_field    : 0(Successful Completion: The command completed without error)
phase_tag       : 0
parm_err_loc    : 0
lba             : 0
nsid            : 0
vs              : 0
trtype          : The transport type is not indicated or the error is not transport related.
csi             : 0
opcode          : 0
cs              : 0
trtype_spec_info: 0
log_page_version: 0
.................
~ #

ganou2007, 2024-07-04:
There is only one SSD in the unit, which is NVMe SSD.

igaw, 2024-07-04:
Okay, the `dmesg` lists the nvme device and since this device contains your rootfs all seems to be good.

That means for some reason nvme-cli is not able to collect any data from sysfs. I suspect it is not able to access it properly. Maybe there is a LSM (e.g. SELinux) active preventing access. Because accessing the block device node works fine; `nvme error-log` opens `/dev/nvme0` and issues a admin command. 

Anyway, let's try to figure out why nvme-cli can't access sysfs. Could you run the command below and upload the `trace.txt` file?

`strace -o /tmp/trace.txt nvme list`

ganou2007, 2024-07-04:
~ # 
~ # strace -o /tmp/trace.txt nvme list
/bin/sh: strace: not found
~ #

ganou2007, 2024-07-04:
Does it need to specify where 'strace' installed? or it maybe did not install.

igaw, 2024-07-04:
You need to install the strace tool, which is usally package name strace.

ganou2007, 2024-07-08:
diag_fld > $cat /tmp/trace.txt
execve("/sbin/nvme", ["nvme", "list"], 0x7fffffffe9c8 /* 37 vars */) = 0
brk(NULL)                               = 0x555555644000
arch_prctl(0x3001 /* ARCH_??? */, 0x7fffffffe8b0) = -1 EINVAL (Invalid argument)
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7ffff7fbf000
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/glibc-hwcaps/x86-64-v4/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/glibc-hwcaps/x86-64-v4", 0x7fffffffdae0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/glibc-hwcaps/x86-64-v3/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/glibc-hwcaps/x86-64-v3", 0x7fffffffdae0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/glibc-hwcaps/x86-64-v2/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/glibc-hwcaps/x86-64-v2", 0x7fffffffdae0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/tls/haswell/avx512_1/x86_64/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/tls/haswell/avx512_1/x86_64", 0x7fffffffdae0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/tls/haswell/avx512_1/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/tls/haswell/avx512_1", 0x7fffffffdae0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/tls/haswell/x86_64/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/tls/haswell/x86_64", 0x7fffffffdae0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/tls/haswell/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/tls/haswell", 0x7fffffffdae0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/tls/avx512_1/x86_64/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/tls/avx512_1/x86_64", 0x7fffffffdae0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/tls/avx512_1/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/tls/avx512_1", 0x7fffffffdae0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/tls/x86_64/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/tls/x86_64", 0x7fffffffdae0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/tls/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/tls", 0x7fffffffdae0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/haswell/avx512_1/x86_64/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/haswell/avx512_1/x86_64", 0x7fffffffdae0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/haswell/avx512_1/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/haswell/avx512_1", 0x7fffffffdae0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/haswell/x86_64/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/haswell/x86_64", 0x7fffffffdae0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/haswell/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/haswell", 0x7fffffffdae0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/avx512_1/x86_64/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/avx512_1/x86_64", 0x7fffffffdae0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/avx512_1/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/avx512_1", 0x7fffffffdae0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/x86_64/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/x86_64", 0x7fffffffdae0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/libnvme.so.1", O_RDONLY|O_CLOEXEC) = 10
read(10, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0p\253\0\0\0\0\0\0"..., 832) = 832
newfstatat(10, "", {st_mode=S_IFREG|0755, st_size=178736, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 176336, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 10, 0) = 0x7ffff7f93000
mprotect(0x7ffff7f9c000, 118784, PROT_NONE) = 0
mmap(0x7ffff7f9c000, 77824, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 10, 0x9000) = 0x7ffff7f9c000
mmap(0x7ffff7faf000, 36864, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 10, 0x1c000) = 0x7ffff7faf000
mmap(0x7ffff7fb9000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 10, 0x25000) = 0x7ffff7fb9000
close(10)                               = 0
openat(AT_FDCWD, "/lib64/libnvme-mi.so.1", O_RDONLY|O_CLOEXEC) = 10
read(10, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\260$\0\0\0\0\0\0"..., 832) = 832
[... cut ...]
close(10)                               = 0
newfstatat(1, "", {st_mode=S_IFCHR|0600, st_rdev=makedev(0x4, 0x40), ...}, AT_EMPTY_PATH) = 0
ioctl(1, TCGETS, {c_iflag=ICRNL|IXON|IXOFF, c_oflag=NL0|CR0|TAB0|BS0|VT0|FF0|OPOST|ONLCR, c_cflag=B115200|CS8|CREAD|HUPCL|CLOCAL, c_lflag=ISIG|ICANON|ECHO|ECHOE|ECHOK|IEXTEN|ECHOCTL|ECHOKE, ...}) = 0
write(1, "Node                  Generic   "..., 170) = 170
write(1, "--------------------- ----------"..., 170) = 170
exit_group(0)                           = ?
+++ exited with 0 +++
diag_fld >

igaw, 2024-07-09:
libnvme is able to read the sysfs and finds the nvme device. Thus after the sysfs scan the device should be in the tree which nvme-cli uses to print. The list command has no obvious way to skip any entry which is the tree structure. 

The only thing I spotted a bit strange is that you don't have any `/etc/nvme/hostnqn` or equivalent configuration and thus libnvme falls back to generate a random one. Not sure if this is contributing to this problem (it shouldn't).

ganou2007, 2024-07-09:
When ran command "nvme list -o normal -v", console output as below:
[image]

ganou2007, 2024-07-09:
Command "strace -o /tmp/trace1.txt nvme list -o normal -v". strace output as below:

execve("/sbin/nvme", ["nvme", "list", "-o", "normal", "-v"], 0x7fffffffe9b0 /* 37 vars */) = 0
brk(NULL)                               = 0x555555644000
arch_prctl(0x3001 /* ARCH_??? */, 0x7fffffffe890) = -1 EINVAL (Invalid argument)
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7ffff7fbf000
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/glibc-hwcaps/x86-64-v4/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/glibc-hwcaps/x86-64-v4", 0x7fffffffdac0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/glibc-hwcaps/x86-64-v3/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/glibc-hwcaps/x86-64-v3", 0x7fffffffdac0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/glibc-hwcaps/x86-64-v2/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/glibc-hwcaps/x86-64-v2", 0x7fffffffdac0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/tls/haswell/avx512_1/x86_64/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/tls/haswell/avx512_1/x86_64", 0x7fffffffdac0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/tls/haswell/avx512_1/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/tls/haswell/avx512_1", 0x7fffffffdac0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/tls/haswell/x86_64/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/tls/haswell/x86_64", 0x7fffffffdac0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/tls/haswell/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/tls/haswell", 0x7fffffffdac0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/tls/avx512_1/x86_64/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/tls/avx512_1/x86_64", 0x7fffffffdac0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/tls/avx512_1/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/tls/avx512_1", 0x7fffffffdac0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/tls/x86_64/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/tls/x86_64", 0x7fffffffdac0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/tls/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/tls", 0x7fffffffdac0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/haswell/avx512_1/x86_64/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/haswell/avx512_1/x86_64", 0x7fffffffdac0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/haswell/avx512_1/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/haswell/avx512_1", 0x7fffffffdac0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/haswell/x86_64/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/haswell/x86_64", 0x7fffffffdac0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/haswell/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/haswell", 0x7fffffffdac0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/avx512_1/x86_64/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/avx512_1/x86_64", 0x7fffffffdac0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/avx512_1/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/avx512_1", 0x7fffffffdac0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/x86_64/libnvme.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib64/x86_64", 0x7fffffffdac0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib64/libnvme.so.1", O_RDONLY|O_CLOEXEC) = 10
read(10, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0p\253\0\0\0\0\0\0"..., 832) = 832
newfstatat(10, "", {st_mode=S_IFREG|0755, st_size=178736, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 176336, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 10, 0) = 0x7ffff7f93000
mprotect(0x7ffff7f9c000, 118784, PROT_NONE) = 0
mmap(0x7ffff7f9c000, 77824, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 10, 0x9000) = 0x7ffff7f9c000
mmap(0x7ffff7faf000, 36864, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 10, 0x1c000) = 0x7ffff7faf000
mmap(0x7ffff7fb9000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 10, 0x25000) = 0x7ffff7fb9000
close(10)                               = 0
openat(AT_FDCWD, "/lib64/libnvme-mi.so.1", O_RDONLY|O_CLOEXEC) = 10
read(10, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\260$\0\0\0\0\0\0"..., 832) = 832
[... cut ...]
write(1, "nvme-subsys0     nqn.2016-08.com"..., 120) = 120
write(1, "\n", 1)                       = 1
write(1, "Device   SN                   MN"..., 132) = 132
write(1, "-------- -------------------- --"..., 132) = 132
write(1, "nvme0    23264213A3BF         Mi"..., 116) = 116
write(1, "\n", 1)                       = 1
write(1, "Device       Generic      NSID  "..., 98) = 98
write(1, "------------ ------------ ------"..., 98) = 98
exit_group(0)                           = ?
+++ exited with 0 +++

igaw, 2024-07-10:
That is very strange. Can you test with the latest version? We have changed quiet a bit in the output code. And given your results, this where things go wrong.

I've  looked at the v2.4 code base but couldn't find any obvious problem, at least not what `-o normal` could explain.

If you don't want to compile, we also have a test image available: https://oxygen.monom.org/linux-nvme/

ganou2007, 2024-07-10:
Would you tell how to use this image:[nvme-cli-latest-x86_64.AppImage](https://monom.org/linux-nvme/upload/AppImage/nvme-cli-latest-x86_64.AppImage)?

Or could you provide a binary file for nvme? Thanks

igaw, 2024-07-10:
Really? First Google hit: https://docs.appimage.org/introduction/quickstart.html#using-the-terminal

ganou2007, 2024-07-10:
diag_fld > $./nvme.AppImage list
dlopen(): error loading libfuse.so.2

AppImages require FUSE to run. 
You might still be able to extract the contents of this AppImage 
if you run it with the --appimage-extract option. 
See https://github.com/AppImage/AppImageKit/wiki/FUSE 
for more information
diag_fld > 

The system is a customized unit. Is there any other way?

igaw, 2024-07-11:
Sure, you can always build your own version or use a static build from me https://oxygen.monom.org/linux-nvme/upload/static/nvme

ganou2007, 2024-07-11:
I used above static build, but the output was the same as before.

[image]

ganou2007, 2024-07-11:
[image]

igaw, 2024-07-11:
In the first run you add the `-v` option. This will run a different code path. Do see any output with just `nvme list -o normal`?

ganou2007, 2024-07-11:
Indeed, I first ran "/sbin/nvme list", only posted later, I thought I should. 

I just tried again, but there was no difference. 

[image]

igaw, 2024-07-11:
Alright, the problem is that the namespaces are not listed. The `nvme list -v` shows the subsystems and the controller but no namespaces. `nvme list` only shows the namespaces, which are also missing. So this is consistent.

What is the content of the sysfs dir `/sys/class/nvme/nvme0`?

``` shell
$ ls /sys/class/nvme/nvme0
address    dctype  firmware_rev  model      nvme0n1                   queue_count        serial  subsysnqn  uevent
cntlid     dev     hwmon1        ng0n1      passthru_err_log_enabled  rescan_controller  sqsize  subsystem
cntrltype  device  kato          numa_node  power                     reset_controller   state   transport
```

ganou2007, 2024-07-11:
[image]

[image]

igaw, 2024-07-12:
As expected, your kernel does not expose the namespaces in sysfs, thus nvme-cli wont show them. This is a kernel bug. I've checked if there were some hickups in older version of Linux but it doesn't look the case we ever had the problem that the namespace was exposed. Your kernel seems not too old v5.15 (2021) so this part should work. 

Anyway, you have to report this against your kernel vendor.

closed 2024-07-12 by igaw

ganou2007, 2024-07-15:
In my unit, the NVMe SSD was partitioned to p1 and p2 as above screenshot(/dev/nvme0n1p1, and /dev/nvme0n1p2). Does it still have /dev/nvme0n1?

igaw, 2024-07-16:
The block device node `/dev/nvmeX` is not accessed for the `nvme list` operation. nvme-cli depends on an correctly populated sysfs trees. The kernel does discover the nvme-pci device correctly (nvme-cli is not involved, all done by the kernel) and exposes the block device node to userspace but fails to populate the `/sys/class/nvme/nvmeX/nvmeXnY` directory. Your kernel doesn't do this for unknown reasons; you need to talk to your OS vendor.

ganou2007, 2024-07-16:
In your gitbub website, (https://github.com/linux-nvme/nvme-cli/tree/v2.9.1), the newest version is 2.9.1. 
The version in my system is 2.4 as showed above. This version was coming with Oracle Linux. In Oracle Linux, the newest one is version 2.6.5 (nvme-cli-2.6-5.el9.x86_64.rpm). 
Would you think the one from your website (version 2.9.1) can fix the issue? Thanks

ganou2007, 2024-07-16:
Referring to namespace issue:
https://github.com/linux-nvme/nvme-cli/issues/1983

igaw, 2024-07-17:
> Would you think the one from your website (version 2.9.1) can fix the issue?

No. You already tested latest greatest with the static nvme-cli build I provided.

Again, the sysfs tree is not complete as far I can tell and this is way nvme-cli is not able to list your namespaces. This is completely a different thing than #1983. Just check if you can see any namespace node under `/sys/class/nvme/nvme0`. If there is no `nvme0n1`,  there wont be any namespace listed by nvme-cli.

ganou2007, 2024-07-20:
just installed a new Oracle Linux 9.3 unit. 

[gou@localhost ~]$ uname -a
Linux localhost.localdomain 5.15.0-208.159.3.el9uek.x86_64 #2 SMP Wed Jun 19 09:05:13 PDT 2024 x86_64 x86_64 x86_64 GNU/Linux

It has folder: /sys/class/nvme/nvme0/nvme0n1/

But it only works in super user mode. 

/home/gou/Pictures/Screenshot from 2024-07-19 20-53-20.png

In Oracle Linux 8.x, it works on user and superuser modes.
Is it possible to make it working in user mode?

igaw, 2024-07-22:
This sounds like a permissions problem: either directory permissions or a LSM (SELinux?). 

The upstream kernel has the following permissions which allow read access to anyone:

```shell
$ ls -ld /sys/class/nvme/nvme0/nvme0n1/
drwxr-xr-x 12 root root 0 Jul 22 08:44 /sys/class/nvme/nvme0/nvme0n1/
```

Anyway, you should report this to your OS vendor.

ganou2007, 2024-07-23:
Here was I saw in the unit.

[image]

igaw, 2024-07-24:
You wrote that running nvme-cli with root permission works, while running nvme-cli with normal user permission it doesn't. There could be other mechanism in place which prevent access, e.g. LSM. All I can say, if nvme-cli is able to access the namespace directly it all works. I haven't observed any problem with upstream kernels nor did I get any other reports so far that there is an upstream problem concerning the namespace directory in sysfs. 

It would also be very surprising if the namespace directory has different default access permission setting than the rest of the sysfs entries.