linux-nvme/nvme-cli #2405: nvme effects-log shows nothing when set to binary

rwk-unil, 2024-07-08:
The output format "binary" shows nothing when chosen :

```
sudo nvme effects-log -o binary /dev/nvme0
```

Neither does :

```
sudo nvme effects-log -b /dev/nvme0
```

The other formats work :

```
sudo nvme effects-log -o normal /dev/nvme0
NVM Command Set Log Page
--------------------------------------------------------------------------------
Admin Commands
ACS0     [Delete I/O Submission Queue     ] 00000001
ACS1     [Create I/O Submission Queue     ] 00000001
ACS2     [Get Log Page                    ] 00000001
ACS4     [Delete I/O Completion Queue     ] 00000001
ACS5     [Create I/O Completion Queue     ] 00000001
ACS6     [Identify                        ] 00000001
ACS8     [Abort                           ] 00000001
ACS9     [Set Features                    ] 00000001
ACS10    [Get Features                    ] 00000001
ACS12    [Asynchronous Event Request      ] 00000001
ACS16    [Firmware Commit                 ] 00000001
ACS17    [Firmware Image Download         ] 00000001
ACS20    [Device Self-test                ] 00000001
ACS128   [Format NVM                      ] 00020003
ACS129   [Security Send                   ] 00020003
ACS130   [Security Receive                ] 00010001

I/O Commands
IOCS0    [Flush                           ] 00010001
IOCS1    [Write                           ] 00000003
IOCS2    [Read                            ] 00000001
IOCS4    [Write Uncorrectable             ] 00000003
IOCS5    [Compare                         ] 00000001
IOCS8    [Write Zeroes                    ] 00000003
IOCS9    [Dataset Management              ] 00010003
```

```
sudo nvme effects-log -o json /dev/nvme0
[
  {
    "command_set_identifier":0,
    "admin_cmd_set":{
      "ACS_0 (Delete I/O Submission Queue)":1,
      "ACS_1 (Create I/O Submission Queue)":1,
      "ACS_2 (Get Log Page)":1,
      "ACS_4 (Delete I/O Completion Queue)":1,
      "ACS_5 (Create I/O Completion Queue)":1,
      "ACS_6 (Identify)":1,
      "ACS_8 (Abort)":1,
      "ACS_9 (Set Features)":1,
      "ACS_10 (Get Features)":1,
      "ACS_12 (Asynchronous Event Request)":1,
      "ACS_16 (Firmware Commit)":1,
      "ACS_17 (Firmware Image Download)":1,
      "ACS_20 (Device Self-test)":1,
      "ACS_128 (Format NVM)":131075,
      "ACS_129 (Security Send)":131075,
      "ACS_130 (Security Receive)":65537
    },
    "io_cmd_set":{
      "IOCS_0 (Flush)":65537,
      "IOCS_1 (Write)":3,
      "IOCS_2 (Read)":1,
      "IOCS_4 (Write Uncorrectable)":3,
      "IOCS_5 (Compare)":1,
      "IOCS_8 (Write Zeroes)":3,
      "IOCS_9 (Dataset Management)":65539
    }
  }
]
```

```
sudo nvme effects-log -H /dev/nvme0
NVM Command Set Log Page
--------------------------------------------------------------------------------
Admin Commands
ACS0     [Delete I/O Submission Queue     ] 00000001  CSUPP+  LBCC-  NCC-  NIC-  CCC-  USS-  No command restriction
ACS1     [Create I/O Submission Queue     ] 00000001  CSUPP+  LBCC-  NCC-  NIC-  CCC-  USS-  No command restriction
ACS2     [Get Log Page                    ] 00000001  CSUPP+  LBCC-  NCC-  NIC-  CCC-  USS-  No command restriction
ACS4     [Delete I/O Completion Queue     ] 00000001  CSUPP+  LBCC-  NCC-  NIC-  CCC-  USS-  No command restriction
ACS5     [Create I/O Completion Queue     ] 00000001  CSUPP+  LBCC-  NCC-  NIC-  CCC-  USS-  No command restriction
ACS6     [Identify                        ] 00000001  CSUPP+  LBCC-  NCC-  NIC-  CCC-  USS-  No command restriction
ACS8     [Abort                           ] 00000001  CSUPP+  LBCC-  NCC-  NIC-  CCC-  USS-  No command restriction
ACS9     [Set Features                    ] 00000001  CSUPP+  LBCC-  NCC-  NIC-  CCC-  USS-  No command restriction
ACS10    [Get Features                    ] 00000001  CSUPP+  LBCC-  NCC-  NIC-  CCC-  USS-  No command restriction
ACS12    [Asynchronous Event Request      ] 00000001  CSUPP+  LBCC-  NCC-  NIC-  CCC-  USS-  No command restriction
ACS16    [Firmware Commit                 ] 00000001  CSUPP+  LBCC-  NCC-  NIC-  CCC-  USS-  No command restriction
ACS17    [Firmware Image Download         ] 00000001  CSUPP+  LBCC-  NCC-  NIC-  CCC-  USS-  No command restriction
ACS20    [Device Self-test                ] 00000001  CSUPP+  LBCC-  NCC-  NIC-  CCC-  USS-  No command restriction
ACS128   [Format NVM                      ] 00020003  CSUPP+  LBCC+  NCC-  NIC-  CCC-  USS-  No other command for any namespace
ACS129   [Security Send                   ] 00020003  CSUPP+  LBCC+  NCC-  NIC-  CCC-  USS-  No other command for any namespace
ACS130   [Security Receive                ] 00010001  CSUPP+  LBCC-  NCC-  NIC-  CCC-  USS-  No other command for same namespace

I/O Commands
IOCS0    [Flush                           ] 00010001  CSUPP+  LBCC-  NCC-  NIC-  CCC-  USS-  No other command for same namespace
IOCS1    [Write                           ] 00000003  CSUPP+  LBCC+  NCC-  NIC-  CCC-  USS-  No command restriction
IOCS2    [Read                            ] 00000001  CSUPP+  LBCC-  NCC-  NIC-  CCC-  USS-  No command restriction
IOCS4    [Write Uncorrectable             ] 00000003  CSUPP+  LBCC+  NCC-  NIC-  CCC-  USS-  No command restriction
IOCS5    [Compare                         ] 00000001  CSUPP+  LBCC-  NCC-  NIC-  CCC-  USS-  No command restriction
IOCS8    [Write Zeroes                    ] 00000003  CSUPP+  LBCC+  NCC-  NIC-  CCC-  USS-  No command restriction
IOCS9    [Dataset Management              ] 00010003  CSUPP+  LBCC+  NCC-  NIC-  CCC-  USS-  No other command for same namespace
```

Tested with version built from source :

```
./nvme --version
nvme version 2.9.1 (git 2.9.1-87-g24a6bfd)
libnvme version 1.9 (git 1.9-35-ge42b6a8)
```

and the older :

```
nvme --version
nvme version 1.16
```

ikegami-t, 2024-07-08:
Seems previously 1.16 supported by the both options raw-binary and -o binary as below.
```c
static int get_effects_log(int argc, char **argv, struct command *cmd, struct plugin *plugin)
{
...
	const char *raw = "show log in binary format";
...
	struct config {
		int   raw_binary;
...
	};
...
	OPT_ARGS(opts) = {
...
		OPT_FLAG("raw-binary",    'b', &cfg.raw_binary,     raw),
...
void nvme_show_effects_log(struct nvme_effects_log_page *effects,
			   unsigned int flags)
{
...
	if (flags & BINARY)
		return d_raw((unsigned char *)effects, sizeof(*effects));
...
```
But unfortunately currently not implemented by the both options as below.
```c
static struct print_ops binary_print_ops = {
	/* libnvme types.h print functions */
...
	.effects_log_list		= NULL,
```

igaw, 2024-07-10:
The fix is merged.

closed 2024-07-10 by igaw