linux-nvme/nvme-cli #2420: nvme show-regs /dev/nvme0 returns: nvme0 failed to map. Did your kernel enable CONFIG_IO_STRICT_DEVMEM?

yalanliufio, 2024-07-15:
In some of our Linux OS system, when show-regs is issued, it returns:  nvme0 failed to map. Did your kernel enable CONFIG_IO_STRICT_DEVMEM?

Looks like this is kernel related problem. When I search around, I found this conversation in nvme cli github
[https://github.com/linux-nvme/nvme-cli/issues/2092](url)

would like to check does this issue is fixed in nvme cli version below? or we need disable the CONFIG_IO_STRICT_DEVMEM option in kernel?
nvme version 2.7.1 (git 2.7.1)
libnvme version 1.7 (git 1.7)

keithbusch, 2024-07-15:
It's more of a "feature" than an issue. The kernel will not allow user space to map a device's IO memory when that config option is set and the tooling can't get around that limitation.

yalanliufio, 2024-07-15:
thanks, in the conversation:

[https://github.com/linux-nvme/nvme-cli/issues/2092](url)

It says:
No more problems using:
nvme version 2.7.1 (git 2.7.1)
libnvme version 1.7 (git 1.7)

Is it correct?

igaw, 2024-07-16:
When the Linux kernel is configured with CONFIG_IO_STRICT_DEVMEM, the kernel prevents user space to map IO memory to userspace. Either compile a kernel without this option enabled or set the kernel command line option `io_memory=relaxed`.

keithbusch, 2024-07-16:
I believe that comment was referring to the "error logs" incrementing when scanning namespaces is not happening in the newer versions.

But if it's referring to show-regs, then I think that means the suggested kernel parameter (which i'd forgotten about), `io_memory=relaxed`, was used.

yalanliufio, 2024-07-16:
Hello keithbush and igaw,
This is the info I found, could help others
After add io_memory=relaxed in grub boot loader, the show-regs command still not work in my Gen5 server.  but I got with a different error:
nvme show-regs /dev/nvme0
get-property: Operation not permitted
After disable "secure boot", then it works. Thanks for your generous help.

igaw, 2024-07-17:
Thanks. IIRC, secure boot also enabled lockdown, which is also preventing mapping PCI bars to user space.

igaw, 2024-07-17:
which means we should extend the message when `show-regs` fails...

tyy-x, 2025-01-28:
> When the Linux kernel is configured with CONFIG_IO_STRICT_DEVMEM, the kernel prevents user space to map IO memory to userspace. Either compile a kernel without this option enabled or set the kernel command line option `io_memory=relaxed`.

Hello igaw, thanks for your solution.

I encountered the same problem but the error message is
`NVMe status: Invalid Command Opcode: A reserved coded value or an unsupported value in the command opcode field(0x1)`

Anyway I tried the solution here but it didn't work. Then I checked the mainline kernel tree and found that the option is actually called `iomem` in the documentation. After I added `iomem=relaxed`, it works. So is the right option name `iomem` or is just that on other systems `io_memory` works as well? If it is the former case, I though it might be useful to point it out.

nvme-cli version is 2.11

Kernel version:
`Linux bookworm 6.1.0-30-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.124-1 (2025-01-12) x86_64 GNU/Linux`

PR #2444 (2024-08-01): nvme: extend help message when mmap regs fails
Newer kernels enable the lockdown feature when secure boot is enabled. This feature also prevents the register mapping into userspace. Thus it is not enough to have CONFIG_IO_STRICT_DEVMEM disabled. Extend the message with the hint also to disable secure boot.

Fixes: #2420