linux-nvme/nvme-cli #2437: Build problem

OGSD, 2024-07-25:
why i use nvme list and i got the following information 

nvme: /lib/x86_64-linux-gnu/libnvme.so.1: version `LIBNVME_1.10' not found (required by nvme)
nvme: /lib/x86_64-linux-gnu/libnvme.so.1: version `LIBNVME_1.9' not found (required by nvme)
nvme: /lib/x86_64-linux-gnu/libnvme-mi.so.1: version `LIBNVME_MI_1_10' not found (required by nvme)

how do i solve it

igaw, 2024-07-25:
The nvme binary was build against an newer version of libnvme. Though when you run nvme it finds an older version of libnvme.  Did you build it yourself and there is already a version on your system installed? How do you build it, what are your commands?

OGSD, 2024-07-26:
> The nvme binary was build against an newer version of libnvme. Though when you run nvme it finds an older version of libnvme. Did you build it yourself and there is already a version on your system installed? How do you build it, what are your commands?

I build the libnvme by myself. I use sudo apt-get install libnvme-dev

igaw, 2024-07-26:
You are mixing distro libraries and self build libraries and the link loader picks up the wrong library when executing the nvme binary. 

There are some instruction in the REAMDE in the libnvme and nvme-cli project on how to build your own versions. Installing the binary in your OS is not covered, this is outside of the scope of this project. But generally not that difficult. I suggest you look at how your distro is building/configuring the build (e.g. for debian https://sources.debian.org/src/libnvme/1.9-1/debian/). 

Anyway, this is not upstream problem.

closed 2024-07-26 by igaw