linux-nvme/nvme-cli #2441: problem with creating the rpm for 2.5

Arima-Sempai, 2024-07-28:
Trying to build nvme 2.5 rpm but running into the following issue:
Processing files: nvme-2.5-1.el8.x86_64
error: File not found: /root/rpmbuild/BUILDROOT/nvme-2.5-1.el8.x86_64/usr/local/share/man/man1/nvme.1

RPM build errors:
    File not found: /root/rpmbuild/BUILDROOT/nvme-2.5-1.el8.x86_64/usr/local/share/man/man1/nvme.1

To replicate:
Install 2.5.tar.gz

Install the general dependency like meson and others
For meson and ninja pip3 was used

After extracting, do the following command command:
$ meson setup --force-fallback-for=libnvme .build

$ meson compile -C .build

meson install -C .build
After it 
$ make rpm

To get this

ModuleNotFoundError: No module named 'mesonbuild'
error: Bad exit status from /var/tmp/rpm-tmp.ks6q6M (%install)

RPM build errors:
    Bad exit status from /var/tmp/rpm-tmp.ks6q6M (%install)
make: *** [Makefile:48: rpm] Error 1

But after editing meson location by doing this:

sudo pip3 uninstall -y meson && \
sudo pip3 uninstall -y ninja && \
sudo pip3 install meson --prefix=/usr && \
sudo pip3 install ninja --prefix=/usr 

I get the error above

note: I am using rocky 8 docker image as an environment
Note: my aim was to download the nvme Cli in /usr/sbin location instead of /usr/local/sbin, (edited the prefix the first time and edited it in the meson.build file in the second time, and didn’t do anything the third time but the results were the same)
Note: please also recommend me a way to change the download location

igaw, 2024-07-29:
You seem to mix up building from distro source packages and building the project from upstream source. 

Please understand, building the distro source package is out of scope for this upstream project. If you want to do this, please contact distro vendor.

Building upstream source is documented in our README. You need to install the dependencies, using the package manger of you distro. For Rocky it is `dnf`. Thus you install meson using `dnf` and not `pip`!

During the configuration step you set the `--prefix` to `/usr`. 

In short

```shell
$ sudo dnf install meson
$ cd nvme-cli
$ meson setup --force-fallback-for=libnvme --prefix=/usr .build 
```

closed 2024-07-29 by igaw

Arima-Sempai, 2024-07-30:
thanks for clearing my confusion, in case of meson for rocky, u need this command to be able to download it: dnf --enablerepo=powertools install meson

but i wanna ask, is there a way to make build an rpm for 2.5 version, where can i find them for example

igaw, 2024-07-30:
There is plenty of documentation available on the internet how to build rpms. Just use a search engine and go through the examples:

https://wiki.centos.org/HowTos(2f)RebuildSRPM.html

or if you feel extra lazy just use ChatGPT "how to build nvme-cli 2.5 from srpm"... It's that simple.

To build `nvme-cli` 2.5 from the source RPM (SRPM), you will need a suitable build environment. Here are the steps to guide you through the process:

1. **Install the Required Tools and Dependencies**:
   Ensure you have the necessary development tools and dependencies installed. On a Fedora or CentOS/RHEL system, you can use the following commands:
   ```bash
   sudo dnf install rpm-build rpmdevtools
   sudo dnf install git gcc make libuuid-devel json-c-devel
   ```

2. **Set Up the RPM Build Environment**:
   Create the RPM build directory structure in your home directory:
   ```bash
   rpmdev-setuptree
   ```

3. **Download the SRPM**:
   Obtain the `nvme-cli` 2.5 SRPM. You can download it from a repository or website that hosts SRPMs, or use `dnf` if it's available in your repositories:
   ```bash
   wget http://example.com/path/to/nvme-cli-2.5-1.src.rpm
   ```

4. **Install the SRPM**:
   Installing the SRPM will place the source code and the spec file into the appropriate directories under `~/rpmbuild/`:
   ```bash
   rpm -ivh nvme-cli-2.5-1.src.rpm
   ```

5. **Build the RPM**:
   Navigate to the SPECS directory and build the RPM:
   ```bash
   cd ~/rpmbuild/SPECS
   rpmbuild -ba nvme-cli.spec
   ```
   This command will compile the source code and create binary RPM packages. If there are missing dependencies, you will need to install them and rerun the command.

6. **Install the Built RPM**:
   Once the build process completes, you will find the generated RPMs in the `~/rpmbuild/RPMS` directory. You can install the binary RPM using:
   ```bash
   sudo rpm -ivh ~/rpmbuild/RPMS/x86_64/nvme-cli-2.5-1.x86_64.rpm
   ```

### Example Step-by-Step Commands:
Here is a summary of all the commands put together:
```bash
# Install necessary tools and dependencies
sudo dnf install rpm-build rpmdevtools git gcc make libuuid-devel json-c-devel

# Set up the RPM build environment
rpmdev-setuptree

# Download the SRPM
wget http://example.com/path/to/nvme-cli-2.5-1.src.rpm

# Install the SRPM
rpm -ivh nvme-cli-2.5-1.src.rpm

# Build the RPM
cd ~/rpmbuild/SPECS
rpmbuild -ba nvme-cli.spec

# Install the built RPM
sudo rpm -ivh ~/rpmbuild/RPMS/x86_64/nvme-cli-2.5-1.x86_64.rpm
```

Replace `http://example.com/path/to/nvme-cli-2.5-1.src.rpm` with the actual URL where the SRPM is hosted.

### Troubleshooting:
- **Missing Dependencies**: If `rpmbuild` reports missing dependencies, install them using `dnf install <missing-dependency>` and rerun the `rpmbuild -ba nvme-cli.spec` command.
- **Build Errors**: Check the build logs located in `~/rpmbuild/BUILD` for detailed error messages and address any issues.

By following these steps, you should be able to successfully build and install `nvme-cli` 2.5 from its SRPM.