linux-nvme/nvme-cli #2448: Documentation install failure with `nvme-cli 2.10`

jpalus, 2024-08-02:
`nvme-cli 2.10` ships new asciidoc document [nvme-ocp-unsupported-reqs-log-pages.txt](https://github.com/linux-nvme/nvme-cli/blob/v2.10/Documentation/nvme-ocp-unsupported-reqs-log-pages.txt), hower filename and document title do not match:

https://github.com/linux-nvme/nvme-cli/blob/7960a126c2cbada3c65aac1f3381539987cf54b1/Documentation/nvme-ocp-unsupported-reqs-log-pages.txt#L1-L2

This causes issues as install expects to find manpage matching file name while `xmlto` outputs manpage matching title resulting in installation failure:
```
...
Installing Documentation/nvme-ocp-telemetry-string-log-page.1 to /home/users/builder/tmp/nvme-cli-2.10-aarch64-root-builder/usr/share/man/man1

ERROR: File 'Documentation/nvme-ocp-unsupported-reqs-log-pages.1' could not be found
FAILED: meson-internal__install 
...
```

igaw, 2024-08-02:
I run into the same issue. The fix is 4c1e89f14474 ("docs: rename ocp-unsupported-req-log file")

I suppose the doc CI build should also do a install step to find these kind of errors.

closed 2024-08-02 by igaw

igaw, 2024-08-02:
The question is, do you need a bug fix release? Or are you okay with the fix in the tree?

jpalus, 2024-08-02:
Personally I can manage without dedicated release although possibly there might more entities affected.

igaw, 2024-08-02:
Alright, I'll wait a bit longer before deciding if we do bugfix release. Maybe there is another bug lurking around.