linux-nvme/nvme-cli #2450: Building nvme-cli 2.10 with man documentation fails

mjekdi, 2024-08-02:
I suspect Documentation/nvme-ocp-unsupported-reqs-log.1 should be Documentation/nvme-ocp-unsupported-reqs-log-pages.1. Renaming the file fixes the build error.

```
nvme-cli-2.10# meson setup --force-fallback-for=libnvme -Ddocs=man .build
The Meson build system
Version: 0.61.2
Source dir: /home/sped/nvme-cli-2.10
Build dir: /home/sped/nvme-cli-2.10/.build
Build type: native build
Project name: nvme-cli
Project version: 2.10
C compiler for the host machine: cc (gcc 11.4.0 "cc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0")
C linker for the host machine: cc ld.bfd 2.38
Host machine cpu family: x86_64
Host machine cpu: x86_64
Looking for a fallback subproject for the dependency libnvme because:
Use of fallback dependencies is forced.
Cloning into 'libnvme'...
remote: Enumerating objects: 19764, done.
remote: Counting objects: 100% (3237/3237), done.
remote: Compressing objects: 100% (1237/1237), done.
remote: Total 19764 (delta 2186), reused 2062 (delta 2000), pack-reused 16527
Receiving objects: 100% (19764/19764), 8.15 MiB | 19.82 MiB/s, done.
Resolving deltas: 100% (15213/15213), done.
HEAD is now at dd51fa85 Release v1.10

Executing subproject libnvme 

libnvme| Project name: libnvme
libnvme| Project version: 1.10
libnvme| C compiler for the host machine: cc (gcc 11.4.0 "cc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0")
libnvme| C linker for the host machine: cc ld.bfd 2.38
libnvme| C++ compiler for the host machine: c++ (gcc 11.4.0 "c++ (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0")
libnvme| C++ linker for the host machine: c++ ld.bfd 2.38
libnvme| Found pkg-config: /usr/bin/pkg-config (0.29.2)
libnvme| Run-time dependency json-c found: YES 0.15
libnvme| Run-time dependency openssl found: YES 3.0.2
libnvme| Header <openssl/opensslv.h> has symbol "LIBRESSL_VERSION_NUMBER" with dependency openssl: NO
libnvme| Has header "openssl/core_names.h" with dependency openssl: YES
libnvme| Did not find CMake 'cmake'
libnvme| Found CMake: NO
libnvme| Run-time dependency libkeyutils found: NO (tried pkgconfig and cmake)
libnvme| Checking if "__builtin_type_compatible_p" compiles: YES
libnvme| Checking if "typeof" compiles: YES
libnvme| Checking if "byteswap.h" compiles: YES
libnvme| Checking if "bswap64" : links: YES
libnvme| Checking if "statement-expr" compiles: YES
libnvme| Checking if "isblank" : links: YES
libnvme| Checking if "linux/mctp.h" compiles: YES
libnvme| Checking if "netdb" : links: YES
libnvme| Checking if "ioctl has glibc-style prototype" compiles: YES
libnvme| Compiler for C supports function attribute fallthrough: YES
libnvme| Configuring libnvme.spec using configuration
libnvme| Configuring config.h using configuration
libnvme| Program python3 found: YES (/usr/bin/python3)
libnvme| WARNING: Python files installed by Meson might not be found by python interpreter.
libnvme| This warning can be avoided by setting "python.platlibdir" option.
libnvme| WARNING: Python files installed by Meson might not be found by python interpreter.
libnvme| This warning can be avoided by setting "python.purelibdir" option.
libnvme| Program swig found: NO
libnvme| Has header "Python.h" with dependency python: YES
libnvme| Configuring nbft-dump-diff.sh using configuration
libnvme| Configuring gen-nbft-diffs.sh using configuration
libnvme| Program diff found: YES (/usr/bin/diff)
libnvme| Program diff found: YES (/usr/bin/diff)
libnvme| Program tree-diff.sh found: YES (/bin/bash -e /home/sped/nvme-cli-2.10/subprojects/libnvme/test/sysfs/tree-diff.sh)
libnvme| Program diff found: YES (/usr/bin/diff)
libnvme| Program config-diff.sh found: YES (/bin/bash -e /home/sped/nvme-cli-2.10/subprojects/libnvme/test/config/config-diff.sh)
libnvme| Configuring conf.py using configuration
libnvme| Configuring api.rst using configuration
libnvme| Configuring index.rst using configuration
libnvme| Configuring quickstart.rst using configuration
libnvme| Configuring installation.rst using configuration
libnvme| Configuring mi.rst using configuration
libnvme| Configuring config-schema.json using configuration
libnvme| Program /home/sped/nvme-cli-2.10/subprojects/libnvme/doc/../scripts/kernel-doc found: YES (/home/sped/nvme-cli-2.10/subprojects/libnvme/doc/../scripts/kernel-doc)
libnvme| Program /home/sped/nvme-cli-2.10/subprojects/libnvme/doc/../scripts/kernel-doc-check found: YES (/home/sped/nvme-cli-2.10/subprojects/libnvme/doc/../scripts/kernel-doc-check)
libnvme| Build targets in project: 30
libnvme| Subproject libnvme finished.

Dependency libnvme from subproject subprojects/libnvme found: YES 1.10
Dependency libnvme-mi from subproject subprojects/libnvme found: YES 1.10
Dependency json-c found: YES 0.15 (cached)
Checking if "__builtin_type_compatible_p" compiles: YES (cached)
Checking if "typeof" compiles: YES (cached)
Checking if "byteswap.h" compiles: YES (cached)
Checking if "bswap64" : links: YES (cached)
Checking if "isblank" : links: YES (cached)
Checking if "sys/random.h" compiles: YES 
Checking if "linux/sed-opal.h" compiles: YES 
Checking if "key_type" compiles: NO 
Compiler for C supports function attribute fallthrough: YES (cached)
Configuring config.h using configuration
Configuring nvme.spec using configuration
Configuring discovery.conf using configuration
Configuring 70-nvmf-autoconnect.conf using configuration
Configuring nvmefc-boot-connections.service using configuration
Configuring nvmf-autoconnect.service using configuration
Configuring nvmf-connect-nbft.service using configuration
Configuring nvmf-connect.target using configuration
Configuring nvmf-connect@.service using configuration
Configuring 65-persistent-net-nbft.rules using configuration
Configuring 70-nvmf-autoconnect.rules using configuration
Configuring 71-nvmf-netapp.rules using configuration
Program asciidoc found: NO

Documentation/meson.build:296:8: ERROR: File nvme-ocp-unsupported-reqs-log-pages.1 does not exist.
```

igaw, 2024-08-05:
Thanks for your report. Already fixed: #2448

closed 2024-08-05 by igaw

mjekdi, 2024-08-05:
Ooops, I should have checked. Thank you!

igaw, 2024-08-05:
No worries. Duplicates are easy to deal with :)