linux-nvme/nvme-cli #2456: nvme smart-log does not show temps in Celsius

Strykar, 2024-08-09:
Temps in Fahrenheit and Kelvin but not Celsius?
```
$ sudo nvme smart-log /dev/nvme0
Smart Log for NVME device:nvme0 namespace-id:ffffffff
critical_warning			: 0
temperature				: 91 °F (306 K)
available_spare				: 100%
available_spare_threshold		: 5%
percentage_used				: 1%
endurance group critical warning summary: 0
Data Units Read				: 42,210,392 (21.61 TB)
Data Units Written			: 20,246,705 (10.37 TB)
host_read_commands			: 2,247,662,721
host_write_commands			: 1,572,136,596
controller_busy_time			: 150
power_cycles				: 111
power_on_hours				: 5,221
unsafe_shutdowns			: 78
media_errors				: 0
num_err_log_entries			: 1,124
Warning Temperature Time		: 0
Critical Composite Temperature Time	: 0
Thermal Management T1 Trans Count	: 0
Thermal Management T2 Trans Count	: 0
Thermal Management T1 Total Time	: 0
Thermal Management T2 Total Time	: 0

$ sudo nvme version 
nvme version 2.10.2 (git 2.10.2)
libnvme version 1.10 (git 1.10)

$ uname -a
Linux r912 6.10.3-arch1-2 #1 SMP PREEMPT_DYNAMIC Tue, 06 Aug 2024 07:21:19 +0000 x86_64 GNU/Linux
```

igaw, 2024-08-09:
The unit type is chosen depending of your locale settings, so if you use `US` locales you get Fahrenheit.

Strykar, 2024-08-09:
> The unit type is chosen depending of your locale settings, so if you use `US` locales you get Fahrenheit.

Thanks for clearing that up, setting `LC_MEASUREMENT=metric` in `/etc/locale.conf` works along side `LANG=en_US.UTF-8`,

igaw, 2024-08-09:
Oh, we might also need to extend our check, though. I don't think we are actually looking at `LC_MEASURMENT` instead just on the country. This is wrong I think.

closed 2024-08-13 by igaw

wdoekes, 2025-09-18:
@Strykar : I'm not so sure `LC_MEASUREMENT` takes a `metric` argument. But it does take `en_GB.UTF-8` for instance.

And the first hit on google for `LC_MEASUREMENT=metric` is the [Arch Wiki](https://wiki.archlinux.org/title/Locale#toc-Metric_measurements_with_US_locale) which references this post!

The [commit that was merged](https://github.com/linux-nvme/nvme-cli/pull/2458/commits/decd39f50b6615659c98972d1e1014f90d5380db) now narrows the affected locale scope from `LC_ALL` to `LC_MEASUREMENT`. Interestingly, it doesn't use `setlocale()` to *set* the category. Instead it uses it to *get* the current country. Which is non-standard. From there it uses custom `is_fahrenheit_country()` to check against e.g. the `US` in `en_US`. (Which is unreached code when you supply the value `metric`.) But see below, for proper usage of `LC_MEASUREMENT`.

If I look inside `/usr/share/i18n/locales/i18n`, I find:
```
LC_MEASUREMENT
% This is the ISO/IEC 14652 "i18n" definition for
% the LC_MEASUREMENT category.  It indicates metric.
measurement 1
END LC_MEASUREMENT
```
versus
```
LC_MEASUREMENT
% US customary units.
measurement 2
END LC_MEASUREMENT
```
which is in `/usr/share/i18n/locales/en_US`.

If nothing matches we will get `C` or `ANSI_X3.4-1968` (default) locale, which happens to have `measurement 1`. So `LC_MEASUREMENT=whatever` will work to get metric, as will `metric`. But it's still wrong:
```
# LC_MEASUREMENT=metric locale -ck LC_MEASUREMENT
locale: Cannot set LC_ALL to default locale: No such file or directory
LC_MEASUREMENT
measurement=1
measurement-codeset="ANSI_X3.4-1968"
```
```
# LC_MEASUREMENT=en_US.UTF-8 locale -ck LC_MEASUREMENT
LC_MEASUREMENT
measurement=2
measurement-codeset="UTF-8"
```

### Proper usage of LC_MEASUREMENT

In a modern system, one would use this:
```c
#include <langinfo.h>
#include <locale.h>
#include <stdio.h>

int main(void) {
    // In nvme.c this is already called. Nothing more is needed.
    setlocale(LC_ALL, "");

    // At the appropriate place, you check for the correct measurement.
    const char *m = nl_langinfo(_NL_MEASUREMENT_MEASUREMENT);
    if (m && m[0] == 2) {
        printf("fahrenheit\n");
    } else {
        printf("celsius\n");
    }
    return 0;
}
```
Example where you see the precedence of LC_MEASUREMENT over LANG.
```
$ env -i ./a.out 
celsius

$ env -i LANG=en_US.UTF-8 ./a.out 
fahrenheit

$ env -i LANG=en_US.UTF-8 LC_MEASUREMENT=en_GB.UTF-8 ./a.out 
celsius
```
And how LC_ALL overrides all:
```
$ env -i LC_ALL=en_US.UTF-8 LC_MEASUREMENT=en_GB.UTF-8 ./a.out 
fahrenheit
```

### TL;DR (1)

- ✅ `LC_MEASUREMENT=en_GB.UTF-8`
- 🚫 `LC_MEASUREMENT=metric`

### TL;DR (2)

The code in question should prefer `_NL_MEASUREMENT_MEASUREMENT` over custom `is_fahrenheit_country`.

martin-belanger, 2025-09-18:
Just my 2¢... :smirk:

We could also print all 3 units (°C, °F, K) and forget about the locale altogether.

```
temperature				: 33 °C, 91 °F, 306 K
```

igaw, 2025-09-18:
As long we only print three different units I am fine with this solution. I swear my next open source project will only accept SI units.

ell1e, 2025-10-01:
I tried `LC_MEASUREMENT=metric nvme smart-log /dev/nvme0` but get Fahrenheit still, will this be fixed in an upcoming version? Is there any good way to do this with the older NVMe 2.15?

wdoekes, 2025-10-02:
Don't do `metric` in `LC_MEASUREMENT` 😩 -- see https://github.com/linux-nvme/nvme-cli/issues/2456#issuecomment-3307263950

And before https://github.com/linux-nvme/nvme-cli/pull/2458/commits/decd39f50b6615659c98972d1e1014f90d5380db you'll need `LC_ALL=en_GB.UTF-8` or `LC_ALL=C`

Strykar, 2025-10-02:
> Huh, but isn't metric nowadays also a common measurement in the US, and the UK also uses non-metric for some purposes? `LC_MEASUREMENT=metric` seems to make more sense than `LC_MEASUREMENT=en_GB.UTF-8`.

That was my initial thought too.

wdoekes, 2025-10-02:
I'm sorry. I tried to explain why `LC_MEASUREMENT=metric` is invalid in my verbose comment above. I don't think I have more words to elucidate. Maybe reread it?

Or ask Mr. Chat?
https://chatgpt.com/share/68de60e1-6f7c-800d-8c0b-ec6b16f4ac86

ell1e, 2025-10-02:
Based on [searching the POSIX locale standard](https://pubs.opengroup.org/onlinepubs/9799919799/) (click "Word search..."), LC_MEASUREMENT doesn't exist. Therefore, it seems to me like a program would be free to interpret it differently. But my apologies if it's actually standardized somewhere I'm not aware of.

ell1e, 2025-10-03:
Based on the many people here being confused, perhaps that alone seems like enough of a reason to consider adding the switch back, maybe in addition to the locale setting? Sorry if I'm missing something here.

> glibc (2.2) has supported it since 2000. 

I'm pretty sure that musl libc doesn't, for example: https://bugs.gentoo.org/942668

wdoekes, 2025-10-03:
> If you also support the country codes, it might work out well.

No, by doing that you're _adding expectations_ for other (users of) programs to support your new _non-standard_ invocation. That's very unfriendly to the ecosystem. You'd be making the world more confusing.

ell1e, 2025-10-03:
Thanks for elaborating. I still don't share that angle on LC_MEASUREMENT, but I guess perhaps it doesn't matter.

What probably matters is there isn't a single program I could find on my entire system where I had to manipulate locale options to see Celsius. I checked lm-sensors's `sensors`, smartmontool's `smartctl`, and fastfetch's `cpu` module temperature.