linux-nvme/nvme-cli #2465: How to Re-enable NVMe Write Cache via `nvme-cli`?

JigaoLuo, 2024-08-16:
I recently discovered it is possible to enable and disable the NVMe Volatile Write Cache (VWC). I successfully disabled it with the following commands:

```sh
# nvme get-feature /dev/nvme0n1 -f 6
get-feature:0x06 (Volatile Write Cache), Current value:0x00000001

# nvme set-feature /dev/nvme0 -f 6 -v 0x01
set-feature:0x06 (Volatile Write Cache), value:00000000, cdw12:00000000, save:0

# nvme get-feature /dev/nvme0 -f 6
get-feature:0x06 (Volatile Write Cache), Current value:00000000
```

The VWC was disabled as expected. However, when I tried to re-enable it, the command did not work, that the value still **00000000**:

```sh
# nvme set-feature /dev/nvme0 -f 0x06 -v 0x01
opcode       : 09
flags        : 00
rsvd1        : 0000
nsid         : ffffffff
cdw2         : 00000000
cdw3         : 00000000
data_len     : 00000000
metadata_len : 00000000
addr         : 0
metadata     : 0
cdw10        : 00000006
cdw11        : 00000000
cdw12        : 00000000
cdw13        : 00000000
cdw14        : 00000000
cdw15        : 00000000
timeout_ms   : 00000000
result       : 00000000
err          : 0
latency      : 8113 us
set-feature:0x06 (Volatile Write Cache), value:00000000, cdw12:00000000, save:0

# nvme get-feature /dev/nvme0n1 -f 0x06
get-feature:0x06 (Volatile Write Cache), Current value:00000000
```
The VWC remains disabled. **So, my question: How can I re-enable the VWC without rebooting?**

---

Additional Information:

 - NVMe disk: Samsung SSD 980 PRO 1TB
 - Linux version: Linux c01 6.2.0-37-generic #38~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Nov 2 18:01:13 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

NVMe controller details:
```sh
# nvme id-ctrl /dev/nvme0n1 -H
...
    vwc       : 0x7
      [2:1] : 0x3 The Flush command supports NSID set to FFFFFFFFh
      [0:0] : 0x1 Volatile Write Cache Present
```

Any insights or suggestions would be greatly appreciated!

ikegami-t, 2024-08-18:
Sorry please change the value option as below.
```diff
-  nvme set-feature /dev/nvme0 -f 0x06 -v 0x01
+ nvme set-feature /dev/nvme0 -f 0x06 -V 0x01
or 
+ nvme set-feature /dev/nvme0 -f 0x06 --value=0x01
```

keithbusch, 2024-08-18:
Or use the long options.

JigaoLuo, 2024-08-18:
> ```diff
> nvme set-feature /dev/nvme0 -f 0x06 --value=0x01
> ```

Thanks.
The `--value=0x01` works!
(At my nvmecli version the `-V` doesnot exist)

closed 2024-08-18 by JigaoLuo

keithbusch, 2024-08-18:
Yes, so, with the best intentions, unifying short options across sub-commands is a good idea, but that kind of change on any command line based utility code risks collisions. This is a hindsight thing: it makes sense to do for consistency, but it is an unfortunate API breakage.