linux-nvme/nvme-cli #2466: nvme-cli  "nvme list" command shows wrong usage information

jmvaswani, 2024-08-16:
Hi, currently running nvme-cli  with the following version on RHEL 9.3 

- nvme version 2.4 (git 2.4)
- libnvme version 1.4 (git 1.4)

The nvme list command shows the namespace is fully used (It is an empty namespace), but we are still able to perform IO operations on it. 

P.s, the "df -h" does show the correct usage information. 
Can someone take a look at this behavior ?

[image]

[image]

ikegami-t, 2024-08-18:
Seems the namespace is formatted by the file system then the usage is full as written the file system structure. Also the df command usage is just showed the file system data usage only. (If you can execute the format command for the namespace the usage will be cleared. Note: The file system structure including the data used will be formatted by the command.)

keithbusch, 2024-08-18:
File system usage and low level media utilization are not the same thing. Media usage is always device specific. The filesystem can help if you've enabled discards, but that's not guaranteed to accomplish anything observable from the host though.

jmvaswani, 2024-08-19:
Hi, Thanks for the quick response! 

I tried performing a namespace format command, but it was throwing the following error.  Any idea about this? 

P.s, the namespace is provided by Ceph's NVME/TCP gateway.

[image]

ikegami-t, 2024-08-19:
Sorry you need to check the oacs and fna support values as below.
```sh
tokunori@tokunori-desktop:~/nvme-cli$ nvme id-ctrl /dev/nvme0 -H
...
oacs      : 0x5f
...
  [1:1] : 0x1   Format NVM Supported
  [0:0] : 0x1   Security Send and Receive Supported
...
fna       : 0x6
  [3:3] : 0     Format NVM Broadcast NSID (FFFFFFFFh) Supported
  [2:2] : 0x1   Crypto Erase Supported as part of Secure Erase
  [1:1] : 0x1   Crypto Erase Applies to All Namespace(s)
  [0:0] : 0     Format Applies to Single Namespace(s)
...
tokunori@tokunori-desktop:~/nvme-cli$ nvme-build id-ctrl /dev/nvme1 -H
...
oacs      : 0x17
  [10:10] : 0   Lockdown Command and Feature Not Supported
...
  [1:1] : 0x1   Format NVM Supported
  [0:0] : 0x1   Security Send and Receive Supported
...
fna       : 0x4
  [3:3] : 0     Format NVM Broadcast NSID (FFFFFFFFh) Supported
  [2:2] : 0x1   Crypto Erase Supported as part of Secure Erase
  [1:1] : 0     Crypto Erase Applies to Single Namespace(s)
  [0:0] : 0     Format Applies to Single Namespace(s)
```

keithbusch, 2024-08-19:
Sounds like your NVMe target may not implement namespace utilization. Many of them don't, so that's not uncommon

closed 2024-08-20 by igaw