linux-nvme/nvme-cli #2474: nvme list can't refresh the firmware version

banklpjhgb, 2024-08-23:
I'm using nvme 2.7.1 and there's some problems :

/ # nvme fw-download /dev/nvme0 -f /cytest/52F1U1TC.bin                       
Firmware download success
/ # nvme fw-commit /dev/nvme0 -s=1 -a=3                                             
Success committing firmware action:3 slot:1
/ # nvme list
Node                  Generic               SN                   Model                                    Namespace  Usage                      Format           FW Rev  
--------------------- --------------------- -------------------- ---------------------------------------- ---------- -------------------------- ---------------- --------
/dev/nvme0n1          ng0n1                 H551482004           TS32GMTE452TI                        0x1          9.71  GB /  32.02  GB    512   B +  0 B   52F1U1TB
/ # nvme list-ctrl /dev/nvme0                      
NVMe status: Invalid Field in Command: A reserved coded value or an unsupported value in a defined field(0x2)s: Invalid Field in Command: A reserved coded value or an unsupported value in a defined field(0x2)

Why the FW Rev still be old name after fw-commit? In fact the fw-commit is cucceed, because I can see the new version using smartctl, and why the list-ctrl is faild?
Thranks for your help.

igaw, 2024-08-23:
> Why the FW Rev still be old name after fw-commit? 

nvme-cli/libnvme reads the firmware version from sysfs. The kernel didn't update the firmware version entry in older version. a8eb6c1ba48b ("nvme: copy firmware_rev on each init")

Is there anything in the kernel messages? Does the problem go away after a reboot (device reset)?

banklpjhgb, 2024-08-23:
yes, the output will be right after reboot. I checked the code and the patch is not applied， I will test if it's OK when patch applied.
Thanks for your help!

igaw, 2024-08-23:
The kernel message should contain a message if a reboot/reset of the device is necessary after updating the firmware as far I know.

mi-fan, 2025-01-20:
I also encountered this issue. After a successful firmware upgrade, using nvme-cli 2.11 to execute nvme list still shows the old FW rev, but when I use an earlier version, nvme-cli 1.8, it shows the new version. However, using nvme id-ctrl shows the new version in both cases. Does anyone know why?

igaw, 2025-01-20:
https://github.com/linux-nvme/nvme-cli/wiki/FAQ#nvme-list-shows-old-firmware-version-after-firmware-udpate

banklpjhgb, 2025-01-20:
> I also encountered this issue. After a successful firmware upgrade, using nvme-cli 2.11 to execute `nvme list` still shows the old FW rev, but when I use an earlier version, nvme-cli 1.8, it shows [...]

a8eb6c1ba48b ("nvme: copy firmware_rev on each init")
This patch do work.

mi-fan, 2025-01-21:
The kernel version on my machine is 5.4, and it is difficult to apply a kernel patch on this OS. 
Is it possible to achieve the same effect by modifying the nvme-cli v2 source code?

igaw, 2025-01-21:
If you want to use the latest upstream version of nvme-cli with an older kernel, something might not work. You have two options, use the version provided by your distro or update/patch your kernel.

In this particular case, it's a deliberate decision not to support older kernels which do not update the sysfs firmware version after an update, because it causes problems. As I said above, after a reboot all is good or you use `nvme id-ctrl` to verify the firmware version.

closed 2025-01-21 by igaw