linux-nvme/nvme-cli #2494: [NVMeOF Host] Question with `nr-write-queues` in `nvme connect`

JigaoLuo, 2024-09-05:
Hi,

I am testing NVMe-oF using two nodes connected via RDMA and noticed some confusion regarding queue parameters when using `nvme connect`, specifically `nr-write-queues`, `nr-io-queues`, and `nr-poll-queues`. I set these parameters during the connection setup to observe their effects.

After setting up the NVMe-oF target using the Linux Kernel, I attempted to connect the host to the target SSD with the following command:
```bash
sudo /usr/local/sbin/nvme connect -t rdma -s 4420 -n nqn.2018-09.io.spdk:cnode0 -a 172.18.94.70 -w 172.18.94.81 -k 30 --nr-io-queues=32 --nr-write-queues=16
```

However, on the host side, I observed the following dmesg logs:
```
[9909557.917203] nvme nvme4: creating 48 I/O queues.
[9909569.324911] nvme nvme4: mapped 16/32/0 default/read/poll queues.
[9909569.335139] nvme nvme4: new ctrl: NQN "nqn.2018-09.io.spdk:cnode0", addr 172.18.94.70:4420
```

There are two main points of confusion:
1. **Write Queues Not Explicitly Mentioned**: Despite setting `nr-write-queues` in the `nvme connect` command, the dmesg output only shows read queues being created. There is no clear mention of the write queues that were specified.
2. **Inconsistent and Non-Intuitive Queue Mapping**: The number of queues shown in the dmesg logs doesn’t match intuitively with the parameters set in the `nvme connect` command. For instance, it is unclear if the 32 read queues in dmesg correspond to the `--nr-io-queues` flag and if the 16 default queues correspond to the `--nr-write-queues` flag (?)

Could you please clarify how the queue parameters are interpreted and mapped in the connection process? Specifically, how are `nr-write-queues` and `nr-io-queues` handled, and why do the dmesg logs not explicitly reflect the write queues set in the command?

Thank you for your assistance!

---

Here's my nvme version:
```bash
$ nvme --version
nvme version 2.6 (git 2.6-33-gd812d09)
libnvme version 1.6 (git 1.6-9-gda8c28e)
```

JigaoLuo, 2024-09-06:
I found this function which explains a little: https://github.com/torvalds/linux/blob/b831f83e40a24f07c8dcba5be408d93beedc820f/drivers/nvme/host/fabrics.c#L1128-L1169

igaw, 2024-09-09:
Indeed this part is not very easy to grasp. The kernel has three types of queues `default`, `read` and `poll`. 
- `default` read and write ops share the same queue
- `read` read ops only
- `poll` only for poll commands

For some reason, the kernel exposes `nr_write_queues` and not `nr_read_queues`. That means the number of `read` queues are defined by `nr_queues - nr_write_queues - nr_poll_queues`.

The kernel reports the number of queues as `default/read/poll`.

JigaoLuo, 2024-09-09:
Hi @igaw ,

Thank you for your response; it aligns well with the code above and has provided me with a clear understanding. I will go ahead and close this discussion.

Additionally, I found that all PCIe SSDs use the queue parameters specified by the kernel module. However, NVMe-oF SSDs connected via `nvme connect` can have individual and different queue parameters specified directly within the `nvme connect` command.

Thanks again!

closed 2024-09-09 by JigaoLuo