linux-nvme/nvme-cli #2520: wdc plugin commands are not working with nvme-cli 2.10.2

sarabjotsoit, 2024-10-02:
wdc plugin commands are not working with nvme-cli 2.10.2

With nvme-cli 1.16

root@blr-r29-26u:~/NVME_1.16/nvme-cli-1.16# ./nvme  wdc get-drive-status /dev/nvme1
ERROR : WDC : wdc_get_pci_ids : Open vendor file failed
ERROR : WDC : wdc_get_pci_ids : Open vendor file failed
ERROR : WDC : wdc_get_pci_ids : Open vendor file failed
ERROR : WDC : wdc_get_pci_ids : Open vendor file failed
ERROR : WDC : wdc_get_pci_ids : Open vendor file failed
ERROR : WDC : wdc_get_pci_ids : Open vendor file failed
ERROR : WDC : wdc_get_pci_ids : Open vendor file failed
ERROR : WDC : wdc_get_pci_ids : Open vendor file failed
ERROR : WDC : wdc_get_pci_ids : Open vendor file failed
ERROR : WDC : wdc_get_pci_ids : Open vendor file failed
ERROR : WDC : wdc_get_pci_ids : Open vendor file failed
ERROR : WDC : wdc_get_pci_ids : Open vendor file failed
ERROR : WDC : wdc_get_pci_ids : Open vendor file failed
  Drive Status :- 
  Percent Used:                         0%
  Drive Life Status:                    Normal
  Assert Dump Status:                   Not Present
  Thermal Throttling Status:            Off
  Format Corrupt Reason:                Format Not Corrupted

With nvme-cli 2.10.2

root@blr-r29-26u:~/NVME_1.16/nvme-cli-1.16# nvme  wdc get-drive-status /dev/nvme1
ERROR: WDC: wdc_get_pci_ids : Open vendor file failed
ERROR: WDC: wdc_get_pci_ids : Open vendor file failed
ERROR: WDC: get pci ids: -1
ERROR: WDC: 0xC2 Log Page entry ID 0x8 not found
ERROR: WDC: wdc_get_pci_ids : Open vendor file failed
ERROR: WDC: get pci ids: -1
ERROR: WDC: 0xC2 Log Page entry ID 0x8 not found
ERROR: WDC: wdc_get_pci_ids : Open vendor file failed
ERROR: WDC: get pci ids: -1
ERROR: WDC: 0xC2 Log Page entry ID 0x8 not found
ERROR: WDC: wdc_get_pci_ids : Open vendor file failed
ERROR: WDC: get pci ids: -1
ERROR: WDC: 0xC2 Log Page entry ID 0x8 not found
ERROR: WDC: wdc_get_pci_ids : Open vendor file failed
ERROR: WDC: get pci ids: -1
wdc_get_fw_cust_id: ERROR: WDC: 0xC2 Log Page entry ID 0x15 not found
wdc_get_enc_drive_capabilities: ERROR: WDC: invalid customer id
ERROR: WDC: wdc_get_pci_ids : Open vendor file failed
ERROR: WDC: get pci ids: -1
ERROR: WDC: 0xC2 Log Page entry ID 0x8 not found
ERROR: WDC: 0xC2 Log Page not supported

igaw, 2024-10-02:
Are you sure that v1.16 is actually working? The output looks like that the command fails and just prints out 0 values.

@jeff-lien-wdc

sarabjotsoit, 2024-10-02:
HI, @igaw  @jeff-lien-wdc 

yes, 1.16 is working.

Adding o/p for id-ctrl with 2.10.2

nvme id-ctrl /dev/nvme1
NVME Identify Controller:
vid       : 0x1b96
ssvid     : 0x1b96
sn        : 243302900004        
mn        : WUS5EC1B1ESP7Y4                         
fr        : RC910001
rab       : 4
ieee      : 0014ee
cmic      : 0x3
mdts      : 3
cntlid    : 0xc002
ver       : 0x10400
rtd3r     : 0
rtd3e     : 0
oaes      : 0x300
ctratt    : 0x41
rrls      : 0
cntrltype : 1
fguid     : 00000000-0000-0000-0014-ee8312ff6980
crdt1     : 400
crdt2     : 270
crdt3     : 140
nvmsr     : 1
vwci      : 138
mec       : 3
oacs      : 0x5f
acl       : 3
aerl      : 3
frmw      : 0x19
lpa       : 0x5f
elpe      : 255
npss      : 0
avscc     : 0x1
apsta     : 0
wctemp    : 350
cctemp    : 355
mtfa      : 100
hmpre     : 0
hmmin     : 0
tnvmcap   : 30720827326464
unvmcap   : 896574423040
rpmbs     : 0
edstt     : 30
dsto      : 1
fwug      : 255
kas       : 10
hctma     : 0
mntmt     : 0
mxtmt     : 0
sanicap   : 0x60000003
hmminds   : 0
hmmaxd    : 0
nsetidmax : 0
endgidmax : 0
anatt     : 0
anacap    : 0
anagrpmax : 0
nanagrpid : 0
pels      : 160
domainid  : 0
megcap    : 0
sqes      : 0x66
cqes      : 0x44
maxcmd    : 16
nn        : 128
oncs      : 0x54
fuses     : 0
fna       : 0x4
vwc       : 0x4
awun      : 63
awupf     : 63
icsvscc   : 1
nwpc      : 0
acwu      : 63
ocfs      : 0
sgls      : 0x100005
mnan      : 0
maxdna    : 0
maxcna    : 0
oaqd      : 0
subnqn    : nqn.1992-05.com.wdc.ofdata24-4213-uscos02823sb0009-Sarabjot--------S:nvme.1
ioccsz    : 4
iorcsz    : 1
icdoff    : 0
fcatt     : 0
msdbd     : 1
ofcs      : 0
ps      0 : mp:20.00W operational enlat:400 exlat:400 rrt:0 rrl:0
            rwt:0 rwl:0 idle_power:- active_power:-
            active_power_workload:-

jeff-lien-wdc, 2024-10-02:
@sarabjotsoit What is the pci device id of the WD drive are you running this command against?
  
The "Open vendor file failed" message is because the code can not access the sys fs file - /sys/class/nvme/nvme1/device/vendor. 

Even though the command seems to work on version 1.16, the code is unable to read the pci vendor id and therefore not getting valid data.  It should be failing with an unsupported device error.   

On version 2.10.2, the code is also not able to read the vendor id.  In addition to that, it appears the WD Vendor Unique C2 log page is not supported.  

I'd say this command is not supported by the wdc plugin for the drive you have.  The failure  messages are not very clear though and could be improved.

igaw, 2024-10-03:
IIRC, `vid` and `ssvid` should map the the PCI ids.

sarabjotsoit, 2024-10-04:
HI, @jeff-lien-wdc  Ping @igaw 

Got a private fix from Jeff and the drive status is coming up fine. Thank you.

root@blr-r29-26u:~# nvme wdc get-drive-status /dev/nvme1
ERROR: WDC: wdc_get_pci_ids : Open vendor file failed
ERROR: WDC: wdc_get_pci_ids : Open vendor file failed
ERROR: WDC: wdc_get_pci_ids : Open vendor file failed
ERROR: WDC: wdc_get_pci_ids : Open vendor file failed
ERROR: WDC: wdc_get_pci_ids : Open vendor file failed
ERROR: WDC: wdc_get_pci_ids : Open vendor file failed
ERROR: WDC: wdc_get_pci_ids : Open vendor file failed
ERROR: WDC: wdc_get_pci_ids : Open vendor file failed
ERROR: WDC: wdc_get_pci_ids : Open vendor file failed
ERROR: WDC: wdc_get_pci_ids : Open vendor file failed
ERROR: WDC: wdc_get_pci_ids : Open vendor file failed
ERROR: WDC: wdc_get_pci_ids : Open vendor file failed
ERROR: WDC: wdc_get_pci_ids : Open vendor file failed
  Drive Status :-
  Percent Used:                         0%
  Drive Life Status:                    Normal
  Assert Dump Status:                   Not Present
  Thermal Throttling Status:            Off
  Format Corrupt Reason:                Format Not Corrupted
root@blr-r29-26u:~# nvme version
nvme version 2.10.2 (git 2.10.2)
libnvme version 1.10 (git 1.10)

Although wdc_get_pci_id are showing failure, can we improve them as well.

jeff-lien-wdc, 2024-10-04:
I'm working on a fix for this issue now.  I hope to have a PR up for review early next week.

jeff-lien-wdc, 2024-10-07:
Created PR - https://github.com/linux-nvme/nvme-cli/pull/2528 to resolve this problem.

closed 2024-10-08 by igaw