linux-nvme/nvme-cli #2566: cli ocp 0xc6

YUEyuan310, 2024-11-13:
### There is an issue with the display of CLI OCP 0xc6.Failure to comply with protocol OCP2.5.
long double log_size = uint128_t_to_double(le128_to_cpu(log->size)) * sizeof(__le32);
According to the OCP2.5 protocol, the total length of Component Descriptions should be log_size-64.
When displayed in CLI, 64 was not subtracted, resulting in the display of redundant invalid Component Descriptions events.
[image]
[image]
The red box in the picture should not appear.
The following is environmental information
[image]

``
static void stdout_hwcomp_log(struct hwcomp_log *log, __u32 id, bool list)
{
	size_t date_lot_code_offset = sizeof(struct hwcomp_desc);
	int num = 1;
	struct hwcomp_desc_entry e = { log->desc };

	long double log_size = uint128_t_to_double(le128_to_cpu(log->size)) * sizeof(__le32);

	printf("Log Identifier: 0x%02xh\n", LID_HWCOMP);
	printf("Log Page Version: 0x%x\n", le16_to_cpu(log->ver));
	print_array("Reserved2", log->rsvd2, ARRAY_SIZE(log->rsvd2));
	print_array("Log page GUID", log->guid, ARRAY_SIZE(log->guid));
	printf("Hardware Component Log Size: 0x%"PRIx64"\n", (uint64_t)log_size);
	print_array("Reserved48", log->rsvd48, ARRAY_SIZE(log->rsvd48));
	printf("Component Descriptions\n");
	while (log_size > 0) {     **// log_size was not subtracted by 64 in advance.**
		e.date_lot_size = le64_to_cpu(e.desc->date_lot_size) * sizeof(__le32);
		e.date_lot_code = e.date_lot_size ? (__u8 *)e.desc + date_lot_code_offset : NULL;
		e.add_info_size = le64_to_cpu(e.desc->add_info_size) * sizeof(__le32);
		e.add_info = e.add_info_size ? e.date_lot_code ? e.date_lot_code + e.date_lot_size :
		    (__u8 *)e.desc + date_lot_code_offset : NULL;
		if (!id || id == le32_to_cpu(e.desc->id))
			print_hwcomp_desc(&e, list, num++);
		e.desc_size = date_lot_code_offset + e.date_lot_size + e.add_info_size;
		e.desc = (struct hwcomp_desc *)((__u8 *)e.desc + e.desc_size);
		log_size -= e.desc_size;
	}
}

igaw, 2024-11-13:
@ikegami-t I think you are working on ocp, do you want to take this?

YUEyuan310, 2024-11-13:
> @ikegami-t I think you are working on ocp, do you want to take this?

yes, I am developing OCP 0xC6. ok

YUEyuan310, 2024-11-13:
> @ikegami-t I think you are working on ocp, do you want to take this?

Yes, I am developing OCP 0xC6. OK.

ikegami-t, 2024-11-13:
@YUEyuan310 Could you please try the PR changes #2568?

YUEyuan310, 2024-11-13:
> @YUEyuan310 Could you please try the PR changes #2568?

OK，I have also submitted a change to fix a display error.

igaw, 2024-11-13:
Oops, I've already merged #2669. Let me know if I should revert it and apply #2568 instead (though I'd prefer @ikegami-t  version)

YUEyuan310, 2024-11-13:
> Oops, I've already merged #2669. Let me know if I should revert it and apply #2568 instead (though I'd prefer @ikegami-t version)

@ikegami-t modified the memory allocation, and I submitted a fix for the display error. Both the changes need to be merged.

YUEyuan310, 2024-11-13:
> Oops, I've already merged #2669. Let me know if I should revert it and apply #2568 instead (though I'd prefer @ikegami-t version)

@ikegami-t  modified the memory allocation, and I submitted a fix for the display error. Both  changes need to be merged. If @ikegami-t makes changes and I don't, it could lead to issues like memory overruns.

sc108-lee, 2024-11-19:
Based on PR #2570, I added ocp 2.6 spec changes for this.

#2580 Log size is currently changed ( in Dwords -> in bytes )
HWCLOG-4 | 47:32 | Hardware Component Log Size | 16 | This is the size of the Hardware Component Log in bytes.

spec : https://www.opencompute.org/documents/datacenter-nvme-ssd-specification-v2-6-1-pdf

igaw, 2024-11-21:
I think I merged now the correct PR. If not just reopen this issue and let me know what I should merge. :)

closed 2024-11-21 by igaw