linux-nvme/nvme-cli #2573: connect-all fails when there is no .nvme keyring because CONFIG_NVME_TCP_TLS and CONFIG_NVME_HOST_AUTH are not enabled

bluca, 2024-11-14:
Since nvme-cli 2.11 I am seeing this issue in the systemd CI where nvme-cli is used. The nvme-ctp is loaded module is loaded:

```
[    4.741966] TEST-84-STORAGETM.sh[577]: + modprobe -v nvmet-tcp
[    4.750298] TEST-84-STORAGETM.sh[580]: insmod /lib/modules/6.11.5-cloud-amd64/kernel/drivers/nvme/target/nvmet.ko.xz
[    4.750298] TEST-84-STORAGETM.sh[580]: insmod /lib/modules/6.11.5-cloud-amd64/kernel/drivers/nvme/target/nvmet-tcp.ko.xz
[    4.750828] TEST-84-STORAGETM.sh[577]: + modprobe -v nvme-tcp
[    4.757204] TEST-84-STORAGETM.sh[585]: insmod /lib/modules/6.11.5-cloud-amd64/kernel/drivers/nvme/host/nvme-fabrics.ko.xz
[    4.757204] TEST-84-STORAGETM.sh[585]: insmod /lib/modules/6.11.5-cloud-amd64/kernel/drivers/nvme/host/nvme-tcp.ko.xz
```

The command, which used to work with nvme-cli 2.10, started failing since Debian upgraded to 2.11:

```
6234s [   25.416521] TEST-84-STORAGETM.sh[548]: + nvme connect-all -t tcp -a 127.0.0.1 -s 16858 --hostid=158f4d7d-6759-499e-b260-b152e49fa9a2
6234s [   25.446043] TEST-84-STORAGETM.sh[567]: Failed to set keyring
6234s [   25.446699] TEST-84-STORAGETM.sh[567]: failed to add controller, error Required key not available
```

Full strace output at https://github.com/systemd/systemd/issues/35130#issuecomment-2476359973

Is enabling those kconfigs and setting up TLS keys required now to use `nvme connect-all -t tcp`?

This is on kernel 6.11.5. Note that Debian does _not_ enable `CONFIG_NVME_TCP_TLS` and `CONFIG_NVME_HOST_AUTH`

igaw, 2024-11-15:
Sorry about this. From a quick glance, I'd say it should not fail hard. Let me look into it.

igaw, 2024-11-15:
I suppose I should do a libnvme 1.11.1 release after merging the fix.

bluca, 2024-11-15:
That would be very nice, thanks for the quick fix!

igaw, 2024-11-15:
Thanks for the great bug report! I've just released a 1.11.1. Hopefully Debian picks this up soon.

closed 2024-11-15 by igaw