linux-nvme/nvme-cli #2577: Field descriptors in OCP SMART Log Page are different to those in standard SMART Log Page

sbates130272, 2024-11-15:
There is a lack of consistency between the field descriptors in the standard SMART log page and the OCP SMART extended log page. This is rather annoying but also poses a bit of a challenge when trying to scrape the OCP metrics via a [Prometheus node-exporter text collector][ref-node-exporter].

I *think* it would be lovely to alter the format of the OCP output (both stdout and JSON) but I am concerned that might cause an issue for people who have already written tools that parse the existing log page output. So I am creating this issue so we can discuss if such a change is welcome or not.
```
$ nvme smart-log /dev/nvme0n1 -o json
{
  "critical_warning":0,
  "temperature":323,
<snip>
}
```
versus
```
$ nvme ocp smart-add-log /dev/nvme0n1 -o json
{
<snip>
  "Bad user nand blocks - Raw":0,
  "Bad user nand blocks - Normalized":0,
<snip>
}
```
[ref-node-exporter]: https://github.com/sbates130272/node-exporter-textfile-collector-scripts/tree/dev/stephen/nvme-ocp-update

igaw, 2024-11-21:
Changing the existing output is likely to cause regression, which I really like to avoid. We could introduce a output format version flag so the use can decide which version should be used.

sbates130272, 2024-11-21:
Note there is a conversation going on in prometheus-community/node-exporter-textfile-collector-scripts#228 that also deals with JSON output format changes and some issues around per controller and per namespace SMART log pages.

sbates130272, 2024-11-21:
@igaw I will try and put together a PR for this in the next few days. I've noted your comments above and will do something that aligns with those.

igaw, 2024-11-21:
Cool. You don't need to go overboard in the first version. I think we have to figure out how to make this maintainable somehow without duplicating too much.

sbates130272, 2024-11-22:
@igaw I threw something together that adds a ```--non-legacy``` flag and then selects the JSON fields based on that. Take a look and see if you think it looks semi-sane. Then I can clean it up a bit and add documentation updates.

igaw, 2024-12-02:
Thanks for the PR. I like where this is going. I suggest though, to really introduce a `--format-version=1` for the legacy case and `--format-version=2` for the new. Because legacy and new will eventually replaced by next generation, followed by the even better one :)

igaw, 2024-12-02:
Hmm, I think it should be called `--output-format-version` to avoid collisions with other commands.

sbates130272, 2024-12-05:
Thanks @igaw. I will update the PR with that and include the documentation update. Stay tuned!

sbates130272, 2024-12-09:
@igaw can you take a look at my updated branch? I think I caught your requests and added a documentation update.

igaw, 2024-12-10:
Looks pretty good to me. As usual, Steven is updating the same lines right now. As your PR came before #2604. If this is okay everyone I'd merge this one first and Steven needs to update his PR. Hope this is okay. Sorry Steven. @sc108-lee

igaw, 2024-12-11:
I'm wondering if we should mark/treat the output of `v2` as experimental for the time so we can work on it until it works for your use case. `v1` is the default anyway.

sbates130272, 2024-12-11:
@igaw I can add an experimental comment to the top of the JSON output when version 2 is selected.

igaw, 2024-12-12:
JSON doesn't allow comments, as far I know. That means it would need something like suggest [here](https://stackoverflow.com/questions/244777/can-comments-be-used-in-json). Not sure about this approach though. Maybe just mention the experimental status in documentation?

sbates130272, 2024-12-12:
I like the documentation idea @igaw. I will do something along those lines.

sbates130272, 2024-12-16:
@igaw take a look at the new commit and let me know if it works for you now.

igaw, 2024-12-17:
Looks good to me. Please send a PR so I can merge it. Let's see else we have to change so that the smart log output is easy to process by any monitoring system.

sbates130272, 2024-12-17:
@igaw PR sent. Thanks!

PR #2624 (2024-12-20): ocp-nvme: Add a different formatting for JSON output
The current OCP JSON format for the SMART extended log page is not condusive to metric collection via tools like Prometheus. So we add a new output mode that uses all lower case and underscores (instead of spaces). This should help with metric collection. At the same time we clean up some of the field names. We add a new argument (--output-format-version) to allow us to select which output version we want. Documentation updated to reflect this change and mark this as experimental.

Fixes #2577.