linux-nvme/nvme-cli #2582: ocp uuidx

YUEyuan310, 2024-11-21:
Why do some OCP commands specify UUID, while others do not？

ocp_get_uuid_index(dev, &uuid_index);？

For example：
The command fw activate history specifies uuidx
The command hardware-component-log doesn't specify uuidx.

YUEyuan310, 2024-11-21:
Should the uuid function be completed?

ikegami-t, 2024-11-21:
Because it is required by the OCP specification. For example the error injection set features required it as below by the OCP version 2.6.
[image]
But it was not required by the OCP version 1.0.3 as below.
[image]

YUEyuan310, 2024-11-22:
> Sorry the answer above was not correct. The UUID is described by the set features commands but as @YUEyuan310 mentioned it is required for the get log page command also as below so needed to fix OCP get log page commands not set the uuid index to use it. ![uuid](https://private-user-images.githubusercontent.com/44117581/388579344-034e7faf-85ed-44c5-8b6d-b96eb4bc7fd8.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIyNDAyNDQsIm5iZiI6MTczMjIzOTk0NCwicGF0aCI6Ii80NDExNzU4MS8zODg1NzkzNDQtMDM0ZTdmYWYtODVlZC00NGM1LThiNmQtYjk2ZWI0YmM3ZmQ4LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTIyVDAxNDU0NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTk2ZDZhYzczZGM4YTU1ZGRmOWM2MDAzODdhYzMxZjhkYTBhZDU0YWE5ZmZhMmNlNmI5ODBmZmFiZTRiM2NhMjYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.CyhS5_MrovT7kANt-F6yyr7bSf4TyqapLY2IpZxYGJ8)

I share your opinion that the get-log command in OCP needs to have its uuidx functionality completed.

sc108-lee, 2024-11-22:
Yes I agree, we should add into the OCP get log commands.

igaw, 2024-12-20:
I've merged the PRs. Thanks everyone!

closed 2024-12-20 by igaw