linux-nvme/nvme-cli #2597: Using make static to compile nvme cli, nvme list cannot refresh firmware version

admin-cmd-shell, 2024-12-04:
As stated in the title,
After upgrading the firmware, the nvme list cannot refresh the firmware version
But using nvme id ctrl, you can see the new firmware version

Thranks for your help.

./nvme fw-download -f 9CV10490_WFEM0260_signed.bin /dev/nvme0n1
Firmware download success
[root@test .build]# ./nvme fw-commit /dev/nvme0 -s 0 -a 3
Success committing firmware action:3 slot:0
[root@test .build]# ./nvme reset /dev/nvme0
[root@test .build]# ./nvme list
Node                  Generic               SN                   Model                                    Namespace  Usage                      Format           FW Rev
--------------------- --------------------- -------------------- ---------------------------------------- ---------- -------------------------- ---------------- --------
/dev/nvme0n1          ng0n1                 PHAXXXXXXXXX15PFGN   INTEL SSDPF2KX153T1                      0x1         15.36  TB /  15.36  TB    512   B +  0 B   9CV10494
[root@test .build]# ./nvme list
Node                  Generic               SN                   Model                                    Namespace  Usage                      Format           FW Rev
--------------------- --------------------- -------------------- ---------------------------------------- ---------- -------------------------- ---------------- --------
/dev/nvme0n1          ng0n1                 PHAXXXXXXXXX15PFGN   INTEL SSDPF2KX153T1                      0x1         15.36  TB /  15.36  TB    512   B +  0 B   9CV10494

[root@test .build]# ./nvme id-ctrl /dev/nvme0n1
NVME Identify Controller:
vid       : 0x8086
ssvid     : 0x8086
sn        : PHAXXXXXXXXX15PFGN
mn        : INTEL SSDPF2KX153T1
fr        : 9CV10490
rab       : 0

admin-cmd-shell, 2024-12-04:
I tested the same version and found no issues with nvme cli installed using rpm

igaw, 2024-12-04:
How do you build the binary.  Please provide all commands and output. 

Are you using the same kernel for both tests? Older kernels are missing to update the firmware version in sysfs. In this case it will be correct after a reboot.

igaw, 2024-12-04:
Which version is the binary from rpm? Older rpm versions of nvme-cli will issue a command when doing an `nvme list`.

admin-cmd-shell, 2024-12-05:
Hi 
The commands used for compilation are as follows.
[root@Test opt]# git clone https://github.com/linux-nvme/nvme-cli.git
[root@Test opt]# cd nvme-cli
[root@Test nvme-cli]# ls
[root@Test nvme-cli]# git clone v2.10
[root@Test nvme-cli]# git checkout  v2.10
[root@Test nvme-cli]# make static
[root@Test nvme-cli]# cd .build/
[root@Test .build]# ls
[root@Test .build]# md5sum ./nvme
[root@Test .build]# md5sum /nvme
[root@Test .build]# ls
[root@Test .build]# nvme list
[root@Test .build]# ./nvme list
[root@Test .build]# ./nvme --version
[root@Test .build]# ./nvme fw-download -f 9CV10480_WFEM0260_signed.bin /dev/nvme0
[root@Test .build]# ./nvme fw-commit /dev/nvme0 -s 0 -a 3
[root@Test .build]# ./nvme list
[root@Test .build]# ./nvme reset /dev/nvme0
[root@Test .build]# ./nvme list
[root@Test .build]# grubby --set-default-index=2    _## Switching to kernel 4.18_
[root@Test .build]# reboot
[root@Test home]# cd /opt/
[root@Test opt]# cd nvme-cli/.build/
[root@Test .build]# ls
[root@Test .build]# ./nvme list
[root@Test .build]# ./nvme fw-download -f 9CV10490_WFEM0260_signed.bin /dev/nvme0
[root@Test .build]# ./nvme fw-commit /dev/nvme0 -s 0 -a 3
[root@Test .build]# ./nvme reset /dev/nvme0
[root@Test .build]# ./nvme list

[make_static_build_nvme-cli-v2.10.log](https://github.com/user-attachments/files/18017875/make_static_build_nvme-cli-v2.10.log)

I tested kernel4.18 and kernel6.12 in the same environment
There is an issue with kernel 4.18 not being able to refresh, but kernel 6.12 does not have this problem,
Is this as expected?
In the attached text, you can see that I switched the kernel,
Before restarting, the kernel 6.12 was compiled and tested,
After switching to kernel 4.18, I restarted and attempted to flash in the new firmware, but the nvme list did not display the new firmware version
Thank you.

admin-cmd-shell, 2024-12-05:
[root@Test .build]# grubby --info=ALL
index=0
kernel="/boot/vmlinuz-6.12.0"
args="ro crashkernel=auto resume=/dev/mapper/cl-swap rd.lvm.lv=cl/root rd.lvm.lv=cl/swap rhgb quiet $tuned_params"
root="/dev/mapper/cl-root"
initrd="/boot/initramfs-6.12.0.img $tuned_initrd"
title="CentOS Linux (6.12.0) 8"
id="f237b3229d2b41538e48e10038338181-6.12.0"
index=1
kernel="/boot/vmlinuz-4.18.0-348.7.1.el8_5.x86_64"
args="ro crashkernel=auto resume=/dev/mapper/cl-swap rd.lvm.lv=cl/root rd.lvm.lv=cl/swap rhgb quiet $tuned_params"
root="/dev/mapper/cl-root"
initrd="/boot/initramfs-4.18.0-348.7.1.el8_5.x86_64.img $tuned_initrd"
title="CentOS Linux (4.18.0-348.7.1.el8_5.x86_64) 8"
id="f237b3229d2b41538e48e10038338181-4.18.0-348.7.1.el8_5.x86_64"
index=2
kernel="/boot/vmlinuz-4.18.0-348.el8.x86_64"
args="ro crashkernel=auto resume=/dev/mapper/cl-swap rd.lvm.lv=cl/root rd.lvm.lv=cl/swap rhgb quiet $tuned_params"
root="/dev/mapper/cl-root"
initrd="/boot/initramfs-4.18.0-348.el8.x86_64.img $tuned_initrd"
title="CentOS Stream (4.18.0-348.el8.x86_64) 8"
id="f237b3229d2b41538e48e10038338181-4.18.0-348.el8.x86_64"
index=3
kernel="/boot/vmlinuz-0-rescue-f237b3229d2b41538e48e10038338181"
args="ro crashkernel=auto resume=/dev/mapper/cl-swap rd.lvm.lv=cl/root rd.lvm.lv=cl/swap rhgb quiet"
root="/dev/mapper/cl-root"
initrd="/boot/initramfs-0-rescue-f237b3229d2b41538e48e10038338181.img"
title="CentOS Stream (0-rescue-f237b3229d2b41538e48e10038338181) 8"
id="f237b3229d2b41538e48e10038338181-0-rescue"

admin-cmd-shell, 2024-12-05:
BTW，I am not very proficient in using GitHub, and I apologize for any inconvenience caused to you.

closed 2024-12-05 by igaw