linux-nvme/nvme-cli #2603: Unable to disconnect from NVMeoF subsystem

derekbit, 2024-12-05:
If using an NVMe initiator to connect to an NVMe target via NVMe-TCP, and the NVMe target node goes offline, resulting in the Paths in the subsystem displayed by `nvme list-subsys` being empty, how can I delete this subsystem? Or, what parameters should be provided during the connection to handle this situation?

```
# nvme list-subsys -o json
[
  {
    "HostNQN":"nqn.2014-08.org.nvmexpress:uuid:a48f0944-3ab0-4e5b-8019-79192500a44e",
    "Subsystems":[
      {
        "Name":"nvme-subsys0",
        "NQN":"nqn.2023-01.io.longhorn.spdk:pvc-4aa94de4-2f75-4cbb-855d-3737a488be50-e-0",
        "IOPolicy":"numa",
        "Paths":[]
      }
    ]
  }
]
```

I've tried the `nvme disconnect` function, but it doesn't work.
```
# nvme disconnect -n nqn.2023-01.io.longhorn.spdk:pvc-4aa94de4-2f75-4cbb-855d-3737a488be50-e-0
NQN:nqn.2023-01.io.longhorn.spdk:pvc-4aa94de4-2f75-4cbb-855d-3737a488be50-e-0 disconnected 0 controller(s)

# nvme list-subsys -o json
[
  {
    "HostNQN":"nqn.2014-08.org.nvmexpress:uuid:a48f0944-3ab0-4e5b-8019-79192500a44e",
    "Subsystems":[
      {
        "Name":"nvme-subsys0",
        "NQN":"nqn.2023-01.io.longhorn.spdk:pvc-4aa94de4-2f75-4cbb-855d-3737a488be50-e-0",
        "IOPolicy":"numa",
        "Paths":[]
      }
    ]
  }
]
```

igaw, 2024-12-06:
```
[ 1072.479893] nvme nvme1: Removing ctrl
```
This messages says the nvme subsystem got informed from userspace to release all resources to the mentioned controller. I suspect that the transport driver is not performing the cleanup tasks. Is this with nvme-tcp?

derekbit, 2024-12-06:
> ```
> [ 1072.479893] nvme nvme1: Removing ctrl
> ```
> 
> This messages says the nvme subsystem got informed from userspace to release all resources to the mentioned controller. I suspect that the transport driver is not performing the cleanup tasks. Is this with nvme-tcp?

Yes, it is with nvme-tcp.

igaw, 2024-12-09:
Ah sorry, stupid me, you already mentioned that it it's nvme-tcp. Anyway, I've tried to replicated this with the current Linux head (6.13-rc1) and,

```shell 
# nvme connect -t tcp -a 192.168.154.145 -s 4420 -n nqn.io-1 --hostnqn  nqn.2014-08.org.nvmexpress:uuid:befdec4c-2234-11b2-a85c-ca77c773af3 
[block traffic between host and controller]
# nvme disconnect-all
# nvme list-subsys 
[no output]
```

and the kernel log doesn't show any thing suspecious.

```
[   65.062189] nvme nvme1: creating 8 I/O queues.
[   65.071495] nvme nvme1: mapped 8/0/0 default/read/poll queues.
[   65.082160] nvme nvme1: new ctrl: NQN "nqn.io-1", addr 192.168.154.145:4420, hostnqn: nqn.2014-08.org.nvmexpress:uuid:befdec4c-2234-11b2-a85c-ca77c773af36
[   75.573965] nvme nvme1: Removing ctrl: NQN "nqn.io-1"
[   90.188630] nvme nvme1: creating 8 I/O queues.
[   90.197960] nvme nvme1: mapped 8/0/0 default/read/poll queues.
[   90.205534] nvme nvme1: new ctrl: NQN "nqn.io-1", addr 192.168.154.145:4420, hostnqn: nqn.2014-08.org.nvmexpress:uuid:befdec4c-2234-11b2-a85c-ca77c773af36
[  148.957642] nvme nvme1: I/O tag 1 (7001) type 4 opcode 0x18 (Keep Alive) QID 0 timeout
[  148.958107] nvme nvme1: starting error recovery
[  148.959373] nvme nvme1: failed nvme_keep_alive_end_io error=10
[  148.968537] nvme nvme1: Reconnecting in 10 seconds...
[  155.864828] nvme nvme1: Removing ctrl: NQN "nqn.io-1"
[  156.035038] nvme nvme1: Property Set error: 880, offset 0x14
```

Is the above command sequence what you are doing? If not, please provide those. Which kernel are  you using?

igaw, 2024-12-16:
Okay, I haven't tested the path, when the retry counter hits the limit and it gets auto removed. Let's see...

derekbit, 2025-01-03:
@igaw
Thanks for the help.
We also encounter the issue. The device is unable to be disconnected. Do you have any thoughts? 
```
instance-manager-76f5303efa69a5131572cedfe2bee640:/ # nvme list-subsys
nvme-subsys0 - NQN=nqn.2023-01.io.longhorn.spdk:e2e-test-volume-0-e-0
               hostnqn=nqn.2014-08.org.nvmexpress:uuid:ec2cad69-0c04-a806-2dc6-585615cc07ff
               iopolicy=numa
\
 +- nvme0 tcp traddr=10.42.3.6,trsvcid=20108 deleting (no IO)
```

igaw, 2025-01-03:
Is there a way to reproduce it?

Hmm, so the connection is in deleting (no IO) state. That might help to identify the problem. Maybe we are waiting on an request to complete but we never end it...

Omar007, 2025-01-23:
I wish I could provide more info but the 'have the remote disconnect/become unavailable and wait past its retries' seems to be all it takes. Not aware of anything special at the moment. Currently on kernel 6.12.10 and it had happened again.  

Looking at the `/sys/class/nvme-subsystem` tree, the whole node is still present when this happens. The only difference with an active subsystem is the absence of the controller symlink.  

Could the `/sys` tree be used to force it to try again or reinitialize the connection and/or controller?

Omar007, 2025-05-22:
I just had it happen on 6.14.4. In this instance the remote had hard-crashed (unrelated crash) instead of a normal termination.  

Is there a way to set it to just retry indefinitely and not give up or at least greatly extend the retry period?

hreinecke, 2025-07-25:
This sound like the issue I had been chasing in f42d4796ee10 ("nvme-tcp: fix I/O stalls on congested sockets"). So I really think this is a kernel issue, and nothing which we can do to fix it here.

igaw, 2025-10-07:
Thanks for testing and confirming that newer kernels don't have this behavior. So I am going to close this one for now.

Omar007, 2025-10-07:
Would be nice if I wasn't the only one confirming (the whole N=1 thing xD) but it also looks like it's included in 6.16.8; I had a node running that and it also did not show the issue 👍

derekbit, 2025-10-07:
Thanks @Omar007. [Longhorn team ](https://github.com/longhorn)will check it soon.

igaw, 2026-09-24:
1. Subsystem shows Paths: [] and nvme disconnect reports 0 controllers.
nvme disconnect works by deleting controllers (/sys/class/nvme/nvmeX/delete_controller). Here the controller was already removed by the kernel after reaching ctrl_loss_tmo ("Removing controller..." in the log), so nothing is left for nvme-cli to disconnect. The kernel has no interface to delete a subsystem. The subsystem object goes away by itself once nothing references it. If it doesn't, something is still holding one of its namespaces open: a mounted filesystem, a process, or a stacked device such as device-mapper (check /sys/block/nvme*n*/holders/, lsof, dmsetup ls). Once that holder is released, the subsystem disappears.
To avoid the controller being removed in the first place when the target is gone temporarily, connect with a longer or infinite loss timeout, e.g. nvme connect ... --ctrl-loss-tmo=-1 (reconnect forever), or pick a value that matches your expected outage window. The log here shows the controller gave up after roughly 30 seconds.
 
2. Controller stuck in deleting (no IO).
This is a kernel bug in nvme-tcp teardown. It is fixed upstream (see f42d4796ee10 "nvme-tcp: fix I/O stalls on congested sockets" and related changes). It has been confirmed gone on 6.14, 6.16.8 and 6.18. The SLES 16 6.12 kernel still shows it despite having that backport, so some other fix between 6.12 and 6.14 is missing there. Please report that to SUSE, since it can't be fixed from userspace.
Neither case is something nvme-cli can fix, so I'm closing this. Feel free to reopen if you see it on a current upstream kernel.

closed 2026-09-24 by igaw