linux-nvme/nvme-cli #2606: nvme discovery is not working with cli v2.11

sarabjotsoit, 2024-12-06:
Steps to reproduce:

1. Power ON the chassis with NVMe drives.
2. Perform the discover operation with nvme cli 2.11

Observation:
nvme discovery is not working with cli v2.11

**# nvme discover -t rdma -a 10.10.10.1 -s 4420
failed to add controller, error Operation not supported
#nvme version
nvme version 2.11 (git 2.11)
libnvme version 1.11 (git 1.11)**

igaw, 2024-12-09:
Does it work with older versions of `nvme-cli`, is this a regression? Could you post the output from 

  `strace -o /tmp/strace.log nvme discover -t rdma -a 10.10.10.1 -s 4420 -vv`

and the file `/tmp/strace.log` file.

sarabjotsoit, 2024-12-09:
[strace.log](https://github.com/user-attachments/files/18061768/strace.log)
Hi, @igaw 
Yes, we have tried with 1.16 and is working fine. not tried the 2. X series.

Please find the attached strace data.

igaw, 2024-12-10:
I see that you don't have `/etc/nvme/hostnqn` and `/etc/nvme/hostid`, thus the library reads from sysfs the dmi `product_uuid` entry and generates on the fly the hostnqn. And there is no interaction with the kernel. 

```
openat(AT_FDCWD, "/usr/etc/nvme/hostid", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/etc/nvme/hostnqn", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/sys/class/dmi/id/product_uuid", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0400, st_size=4096, ...}) = 0
read(3, "4c4c4544-0056-3010-8053-c2c04f4c"..., 4096) = 37
close(3)                                = 0
fstat(2, {st_mode=S_IFCHR|0620, st_rdev=makedev(0x88, 0), ...}) = 0
lseek(2, 0, SEEK_CUR)                   = -1 ESPIPE (Illegal seek)
write(2, "warning: using auto generated ho"..., 49) = 49
write(2, "failed to add controller, error "..., 56) = 56
```

Properly, the library trips over when having to generate the hostnqn on the fly.

igaw, 2024-12-10:
Could you also post the output from?

  `nvme discover -t rdma -a 10.10.10.1 -s 4420 -vv`

Thanks

igaw, 2024-12-10:
Ah wait, please update libnvme to 1.11.1. This is a duplicate of https://github.com/linux-nvme/nvme-cli/issues/2555

sarabjotsoit, 2024-12-11:
Thank you @igaw. This is working fine with the libnvme 1.11.1 for Redhat 9.2

closed 2024-12-11 by igaw