linux-nvme/nvme-cli #2615: nvme connect-all is not working with /etc/nvme/discovery.conf file

Sayoojntly, 2024-12-12:
**Steps**:

1. Load the nvme modules.
2. Add the connection parameters in /etc/nvme/discovery.conf
3. Connect the drives using nvme connect-all

**Observation**:
nvme connect-all is not working with /etc/nvme/discovery.conf file

nvme cli version
```
nvme version 2.11 (git 2.11)
libnvme version 1.11.1 (git 1.11.1)
```

From config file
cat /etc/nvme/discovery.conf
```
# Used for extracting default parameters for discovery

-t rdma -a 10.10.10.1 -s 4420 -i 8
-t rdma -a 20.20.20.1 -s 4420 -i 8
#
# Example:
# --transport=<trtype> --traddr=<traddr> --trsvcid=<trsvcid> --host-traddr=<host-traddr> --host-iface=<host-iface>
```

Sayoojntly, 2024-12-12:
Working fine with nvnne cli 1.16
```
nvme version
nvme version 1.16
# nvme connect-all
Failed to write to /dev/nvme-fabrics: Input/output error
Failed to write to /dev/nvme-fabrics: Input/output error
```

igaw, 2024-12-18:
It looks like the build for nvme-cli used `/usr` as  prefix:

```
openat(AT_FDCWD, "/usr/etc/nvme/discovery.conf", O_RDONLY) = 3
```

Either put the discovery.conf file to `/user/etc/nvme/discover.conf` or use `--prefix=/` for the build.

closed 2024-12-18 by igaw

sarabjotsoit, 2024-12-18:
HI, @igaw 

The above-mentioned steps are a workaround or are you planning to fix it?

igaw, 2024-12-18:
There is nothing to broken. It works exactly as design. 

In your case placing the `discovery.conf` into `/usr/etc` is a workaround. So if you want nvme-cli to pick up `discovery.conf` in `/etc/nvme/`, you need change your build setup.

During the configuration phase, it's possible to choose the `--prefix`. If it set to `--prefix=/usr` or/and `--sysconfdir=/usr/etc` then the nvme-cli expects the `discovery.conf` file to be in `/usr/etc/nvme`, thus `--prefix=/` will fix this. Be aware if you install the project, it will overwrite the version provided by your distro maker.

sarabjotsoit, 2024-12-18:
Thank you, @igaw  for details.