linux-nvme/nvme-cli #2618: Support for passing in PSK_KE/PSK_DHE_KE

sekar-wdc, 2024-12-12:
Hi,

I was wondering if there is or will be support for selecting the PSK key exchange mode (psk_ke/psk_dhe_ke) when initiating a TLS connection. It appears gnuTLS, which ktls-utils uses, supports both key exchange modes (psk_ke/psk_dhe_ke) but I don't see a way to select this via nvme-cli upon initiating a NVMe/TCP connection with TLS.

[image]

igaw, 2024-12-13:
Depends what the spec says.  I have a hard time to figure out what we are supposed to do

[image]

Is it mandatory to support both or can we choose? Anyway, it's a question what the spec wants us to do. 
@hreinecke do you have some input on this one?

sekar-wdc, 2024-12-13:
I think the spec does not allow it. I was thinking if there was a way to bypass the check but that does not seem possible.  Okay with closing this given the spec disallows it.

igaw, 2024-12-16:
Okay, let's close it then. And I suspect the storage vendors will use the Linux implementation to tests against, so if it doesn't implement/support it, no target will support it either :)

closed 2024-12-16 by igaw