linux-nvme/nvme-cli #2626: nvme telemetry-log is not working with nvme cli 2.11

Sayoojntly, 2024-12-18:
collect the  nvme telemetry-log using below command 

```
nvme version
nvme version 2.11 (git 2.11)
libnvme version 1.11.1 (git 1.11.1)
##########################################################################################
nvme telemetry-log /dev/nvme4  -g 0 -O jj.bin
ERROR: No telemetry data block
get-telemetry-log: No such device or address
##########################################################################################
cat /tmp/strace.log
execve("/usr/local/sbin/nvme", ["nvme", "telemetry-log", "/dev/nvme4", "-g", "0", "-O", "jj.bin"], 0x7ffd1f8ba720 /* 37 vars */) = 0
brk(NULL)                               = 0x19a4000
arch_prctl(0x3001 /* ARCH_??? */, 0x7ffe1ef9c490) = -1 EINVAL (Invalid argument)
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fa0c0a93000
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=46139, ...}) = 0
mmap(NULL, 46139, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7fa0c0a87000
close(3)                                = 0
openat(AT_FDCWD, "/lib64/libnvme.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\0\237\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=667432, ...}) = 0
mmap(NULL, 190592, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fa0c0a58000
mmap(0x7fa0c0a61000, 86016, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x9000) = 0x7fa0c0a61000
mmap(0x7fa0c0a76000, 45056, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1e000) = 0x7fa0c0a76000
mmap(0x7fa0c0a81000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x28000) = 0x7fa0c0a81000
close(3)                                = 0
openat(AT_FDCWD, "/lib64/libnvme-mi.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0P\"\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=134896, ...}) = 0
mmap(NULL, 37264, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fa0c0a4e000
mmap(0x7fa0c0a50000, 16384, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x2000) = 0x7fa0c0a50000
mmap(0x7fa0c0a54000, 8192, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x6000) = 0x7fa0c0a54000
mmap(0x7fa0c0a56000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x7000) = 0x7fa0c0a56000
close(3)                                = 0
openat(AT_FDCWD, "/lib64/libjson-c.so", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\220G\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=228304, ...}) = 0
mmap(NULL, 75176, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fa0c0a3b000
mmap(0x7fa0c0a3f000, 36864, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x4000) = 0x7fa0c0a3f000
mmap(0x7fa0c0a48000, 16384, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xd000) = 0x7fa0c0a48000
mmap(0x7fa0c0a4c000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x10000) = 0x7fa0c0a4c000
close(3)                                = 0
openat(AT_FDCWD, "/lib64/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0P\227\2\0\0\0\0\0"..., 832) = 832
pread64(3, "\6\0\0\0\4\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0"..., 784, 64) = 784
pread64(3, "\4\0\0\0 \0\0\0\5\0\0\0GNU\0\2\0\0\300\4\0\0\0\3\0\0\0\0\0\0\0"..., 48, 848) = 48
pread64(3, "\4\0\0\0\24\0\0\0\3\0\0\0GNU\0T\247\253\1\356\366\342\334\242\306\260\332\270\306V\241"..., 68, 896) = 68
fstat(3, {st_mode=S_IFREG|0755, st_size=2592552, ...}) = 0
pread64(3, "\6\0\0\0\4\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0"..., 784, 64) = 784
mmap(NULL, 2133936, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fa0c0800000
mprotect(0x7fa0c0828000, 1892352, PROT_NONE) = 0
mmap(0x7fa0c0828000, 1527808, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x28000) = 0x7fa0c0828000
mmap(0x7fa0c099d000, 360448, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x19d000) = 0x7fa0c099d000
mmap(0x7fa0c09f6000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1f5000) = 0x7fa0c09f6000
mmap(0x7fa0c09fc000, 53168, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7fa0c09fc000
close(3)                                = 0
openat(AT_FDCWD, "/lib64/libcrypto.so.3", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\0\360\n\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=4479984, ...}) = 0
mmap(NULL, 4391960, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fa0c0200000
mprotect(0x7fa0c02ad000, 3309568, PROT_NONE) = 0
mmap(0x7fa0c02ad000, 2469888, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xad000) = 0x7fa0c02ad000
mmap(0x7fa0c0508000, 835584, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x308000) = 0x7fa0c0508000
mmap(0x7fa0c05d5000, 364544, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x3d4000) = 0x7fa0c05d5000
mmap(0x7fa0c062e000, 9240, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7fa0c062e000
close(3)                                = 0
openat(AT_FDCWD, "/lib64/libz.so.1", O_RDONLY|O_CLOEXEC) = 3
[... 60 lines cut ...]
00000080: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000090: 0000 0000 0000 0000 0000 0000 0000 0000  ................
000000a0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
000000b0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
000000c0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
000000d0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
000000e0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
000000f0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000100: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000110: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000120: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000130: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000140: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000150: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000160: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000170: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000180: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000190: 0000 0000 0000 0000 641b 0000 0000 0000  ........d.......
000001a0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
000001b0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
000001c0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
000001d0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
000001e0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
000001f0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
```

igaw, 2024-12-18:
It looks like the device doesn't support it:

```
write(2, "ERROR: No telemetry data block", 30) = 30
write(2, "\n", 1)                       = 1
write(2, "get-telemetry-log: No such devic"..., 44) = 44
```

nvme-cli is working fine from what I see.  And I gave it a quick test on my local nvme disk and it works as expected.

closed 2024-12-18 by igaw

Sayoojntly, 2024-12-19:
Experiencing an issue where a configuration that works correctly with version 1.16 of nvme cli is not functioning as expected with version 2.11, even though the host and drive setup remain unchanged.
```
[root@WH2 nvme-cli-1.16]# ./nvme  telemetry-log /dev/nvme2 -o test3.bin -g 0
[root@WH2 nvme-cli-1.16]#

```
```
[root@WH2 #xxd test3.bin
00000000: 0700 0000 00ee 1400 0000 0000 0000 0000  ................
00000010: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000020: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000030: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000040: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000050: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000060: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000070: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000080: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000090: 0000 0000 0000 0000 0000 0000 0000 0000  ................
000000a0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
000000b0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
000000c0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
000000d0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
000000e0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
000000f0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000100: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000110: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000120: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000130: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000140: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000150: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000160: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000170: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000180: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000190: 0000 0000 0000 0000 cb17 0000 0000 0000  ................
000001a0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
000001b0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
000001c0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
000001d0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
000001e0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
000001f0: 0000 0000 0000 0000 0000 0000 0000 0000  ................

```

########################################################################
```

[root@WH2 ~]# nvme  telemetry-log /dev/nvme2 -o binary -g 0 -O test4
ERROR: No telemetry data block
get-telemetry-log: No such device or address

```

igaw, 2024-12-20:
On my machine, this
  
  `nvme telemetry-log /dev/nvme0n1 -o binary -g 0 -O /tmp/out`  

works fine with the current nvme-cli. 

Please add `-vv` to your command line and you should see something like:

```
opcode       : 02
flags        : 00
rsvd1        : 0000
nsid         : 00000000
cdw2         : 00000000
cdw3         : 00000000
data_len     : 00000200
metadata_len : 00000000
addr         : 686d000
metadata     : 0
cdw10        : 007f0007
cdw11        : 00000000
cdw12        : 00000000
cdw13        : 00000000
cdw14        : 00000000
cdw15        : 00000000
timeout_ms   : 00000000
result       : 00000000
err          : 0
latency      : 11043 us
opcode       : 02
flags        : 00
rsvd1        : 0000
nsid         : 00000000
cdw2         : 00000000
cdw3         : 00000000
data_len     : 00001000
metadata_len : 00000000
addr         : 7fde6ac56000
metadata     : 0
cdw10        : 03ff8007
cdw11        : 00000000
cdw12        : 00000000
cdw13        : 00000000
cdw14        : 00000000
cdw15        : 00000000
timeout_ms   : 00000000
result       : 00000000
err          : 0
latency      : 9813 us
...
```

The error is reported because `dalb3` reports 0. I've compared it what the 1.x branch does and all this looks pretty much identically. Maybe your device has troubles with how much data is requested. The 2.x branch fetches 512 bytes initially and from what I read the 1.x branch uses 4096.

What happens if you do:

```
nvme get-log /dev/nvme0n1 --log-id 0x07 --namespace-id 0 --log-len 512 -vv
```

```
nvme get-log /dev/nvme0n1 --log-id 0x07 --namespace-id 0 --log-len 4096 -vv
```

Sayoojntly, 2024-12-20:
Hi @igaw 
Please find the data requested.

**First command** 
```
[root@WH2 ~]# nvme telemetry-log /dev/nvme2 -o binary -g 0 -O /tmp/out
ERROR: No telemetry data block
get-telemetry-log: No such device or address
```

**With -vv**
```
[root@WH2 ~]#
[root@WH2 ~]# nvme telemetry-log /dev/nvme2 -o binary -g 0 -O /tmp/out -vv
opcode       : 02
flags        : 00
rsvd1        : 0000
nsid         : 00000000
cdw2         : 00000000
cdw3         : 00000000
data_len     : 00000200
metadata_len : 00000000
addr         : 1d6c000
metadata     : 0
cdw10        : 007f0007
cdw11        : 00000000
cdw12        : 00000000
cdw13        : 00000000
cdw14        : 00000000
cdw15        : 00000000
timeout_ms   : 00000000
result       : 00000000
err          : 0
latency      : 50318 us
ERROR: No telemetry data block
get-telemetry-log: No such device or address
[root@WH2 ~]#
[root@WH2 ~]#
```

**get log  command with -l 512**

```
[root@WH2 ~]# nvme get-log /dev/nvme2 --log-id 0x07 --namespace-id 0 --log-len 512 -vv
opcode       : 02
flags        : 00
rsvd1        : 0000
nsid         : 00000000
cdw2         : 00000000
cdw3         : 00000000
data_len     : 00000200
metadata_len : 00000000
addr         : a5f000
metadata     : 0
cdw10        : 007f0007
cdw11        : 00000000
cdw12        : 00000000
cdw13        : 00000000
cdw14        : 00000000
cdw15        : 00000000
timeout_ms   : 00000000
result       : 00000000
err          : 0
latency      : 40540 us
Device:nvme2 log-id:7 namespace-id:0
       0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
0000: 07 00 00 00 00 ee 14 00 00 00 00 00 00 00 00 00 "................"
0010: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0030: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
[... 291 lines cut ...]
0eb0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0ec0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0ed0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0ee0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0ef0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0f00: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0f10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0f20: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0f30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0f40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0f50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0f60: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0f70: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0f80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0f90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0fa0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0fb0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0fc0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0fd0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0fe0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0ff0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
[root@WH2 ~]#
```

Please let me know anything more required

igaw, 2024-12-20:
I got it wrong, also the 1.x branch uses 512 bytes to read the telemetry log header. 

And we see from the log, the device reports 0 for dalb3 (bytes 13-14)

```
       0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
0000: 07 00 00 00 00 ee 14 00 00 00 00 00 00 00 00 00 "................"
```

On my machine this is
```
       0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
0000: 07 00 00 00 00 38 25 00 00 00 00 00 40 1d 00 00 ".....8%.....@..."
```

I don't understand how the 1.x also outputs anything if dalb3 is 0. 

```c
	case 3:
		full_size = (le16_to_cpu(hdr->dalb3) * bs) + offset;
		break;
[...]

	while (offset != full_size) {
		err = nvme_get_telemetry_log(fd, page_log, 0, cfg.ctrl_init, bs, offset);
```

Anyway, 2.x implements and handles this correctly. Why the 1.x branch works for you I don't know. Maybe the version you are using has been patched because the upstream version should not work if I read it correctly.

sarabjotsoit, 2024-12-24:
Hi, @igaw 

Good Morning!

Want to clarify that this is as per design/expected in 2.x.x?

igaw, 2025-01-03:
Not sure what the question is. As I tried to explain I think the 2.x version does the dalb3 parsing correctly. In the raw logs from nvme 1.x we also see that the header (the first 16 bytes) are the same. 

The data structures are also identically.

nvme v1.x
```
struct nvme_telemetry_log_page_hdr {
        __u8                       lpi;                  /*     0     1 */
        __u8                       rsvd[4];              /*     1     4 */
        __u8                       iee_oui[3];           /*     5     3 */
        __le16                     dalb1;                /*     8     2 */
        __le16                     dalb2;                /*    10     2 */
        __le16                     dalb3;                /*    12     2 */
        __u8                       rsvd1[2];             /*    14     2 */
        __le32                     dalb4;                /*    16     4 */
        __u8                       rsvd2[362];           /*    20   362 */
        /* --- cacheline 5 boundary (320 bytes) was 62 bytes ago --- */
        __u8                       ctrlavail;            /*   382     1 */
        __u8                       ctrldgn;              /*   383     1 */
        /* --- cacheline 6 boundary (384 bytes) --- */
        __u8                       rsnident[128];        /*   384   128 */
        /* --- cacheline 8 boundary (512 bytes) --- */
        __u8                       telemetry_dataarea[]; /*   512     0 */

        /* size: 512, cachelines: 8, members: 13 */
};
```

libnvme v1:
```
struct nvme_telemetry_log {
        __u8                       lpi;                  /*     0     1 */
        __u8                       rsvd1[4];             /*     1     4 */
        __u8                       ieee[3];              /*     5     3 */
        __le16                     dalb1;                /*     8     2 */
        __le16                     dalb2;                /*    10     2 */
        __le16                     dalb3;                /*    12     2 */
        __u8                       rsvd14[2];            /*    14     2 */
        __le32                     dalb4;                /*    16     4 */
        __u8                       rsvd20[360];          /*    20   360 */
        /* --- cacheline 5 boundary (320 bytes) was 60 bytes ago --- */
        __u8                       ths;                  /*   380     1 */
        union {
                __u8               hostdgn;              /*   381     1 */
                __u8               tcs;                  /*   381     1 */
        };                                               /*   381     1 */
        __u8                       ctrlavail;            /*   382     1 */
        __u8                       ctrldgn;              /*   383     1 */
        /* --- cacheline 6 boundary (384 bytes) --- */
        __u8                       rsnident[128];        /*   384   128 */
        /* --- cacheline 8 boundary (512 bytes) --- */
        __u8                       data_area[];          /*   512     0 */

        /* size: 512, cachelines: 8, members: 15 */
};
```

Ah I figured out the issue. nvme v2 checks if the dalb has a valid value:

```
	if (dalb == 0) {
		nvme_show_error("ERROR: No telemetry data block");
		return -ENOENT;
	}
```

This check is missing in nvme v1, thus it seems it works but that is not the case. Your device is
misbehaving.