linux-nvme/nvme-cli #2651: Drive block size is not showing correctly after the format

Sayoojntly, 2025-01-09:
**Steps**:

The drive block size should be 4K.
Format the drive from 4K to 512B.
list the drive after format using nvme list.

Observation: 
Drive block size is not showing correctly after the format

nvme cli version
nvme version
nvme version 2.11 (git 2.11)
libnvme version 1.11.1 (git 1.11.1)

igaw, 2025-01-09:
So this could be just another of these here:

https://github.com/linux-nvme/nvme-cli/wiki/FAQ#nvme-list-shows-wrong-diskblock-size

Is the value correct after a reboot?

Sayoojntly, 2025-01-09:
Hi @igaw 
Disconnecting the drive and connecting back is showing the actual format.
 ```
[root@dhcp-#nvme disconnect-all
[root@dhcp-#nvme connect-all -t rdma -a 10.10.10.3 -s 4420 -i  8
[root@dhcp-#nvme connect-all -t rdma -a 20.20.20.3 -s 4420 -i 8
[root@dhcp-# nvme list
Node                  Generic               SN                   Model                                    Namespace  Usage                      Format           FW Rev
--------------------- --------------------- -------------------- ---------------------------------------- ---------- -------------------------- ---------------- --------
/dev/nvme1n1          /dev/ng1n1            23020M900263         WUS5EA138ESP7E1                          0x1          3.83  TB /   3.83  TB    512   B +  0 B   RC610008

```

igaw, 2025-01-09:
Ah this is with nvme-fabric (important info...). This is likely to a kernel issue not updating the sysfs after format command. The kernel is checking the side effects of each command, so it should in theory trigger a rescan after the format command. What kernel are you using? Is this also happening with the latest upstream kernel? 

BTW, nvme list-ns should show the right format, this is what nvme-cli 1.16 is doing.

Sayoojntly, 2025-01-09:
OS version and kernel added below
```

# cat /etc/rocky-release
Rocky Linux release 9.5 (Blue Onyx)
# uname -a
 5.14.0-503.14.1.el9_5.x86_64
```

nvme list 
```
nvme list
Node                  Generic               SN                   Model                                    Namespace  Usage                      Format           FW Rev
--------------------- --------------------- -------------------- ---------------------------------------- ---------- -------------------------- ---------------- --------
/dev/nvme1n1          /dev/ng1n1            23020M900263         WUS5EA138ESP7E1                          0x1         30.68  TB /   3.84  TB      4 KiB +  0 B   RC610008

```
The nvme id-ns is showing the right format
```
]# nvme id-ns /dev/nvme1n1 -H
NVME Identify Namespace 1:
nsze    : 0x1be7c7c60
ncap    : 0x1be7c7c60
nuse    : 0x1be7c7c60
nsfeat  : 0x12
  [5:4] : 0x1   NPWG, NPWA, NPDG, NPDA, and NOWS are Supported
  [3:3] : 0     NGUID and EUI64 fields if non-zero, Reused
  [2:2] : 0     Deallocated or Unwritten Logical Block error Not Supported
  [1:1] : 0x1   Namespace uses NAWUN, NAWUPF, and NACWU
  [0:0] : 0     Thin Provisioning Not Supported

nlbaf   : 3
flbas   : 0
  [6:5] : 0     Most significant 2 bits of Current LBA Format Selected
  [4:4] : 0     Metadata Transferred in Separate Contiguous Buffer
  [3:0] : 0     Least significant 4 bits of Current LBA Format Selected

mc      : 0x1
  [1:1] : 0     Metadata Pointer Not Supported
  [0:0] : 0x1   Metadata as Part of Extended Data LBA Supported

dpc     : 0x17
  [4:4] : 0x1   Protection Information Transferred as Last Bytes of Metadata Supported
  [3:3] : 0     Protection Information Transferred as First Bytes of Metadata Not Supported
  [2:2] : 0x1   Protection Information Type 3 Supported
  [1:1] : 0x1   Protection Information Type 2 Supported
  [0:0] : 0x1   Protection Information Type 1 Supported

dps     : 0
  [3:3] : 0     Protection Information is Transferred as Last Bytes of Metadata
  [2:0] : 0     Protection Information Disabled

nmic    : 0x1
  [0:0] : 0x1   Namespace Multipath Capable

rescap  : 0
  [7:7] : 0     Ignore Existing Key - Used as defined in revision 1.2.1 or earlier
  [6:6] : 0     Exclusive Access - All Registrants Not Supported
  [5:5] : 0     Write Exclusive - All Registrants Not Supported
  [4:4] : 0     Exclusive Access - Registrants Only Not Supported
  [3:3] : 0     Write Exclusive - Registrants Only Not Supported
  [2:2] : 0     Exclusive Access Not Supported
  [1:1] : 0     Write Exclusive Not Supported
  [0:0] : 0     Persist Through Power Loss Not Supported

fpi     : 0x80
  [7:7] : 0x1   Format Progress Indicator Supported
  [6:0] : 0     Format Progress Indicator (Remaining 0%)

dlfeat  : 9
  [4:4] : 0     Guard Field of Deallocated Logical Blocks is set to 0xFFFF
  [3:3] : 0x1   Deallocate Bit in the Write Zeroes Command is Supported
  [2:0] : 0x1   Bytes Read From a Deallocated Logical Block and its Metadata are 0x00

nawun   : 255
nawupf  : 255
nacwu   : 255
nabsn   : 7
nabo    : 0
nabspf  : 7
noiob   : 0
nvmcap  : 3,835,405,795,328
npwg    : 63
npwa    : 63
npdg    : 65535
npda    : 7
nows    : 0
mssrl   : 0
mcl     : 0
msrc    : 0
nulbaf  : 0
anagrpid: 0
nsattr  : 0
nvmsetid: 0
endgid  : 0
nguid   : d0070000000000000014ee840144b200
eui64   : 0014ee840144b200
LBA Format  0 : Metadata Size: 0   bytes - Data Size: 512 bytes - Relative Performance: 0 Best (in use)
LBA Format  1 : Metadata Size: 8   bytes - Data Size: 512 bytes - Relative Performance: 0 Best
LBA Format  2 : Metadata Size: 0   bytes - Data Size: 4096 bytes - Relative Performance: 0 Best
LBA Format  3 : Metadata Size: 64  bytes - Data Size: 4096 bytes - Relative Performance: 0 Best
```

igaw, 2025-01-09:
5.14.0-503.14.1.el9_5.x86_64 is likely missing the patch mentioned in https://github.com/linux-nvme/nvme-cli/wiki/FAQ#nvme-list-shows-wrong-diskblock-size

closed 2025-01-09 by igaw