linux-nvme/nvme-cli #2652: nvme list /nvme list -vv giving inconsistent data, when an nvmeof enclosure is in SLEEP state with all the drives in OFF state.

sarabjotsoit, 2025-01-09:
Put the enclosure into sleep mode with all drives in power state.

# nvme list -vv
scan controller nvme0
warning: using auto generated hostid and hostnqn
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys0/nvme0
scan controller nvme0 namespace nvme0n1
scan controller nvme26
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys0/nvme26
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys10/nvme26
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys11/nvme26
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys12/nvme26
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys13/nvme26
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys14/nvme26
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys15/nvme26
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys16/nvme26
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys17/nvme26
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys18/nvme26
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys19/nvme26
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys2/nvme26
scan controller nvme26 path nvme2c26n3
scan controller nvme26 path nvme2c26n4
scan controller nvme27
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys0/nvme27

OBSERVATION:

a. Sometimes all 24 drives list all the information and other times no drives are listed.
b. Drive information is not persistent on each nvme list /nvme list -vv fetch.

NOTE: Issue not observed with nvme-cli 1.16

sarabjotsoit, 2025-01-09:
[Uploading logs.docx…]()

igaw, 2025-01-09:
You didn't provide any logs. nvme-cli v2 relies on the kernel exporting all the necessary information on sysfs. If `nvme list` doesn't show anyting it's likely that sysfs is missing the devices. Is the kernel able to find those devices again?

igaw, 2025-03-17:
No feedback. Closing.

closed 2025-03-17 by igaw