linux-nvme/nvme-cli #2661: delete-ns on Samsung 990 PRO 1TB

palica, 2025-01-13:
Hi!

 I am unable to delete a namespace.

```
$ nvme delete-ns /dev/nvme0n1 
NVMe status: Invalid Command Opcode: A reserved coded value or an unsupported value in the command opcode field(0x1)

$ nvme list-ns /dev/nvme0
[   0]:0x1

$ nvme list
Node                  Generic               SN                   Model                                    Namespace Usage                      Format           FW Rev  
--------------------- --------------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1          /dev/ng0n1            S6Z1NU0X624009H      Samsung SSD 990 PRO 1TB                  1           2.45  MB /   1.00  TB    512   B +  0 B   4B2QJXD7
```

Can you help me? What does the error code mean?

Thank you

keithbusch, 2025-01-13:
It means your device doesn't support namespace management commands. You can verify if that is the case with command '`sudo nvme id-ctrl /dev/nvme0 -H | grep "NS Management"`'.

palica, 2025-01-13:
Yes,

```
nvme id-ctrl /dev/nvme0 -H | grep "NS Management"
  [3:3] : 0     NS Management and Attachment Not Supported
```

Can I devide the NVMe to smaller parts with other linux tools? Thanks for OT answer.

keithbusch, 2025-01-13:
You want to change how the namespace inventory looks on your device? It says it doesn't support that capability, so I would guess no user tool can get around a implementation limitation. You'd have to check with your vendor if they have some vendor specific back door, but I doubt it.

closed 2025-01-13 by palica