linux-nvme/nvme-cli #2662: nvme flush command returns "invalid argument" when given -n options with nvme-cli 2.11, 2.10.2 and 1.16

sukhi61166, 2025-01-15:
We are connected to our target over fabrics either via RoCEv2 and TCP protocol. We observed that the nvme flush command is not working as expected. It doesnt take -n 0xFFFFFFFF (even if the drive supports it) or -n 1 etc. options and returns invalid argument. If we pass individual name spaces then it works like /dev/nvme1n1. 

```
# nvme --version
    nvme version 2.10.2 (git 2.10.2)
    libnvme version 1.10 (git 1.10)
# nvme --version
nvme version 2.11 (git 2.11+)
libnvme version 1.11.1 (git 1.11.1)

Kernel : 6.8.12-u6.8.0-50.51
OS: Ubuntu 24.04 

$ sudo nvme list-ns /dev/nvme1
 [   0]:0x1
 [   1]:0x2

$sudo nvme id-ctrl /dev/nvme1 -H | grep Flush
[2:1] : 0x3      The Flush command supports NSID set to FFFFFFFFh

$ sudo nvme flush /dev/nvme1n1 -vv
opcode       : 00
flags        : 00
rsvd1        : 0000
nsid         : 00000001
cdw2         : 00000000
cdw3         : 00000000
data_len     : 00000000
metadata_len : 00000000
addr         : 0
metadata     : 0
cdw10        : 00000000
cdw11        : 00000000
cdw12        : 00000000
cdw13        : 00000000
cdw14        : 00000000
cdw15        : 00000000
timeout_ms   : 00000000
result       : 00000000
err          : 0
latency      : 171 us
NVMe Flush: success

$sudo nvme flush /dev/nvme1n2
NVMe Flush: success

$# sudo nvme flush /dev/nvme1 -n 0xFFFFFFFF -vv
opcode       : 00
flags        : 00
rsvd1        : 0000
nsid         : ffffffff
cdw2         : 00000000
cdw3         : 00000000
data_len     : 00000000
metadata_len : 00000000
addr         : 0
metadata     : 0
cdw10        : 00000000
cdw11        : 00000000
cdw12        : 00000000
cdw13        : 00000000
cdw14        : 00000000
cdw15        : 00000000
timeout_ms   : 00000000
result       : 00000000
err          : -1
latency      : 591 us
flush: Invalid argument
$# sudo nvme flush /dev/nvme1 -n 1 -vv
opcode       : 00
flags        : 00
rsvd1        : 0000
nsid         : 00000001
cdw2         : 00000000
cdw3         : 00000000
data_len     : 00000000
metadata_len : 00000000
addr         : 0
metadata     : 0
cdw10        : 00000000
cdw11        : 00000000
cdw12        : 00000000
cdw13        : 00000000
cdw14        : 00000000
cdw15        : 00000000
timeout_ms   : 00000000
result       : 00000000
err          : -1
latency      : 1131 us
flush: Invalid argument

```

keithbusch, 2025-01-15:
The kernel rejects IO commands through the character handle, /dev/nvme1, if you have more than one namespace. It's for security reasons.

sukhi61166, 2025-01-17:
@keithbusch 
  Thanks for the clarification. That is exactly the message we see in the dmesg. We can close this issue to be working as expected.

closed 2025-01-20 by igaw