linux-nvme/nvme-cli #2674: nvme disconnect segmentation fault while specify  [--nqn | -n] and [--device | -d] parameter

yansenliang, 2025-01-24:
controller  is disconnected， but cause nvme command Segmentation fault

> [root@node ~]# nvme connect -t tcp -n nqn.2024-08.io.test -a 192.168.1.100 -s 4420 -i 1
> connecting to device: nvme8
> [root@node  ~]# nvme disconnect -n nqn.2024-08.io.test  -d /dev/nvme8  
> NQN:nqn.2024-08.io.test  disconnected 1 controller(s)
> Segmentation fault (core dumped) 
> 

kernel log
> [root@node ~]#  dmesg 
> [1299497.941170] nvme nvme8: Removing ctrl: NQN "nqn.2024-08.io.test"
> [1299498.007424] nvme[454428]: segfault at 0 ip 00007f3229acd956 sp 00007ffdb22385b8 error 4 in libc.so.6[7f3229a28000+175000]
> [1299498.007437] Code: 00 00 00 00 00 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 44 00 00 f3 0f 1e fa 89 f8 09 f0 c1 e0 14 3d 00 00 00 f8 0f 87 2a 03 00 00 <62> e1 fe 28 6f 0f 62 b2 75 20 26 d1 62 f3 75 22 3f 0e 00 c5 fb 93
>

yansenliang, 2025-01-26:
> Which verison of nvme-cli do you use? Is it latest upstream release?

nvme --version
nvme version 2.9.1 (git 2.9.1)
libnvme version 1.9 (git 1.9)

yansenliang, 2025-01-26:
The function call stack is as follows：
nvmf_disconnect() --> nvmf_disconnect_nqn() --> nvme_disconnect_ctrl() --> nvme_deconfigure_ctrl()
                              --> lookup_nvme_ctrl() --> strcmp(nvme_ctrl_get_name(c), name)
                              --> nvme_disconnect_ctrl()
nvme_ctrl_t struct member variable such as ‘name’ will be free and become nullptr in nvme_deconfigure_ctrl function，so it causes  segmentation fault while compare with device name in lookup_nvme_ctrl function.

igaw, 2025-01-27:
I see, `nvmf_disconnect` offers do disconnect via NQN or the device. I don't know if the idea was that both ways are supposed to work at the same time. 

Anyway, my naive understanding is that `nvme_disconnect_ctrl` should also remove it from the tree structure. The `lookup_nvme_ctrl` call should not operate on stall data.

hreinecke, 2025-01-27:
I would make it exclusive. '-n' disconnects entire subsystems, '-d' individual controllers. Controllers will be contained within specific subsystems, so if the subsystem specified by '-n' contains the controller specified by '-d' it'll be disconnected anyways. If the controller is not contained by the subsystem the user can always issue an additional command.

hreinecke, 2025-01-27:
Should be fixed with https://github.com/linux-nvme/nvme-cli/pull/2678

igaw, 2025-01-28:
I've merged #2678, this workaround the problem that the tool crashes. Though we still have the limitation, that `nvme_disconnect_ctrl` followed by a lookup will crash. I'll open a new issue on libnvme for this.

closed 2025-01-28 by igaw