linux-nvme/nvme-cli #2677: ssd unusable after nvme format, sanitize

paraficial, 2025-01-26:
Hi, I wanted to format / sanitize a second hand and working M.2 nvme SSD using nvme-cli before using it and now the SSD is in an unusable error state.

**Initial situation**:
- SSD: INTEL SSDPEKKF256G8L
- recognized by: BIOS, Ubuntu 24.10, fdisk on device was working (two partitions: ntfs and an efi partition I think)
- using these versions:
>nvme version 2.10.2 (git 2.10.2)
libnvme version 1.10 (git 1.10)

**Step-by-step description**:

1. Check drive supported commands: `sudo nvme id-ctrl /dev/nvme0 -H | grep -E 'Format |Crypto Erase|Sanitize'`:
>   [1:1] : 0x1	Format NVM Supported
  [29:29] : 0	No-Deallocate After Sanitize bit in Sanitize command Supported
    [2:2] : 0	Overwrite Sanitize Operation Not Supported
    [1:1] : 0	Block Erase Sanitize Operation Not Supported
    [0:0] : 0	Crypto Erase Sanitize Operation Not Supported
  [3:3] : 0	Format NVM Broadcast NSID (FFFFFFFFh) Supported
  [2:2] : 0x1	Crypto Erase Supported as part of Secure Erase
  [1:1] : 0	Crypto Erase Applies to Single Namespace(s)
  [0:0] : 0	Format Applies to Single Namespace(s)
  [3:3] : 0	Controller Copy Format 3h Not Supported
  [2:2] : 0	Controller Copy Format 2h Not Supported
  [1:1] : 0	Controller Copy Format 1h Not Supported
  [0:0] : 0	Controller Copy Format 0h Not Supported

2. Trying to format drive: `sudo nvme format /dev/nvme0 -s 2 -n 1`:
> You are about to format nvme0, namespace 0x1.
WARNING: Format may irrevocably delete this device's data.
You have 10 seconds to press Ctrl-C to cancel this operation.
Use the force [--force] option to suppress this warning.
Sending format operation ...
NVMe status: Command Sequence Error: The command was aborted due to a protocol violation in a multi- command sequence(0xc)

This also happened for `-s 0`, `-s 1`, so I though of giving the sanitize command a try, even though it seemed not supported.

3. Trying to sanitize the drive: `sudo nvme sanitize /dev/nvme0 -a start-crypto-erase`
> NVMe status: Invalid Field in Command: A reserved coded value or an unsupported value in a defined field(0x2)

This also happend for `-a start-block-erase`.

4. Nothing worked. So I rebooted and now the SSD is not working anymore at all.

**Current situation: Drive not usable**
* BIOS detect the SSD.
* Ubuntu does, but gets errors, `sudo dmesg`:
> [  579.290531] nvme0n1: I/O Cmd(0x2) @ LBA 0, 8 blocks, I/O Error (sct 0x2 / sc 0x81) 
[  579.290545] critical medium error, dev nvme0n1, sector 0 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
[  579.303512] nvme0n1: I/O Cmd(0x2) @ LBA 0, 8 blocks, I/O Error (sct 0x2 / sc 0x81) 
[  579.303521] critical medium error, dev nvme0n1, sector 0 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[  579.303528] Buffer I/O error on dev nvme0n1, logical block 0, async page read

* `sudo fdisk /dev/nvme0n1`:
> Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.
fdisk: cannot open /dev/nvme0n1: Input/output error

* `lsblk`:
>NAME    MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
...
nvme0n1 259:0    0 238.5G  0 disk 

* `nvme list`, `nvme id-ctrl`, etc all can still detect the SSD like before.

**Question**
How could it happen that from using the nvme format / sanitize commands that did not succeed, the SSD seemingly broke? From the output of the tool during format / sanitize, I did not expected any modifications / changes performed to the SSD. Is it possible to repair the SSD? I tried the Intel SSD Firmware Update Tool hoping updating the firmware might help, but it did not recognize the SSD.

Thank you so much for your time!

igaw, 2025-01-27:
> 2. Trying to format drive: `sudo nvme format /dev/nvme0 -s 2 -n 1`:

So far all looks reasonable to me.

> You are about to format nvme0, namespace 0x1.
> WARNING: Format may irrevocably delete this device's data.
> You have 10 seconds to press Ctrl-C to cancel this operation.
> Use the force [--force] option [...]

So the firmware was not happy with the nvme command it got. The spec says

```
Command Sequence Error: The command was aborted due to a protocol violation in
a multi-command sequence (e.g., a violation of the Security Send and Security Receive
sequencing rules in the TCG Storage Synchronous Interface Communications protocol
(refer to TCG Storage Architecture Core Specification)).
```

Doesn't tell me anything. 

> This also happened for `-s 0`, `-s 1`, so I though of giving the sanitize command a try, even though it seemed not supported.
>
>     3. Trying to sanitize the drive: `sudo nvme sanitize /dev/nvme0 -a [...]

closed 2025-01-27 by igaw

HaroPanosyan, 2025-01-27:
Our software team at Solidigm (formerly Intel SSD Business) will try to investigate locally and provide possible solution.

vgandhi31, 2025-02-27:
Hello,

**Nvme format commands work well with Harris Harbor drive**, and it look like sanitize command is not supported in this drive. **I have confirmed this with my solidigm storage tool (SST)** and you can download SST tool from solidigm website.

https://www.solidigm.com/support-page/drivers-downloads/ka-00085.html

**Os and drive detail**

- Drive  :  INTEL SSDPEKKF256G8L  ( Harris Harbor / Intel SSD Pro 7600p Series )
- OS Version  :  24.04.1 LTS (Noble Numbat)
- nvme version 2.11 (git 2.11)
- libnvme version 1.11.1 (git 1.11.1-41-ga9f667a)
 

**Step-by-step description:**

1.  Check OS  `cat /etc/os-release`

PRETTY_NAME="Ubuntu 24.04.1 LTS"
NAME="Ubuntu"
VERSION_ID="24.04"
VERSION="24.04.1 LTS (Noble Numbat)"
VERSION_CODENAME=noble
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=noble
LOGO=ubuntu-logo

2. Check nvme-cli version `nvme version`
nvme version 2.11 (git 2.11)
libnvme version 1.11.1 (git 1.11.1-41-ga9f667a)

3. Check drive supported commands: `nvme id-ctrl /dev/nvme3 -H | grep -E 'Format |Crypto Erase|Sanitize'`

  [1:1] : 0x1   Format NVM Supported
  [29:29] : 0   No-Deallocate After Sanitize bit in Sanitize command Supported
  [2:2] : 0     Overwrite Sanitize Operation Not Supported
  [1:1] : 0     Block Erase Sanitize Operation Not Supported
  [0:0] : 0     Crypto Erase Sanitize Operation Not Supported
  [3:3] : 0     Format NVM Broadcast NSID (FFFFFFFFh) Supported
  [2:2] : 0x1   Crypto Erase Supported as part of Secure Erase
  [1:1] : 0     Crypto Erase Applies to Single Namespace(s)
  [0:0] : 0     Format Applies to Single Namespace(s)
  [3:3] : 0     Controller Copy Format 3h Not Supported
  [2:2] : 0     Controller Copy Format 2h Not Supported
  [1:1] : 0     Controller Copy Format 1h Not Supported
  [0:0] : 0     Controller Copy Format 0h Not Supported

4. Nvme format command : ` nvme format /dev/nvme0 -s 2 -n 1`                 **NVMeFormat working well**

You are about to format nvme3, namespace 0x1.
WARNING: Format may irrevocably delete this device's data.
You have 10 seconds to press Ctrl-C to cancel this operation.

Use the force [--force] option to suppress this warning.
Sending format operation ...
Success formatting namespace:1

**Sanitize command is not supported in Harris Harbor drive**

5. Trying to sanitize the drive: `nvme sanitize /dev/nvme3 -a start-crypto-erase`

NVMe status: Invalid Field in Command: A reserved coded value or an unsupported value in a defined field(0x2)

                                                         **SST ( Solidigm Storage Tool ) results**

[... 75 lines cut ...]
    [-force|-f] -- Force the operation
    [-output|-o (text|nvmxml|json)] -- Change the output format. One of "text", "nvmxml" or "json".
**Targets:**
    -ssd (Index|SerialNumber|PhysicalPath) -- Device index or serial number or physical path is required.
    -nvmeformat -- Target for issuing NVMe format.
    [-namespace (namespace id)] -- Namespace target is optional. Must specify namespace ID if being used.
**Properties:**
    [LBAFormat = (0-NumLBAFormats)] -- Value corresponds to one of the supported LBA Formats described in Identify Namespace.
    [SecureEraseSetting = (0|1|2)] -- 0 = No Secure erase. 1 = User data erase. 2 = Crypto Erase
    [ProtectionInformation = (0|1|2|3)] -- 0 = Protection Information not enabled. 1 = Type 1 enabled. 2 = Type 2 enabled. 3 = Type 3 enabled.
    [ProtectionInformationLocation = (0|1)] -- 0 = Protection Information transferred as last 8 bytes of metadata. 1 = Protection Information transferred as first
    8 bytes of metadata.
    [MetadataSettings = (0|1)] -- 0 = Metadata is transferred as part of a separate contiguous buffer. 1 = Metadata is transferred as part of an extended data LBA.
    [Timeout = (seconds)] -- Set timeout duration.

- **Trying to format the drive:** `sst start -ssd 3 -nvmeformat -namespace 1 SecureEraseSetting = 2` **NVMeFormat working well**

root@SST-Validation-Test-System:/home/remlab# **sst start -ssd 3 -nvmeformat -namespace 1 SecureEraseSetting = 2**
WARNING! You have selected to format the drive!
Proceed with the format? (Y|N): Y
Formatting...(This can take several minutes to complete)

- Intel SSD Pro 7600p Series AA000000000000000000 -

Status : NVMeFormat successful.