linux-nvme/nvme-cli #2679: nvme-cli compilation errors

martin-gpy, 2025-01-28:
It's been some time since I compiled the latest nvme-cli branch, but I see the following errors now during compilation on a SLES15 SP6 host:

```
# ninja -C build
ninja: Entering directory `build'
[9/110] Compiling C object nvme.p/nvme-print-binary.c.o
FAILED: nvme.p/nvme-print-binary.c.o
cc -Invme.p -I. -I.. -Iccan -I../ccan -I/usr/local/include -I/usr/include/json-c -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=gnu99 -O2 -g -fomit-frame-pointer -D_GNU_SOURCE -include config.h -DOPENSSL_LOAD_CONF -MD -MQ nvme.p/nvme-print-binary.c.o -MF nvme.p/nvme-print-binary.c.o.d -o nvme.p/nvme-print-binary.c.o -c ../nvme-print-binary.c
In file included from ../nvme-print-binary.c:3:0:
../nvme-print.h:91:36: warning: ‘struct nvme_mgmt_addr_list_log’ declared inside parameter list will not be visible outside of this definition or declaration
  void (*mgmt_addr_list_log)(struct nvme_mgmt_addr_list_log *ma_log);
                                    ^~~~~~~~~~~~~~~~~~~~~~~
../nvme-print.h:92:43: warning: ‘struct nvme_rotational_media_info_log’ declared inside parameter list will not be visible outside of this definition or declaration
  void (*rotational_media_info_log)(struct nvme_rotational_media_info_log *info);
                                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../nvme-print.h:334:42: warning: ‘struct nvme_mgmt_addr_list_log’ declared inside parameter list will not be visible outside of this definition or declaration
 void nvme_show_mgmt_addr_list_log(struct nvme_mgmt_addr_list_log *ma_list,
                                          ^~~~~~~~~~~~~~~~~~~~~~~
../nvme-print.h:336:49: warning: ‘struct nvme_rotational_media_info_log’ declared inside parameter list will not be visible outside of this definition or declaration
 void nvme_show_rotational_media_info_log(struct nvme_rotational_media_info_log *info,
                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../nvme-print-binary.c:310:46: warning: ‘struct nvme_mgmt_addr_list_log’ declared inside parameter list will not be visible outside of this definition or declaration
 static void binary_mgmt_addr_list_log(struct nvme_mgmt_addr_list_log *ma_list)
                                              ^~~~~~~~~~~~~~~~~~~~~~~
../nvme-print-binary.c: In function ‘binary_mgmt_addr_list_log’:
../nvme-print-binary.c:312:41: error: dereferencing pointer to incomplete type ‘struct nvme_mgmt_addr_list_log’
  d_raw((unsigned char *)ma_list, sizeof(*ma_list));
                                         ^~~~~~~~
../nvme-print-binary.c: At top level:
../nvme-print-binary.c:315:53: warning: ‘struct nvme_rotational_media_info_log’ declared inside parameter list will not be visible outside of this definition or declaration
 static void binary_rotational_media_info_log(struct nvme_rotational_media_info_log *info)
                                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../nvme-print-binary.c: In function ‘binary_rotational_media_info_log’:
../nvme-print-binary.c:317:38: error: dereferencing pointer to incomplete type ‘struct nvme_rotational_media_info_log’
  d_raw((unsigned char *)info, sizeof(*info));
                                      ^~~~~
../nvme-print-binary.c: At top level:
../nvme-print-binary.c:386:25: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
  .mgmt_addr_list_log  = binary_mgmt_addr_list_log,
                         ^~~~~~~~~~~~~~~~~~~~~~~~~
../nvme-print-binary.c:386:25: note: (near initialization for ‘binary_print_ops.mgmt_addr_list_log’)
../nvme-print-binary.c:387:31: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
  .rotational_media_info_log = binary_rotational_media_info_log,
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../nvme-print-binary.c:387:31: note: (near initialization for ‘binary_print_ops.rotational_media_info_log’)
[11/110] Compiling C object nvme.p/nvme-wrap.c.o
FAILED: nvme.p/nvme-wrap.c.o
cc -Invme.p -I. -I.. -Iccan -I../ccan -I/usr/local/include -I/usr/include/json-c -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=gnu99 -O2 -g -fomit-frame-pointer -D_GNU_SOURCE -include config.h -DOPENSSL_LOAD_CONF -MD -MQ nvme.p/nvme-wrap.c.o -MF nvme.p/nvme-wrap.c.o.d -o nvme.p/nvme-wrap.c.o -c ../nvme-wrap.c
In file included from ../nvme-wrap.c:13:0:
../nvme-wrap.h:150:16: warning: ‘struct nvme_mgmt_addr_list_log’ declared inside parameter list will not be visible outside of this definition or declaration
         struct nvme_mgmt_addr_list_log *ma_list);
                ^~~~~~~~~~~~~~~~~~~~~~~
../nvme-wrap.h:153:16: warning: ‘struct nvme_rotational_media_info_log’ declared inside parameter list will not be visible outside of this definition or declaration
         struct nvme_rotational_media_info_log *info);
                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../nvme-wrap.c:435:16: warning: ‘struct nvme_mgmt_addr_list_log’ declared inside parameter list will not be visible outside of this definition or declaration
         struct nvme_mgmt_addr_list_log *ma_list)
                ^~~~~~~~~~~~~~~~~~~~~~~
../nvme-wrap.c:434:5: error: conflicting types for ‘nvme_cli_get_log_mgmt_addr_list’
 int nvme_cli_get_log_mgmt_addr_list(struct nvme_dev *dev, __u32 len,
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ../nvme-wrap.c:13:0:
../nvme-wrap.h:149:5: note: previous declaration of ‘nvme_cli_get_log_mgmt_addr_list’ was here
 int nvme_cli_get_log_mgmt_addr_list(struct nvme_dev *dev, __u32 len,
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../nvme-wrap.c: In function ‘nvme_cli_get_log_mgmt_addr_list’:
../nvme-wrap.c:24:10: warning: implicit declaration of function ‘nvme_get_log_mgmt_addr_list’; did you mean ‘nvme_cli_get_log_mgmt_addr_list’? [-Wimplicit-function-declaration]
   __rc = nvme_ ## op(d->direct.fd, __VA_ARGS__);  \
          ^
../nvme-wrap.c:437:9: note: in expansion of macro ‘do_admin_op’
[... 668 lines cut ...]
../nvme-print.h:92:43: warning: ‘struct nvme_rotational_media_info_log’ declared inside parameter list will not be visible outside of this definition or declaration
  void (*rotational_media_info_log)(struct nvme_rotational_media_info_log *info);
                                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../nvme-print.h:334:42: warning: ‘struct nvme_mgmt_addr_list_log’ declared inside parameter list will not be visible outside of this definition or declaration
 void nvme_show_mgmt_addr_list_log(struct nvme_mgmt_addr_list_log *ma_list,
                                          ^~~~~~~~~~~~~~~~~~~~~~~
../nvme-print.h:336:49: warning: ‘struct nvme_rotational_media_info_log’ declared inside parameter list will not be visible outside of this definition or declaration
 void nvme_show_rotational_media_info_log(struct nvme_rotational_media_info_log *info,
                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[58/110] Compiling C object nvme.p/fabrics.c.o
In file included from ../fabrics.c:46:0:
../nvme-print.h:91:36: warning: ‘struct nvme_mgmt_addr_list_log’ declared inside parameter list will not be visible outside of this definition or declaration
  void (*mgmt_addr_list_log)(struct nvme_mgmt_addr_list_log *ma_log);
                                    ^~~~~~~~~~~~~~~~~~~~~~~
../nvme-print.h:92:43: warning: ‘struct nvme_rotational_media_info_log’ declared inside parameter list will not be visible outside of this definition or declaration
  void (*rotational_media_info_log)(struct nvme_rotational_media_info_log *info);
                                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../nvme-print.h:334:42: warning: ‘struct nvme_mgmt_addr_list_log’ declared inside parameter list will not be visible outside of this definition or declaration
 void nvme_show_mgmt_addr_list_log(struct nvme_mgmt_addr_list_log *ma_list,
                                          ^~~~~~~~~~~~~~~~~~~~~~~
../nvme-print.h:336:49: warning: ‘struct nvme_rotational_media_info_log’ declared inside parameter list will not be visible outside of this definition or declaration
 void nvme_show_rotational_media_info_log(struct nvme_rotational_media_info_log *info,
                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
```

martin-gpy, 2025-01-28:
Shouldn't such compile errors be caught while submitting the pull request itself? Did something change in the last month or so? What am I missing here?

martin-gpy, 2025-01-28:
Ok, tried compiling with meson instead, and that seems to work successfully:

```
# meson compile -C build
INFO: autodetecting backend as ninja
INFO: calculating backend command to run: /usr/bin/ninja -C /usr/src/packages/BUILD/jan28/nvme-cli/build
ninja: Entering directory `/usr/src/packages/BUILD/jan28/nvme-cli/build'
[110/110] Linking target nvme
```

ninja used to work earlier for me, but no longer the case now. I guess using meson is the recommended approach now as documented in the build details, so will proceed with that. 

Sorry for the noise. Will go ahead and close this issue now.

closed 2025-01-28 by martin-gpy

igaw, 2025-01-28:
`ninja -C build` should also work. It looks more like that libnvme was not in sync.