linux-nvme/nvme-cli #2710: PC801 disk can not be sanitized

berndspeiser, 2025-02-16:
I understand that the safe deletion of data on NVME disks can not be done with the classical tools like dd (and a random number generator or shred and the like). I now have a problem with an NVME SSD that I can not solve with the information I have found in this list or elesewhere.

I have a LG Gram superslim notebook (15Z90RT-G.AA78G) with a 1 TB SSD (SK Hynix PC801, HFS001TEJ9X101N) and a Phoenix BIOS (of 2023/03/22, version R32H0420 X64). This came with a Windows operating system, which I do not want to use. Rather I prefer a Linux installation, e.g. openSuSE. My typical procedure in such a case is to wipe fully the hard disk from the Windows installation using a Linux Live USB stick, and start all over with formatting and installing the preferred operating system. 
Indeed, from the live stick, I can list the hard disk as
```
> sudo nvme list
Node                  Generic               SN                   Model                                    Namespace  Usage                      Format           FW Rev
--------------------- --------------------- -------------------- ---------------------------------------- ---------- -------------------------- ---------------- --------
/dev/nvme0n1          /dev/ng0n1            AMCCN00641240CL42    HFS001TEJ9X101N                          0x1          1.02  TB /   1.02  TB    512   B +  0 B   51000A20
```
However, when I try to sanitize the disk, I get

```
> sudo nvme sanitize /dev/nvme0n1 -a2
NVMe status: Command Sequence Error: The command was aborted due to a protocol violation in a multi- command sequence(0x400c)

```
From [https://github.com/linux-nvme/nvme-cli/issues/423] I learned that the drive is probably security locked and needs to be unlocked. I found a method for unlocking that uses closing and opening the notebook lid, but this didn't work for me. I tried to get some information from the manufacturer of both the hard disk (SK Hynix) and the notebook (LG), but they were not able or willing to supply me further details. 

In the meantime, I have put a preliminary openSuSE installation (kernel version 6.11.0-lp156.10-default) on the hard disk with full LUKS encryption, which makes the original Windows installation unreadable, but of course, it did not really wipe the disk. This might not be a very important issue right now, but it will certainly come back when I want to give the notebook to someone else with a clean disk or generally at end-of live. So, I think still the problem of sanitation should be resolved.

On the preliminaty openSuSE installation, I did some experimentation, such as

```
> sudo cat /boot/config-6.11.0-lp156.10-default | grep BLK_SED
CONFIG_BLK_SED_OPAL=y
> sudo nvme sed discover /dev/nvme0n1
Error: ioctl IOC_OPAL_DISCOVERY failed
> sudo sedutil-cli --scan
Scanning for Opal compliant disks
/dev/nvme0 No  HFS001TEJ9X101N                          51000A20
No more disks present ending scan
```

Seems like this is not an OPAL device. 
I also found some information on the web that there might be some possibility in the BIOS to unlock and that the hard disk would need password protection in the first place. Indeed, I find in the "Security" tab an entry "Storage Security Features", which lists the HFS001TEJ9X101N entry and when I click, I could enter a password. I am, however, unsure what would be the consequences of this. Obviously, I would not want to be forced to enter two passwords (for the hard disk and for the LUKS encryption) on every start-up.

So, my questions are essentially, is this indeed not an OPAL disk and how can I sanitize the hard disk in this notebook before starting to attempt a final full openSuSE installation. Since I have only a preliminary installation right now, I don't care about data loss.
Would it be a sensible scenario to enter a temporary password in the BIOS, use a Linux live stick to boot and (now be able to) sanitize the disk (eventually after using the temporary password) with the approriate nmve command, return to the BIOS, clear the password and start the full Linux installation with LUKS encryption.   

I would appreciate your help in this matter, and I would cdertainly be prepared to provide additional information about the system if needed.
Thank you for considering this problem.

Bernd

keithbusch, 2025-02-16:
Sanitize is usually something I cloud vendor would do before transferring a physical disk to a third party.

igaw, 2025-02-17:
It usually comes down what hard disk vendor implements/supports in their firmware. If they don't implement the NVMe specified sanitize command OPAL commands (note, nvme-cli only supports very few of these). As @keithbusch  said, sometimes some vendors have some third party tools which use a proprietary API. This means you have to talk to your vendor and hope you get an answer.

Sorry, there is much nvme-cli can do here.

closed 2025-02-17 by igaw

berndspeiser, 2025-02-18:
Thanks for the link to the SSD blog. I probed the capabilities of the character device from my opensuse installation (kernel 6.11.0):

```
> sudo nvme id-ctrl /dev/nvme0 -H | grep -E "Format | Crypto Erase | Sanitize"
  [1:1] : 0x1   Format NVM Supported
  [29:29] : 0x1 No-Deallocate After Sanitize bit in Sanitize command Not Supported
    [2:2] : 0   Overwrite Sanitize Operation Not Supported
    [1:1] : 0x1 Block Erase Sanitize Operation Supported
    [0:0] : 0   Crypto Erase Sanitize Operation Not Supported
  [3:3] : 0     Format NVM Broadcast NSID (FFFFFFFFh) Supported
  [0:0] : 0     Format Applies to Single Namespace(s)
  [3:3] : 0     Controller Copy Format 3h Not Supported
  [2:2] : 0     Controller Copy Format 2h Not Supported
  [1:1] : 0     Controller Copy Format 1h Not Supported
  [0:0] : 0     Controller Copy Format 0h Not Supported

```

and the block device:

```
> sudo nvme id-ctrl /dev/nvme0n1 -H | grep -E "Format | Crypto Erase | Sanitize"
  [1:1] : 0x1   Format NVM Supported
  [29:29] : 0x1 No-Deallocate After Sanitize bit in Sanitize command Not Supported
    [2:2] : 0   Overwrite Sanitize Operation Not Supported
    [1:1] : 0x1 Block Erase Sanitize Operation Supported
    [0:0] : 0   Crypto Erase Sanitize Operation Not Supported
  [3:3] : 0     Format NVM Broadcast NSID (FFFFFFFFh) Supported
  [0:0] : 0     Format Applies to Single Namespace(s)
  [3:3] : 0     Controller Copy Format 3h Not Supported
  [2:2] : 0     Controller Copy Format 2h Not Supported
  [1:1] : 0     Controller Copy Format 1h Not Supported
  [0:0] : 0     Controller Copy Format 0h Not Supported

```

I realized that my USB Live stick has a different kernel (6.4.0). From the Live system I get the same response for /dev/nvme0 and /dev/nvme0n1:

```
  [1:1] : 0x1   Format NVM Supported
  [29:29] : 0x1 No-Deallocate After Sanitize bit in Sanitize command Not Supported
    [2:2] : 0   Overwrite Sanitize Operation Not Supported
    [1:1] : 0x1 Block Erase Sanitize Operation Supported
    [0:0] : 0   Crypto Erase Sanitize Operation Not Supported
  [3:3] : 0     Format NVM Broadcast NSID (FFFFFFFFh) Supported
  [2:2] : 0     Crypto Erase Not Supported as part of Secure Erase
  [1:1] : 0     Crypto Erase Applies to Single Namespace(s)
  [0:0] : 0     Format Applies to Single Namespace(s)
  [3:3] : 0     Controller Copy Format 3h Not Supported
  [2:2] : 0     Controller Copy Format 2h Not Supported
  [1:1] : 0     Controller Copy Format 1h Not Supported
  [0:0] : 0     Controller Copy Format 0h Not Supported
```

Seems that the format command would be supported for all situations. But before proceeding, I have now some questions: Why do I see the two additional lines about "crypto erase" with kernel 6.4.0?
Which one (character or block device) should I use for formatting? 
What are the implications of "Block Erase Sanitize Operation Supported" for /dev/nvme0n1 (kernel 6.11.0) and both devices (kernel 6.4.0)?

berndspeiser, 2025-02-19:
I don't know if this still works, since the thread is marked "closed". Please advise if I should open a new issue.

I have tried to perform the "nvme format" operation from the USB stick (Live system). For both nvme0 and nvme0n1 it fails with the same error code:

 ```
> sudo nvme format /dev/nvme0 -s 1 -n 0xffffffff
You are about to format nvme0, namespace 0xffffffff(ALL namespaces).
WARNING: Format may irrevocably delete this device's data.
You have 10 seconds to press Ctrl-C to cancel this operation.

Use the force [--force] option to suppress this warning.
Sending format operation ...
NVMe status: Invalid Format: The LBA Format specified is not supported(0x410a)

```
```
> sudo nvme format /dev/nvme0n1 -s 1 -n 0xffffffff
You are about to format nvme0n1, namespace 0xffffffff(ALL namespaces).
WARNING: Format may irrevocably delete this device's data.
You have 10 seconds to press Ctrl-C to cancel this operation.

Use the force [--force] option to suppress this warning.
Sending format operation ...
NVMe status: Invalid Format: The LBA Format specified is not supported(0x410a)

```

Can we draw any conclusions from that? 

By the way, here are the nvme versions of the USB stick Live System

```
> sudo nvme version
nvme version 2.8 (git 2.8+19.g47882bb)
        libnvme version 1.8 (git 1.8+26.gc96fbe8)
```

and the preliminary openSuSE installation on the disk:

```
> sudo nvme version
nvme version 2.8 (git 2.8+87.g29df38e)
libnvme version 1.8 (git 1.8+79.g69e7772)
```
which resulted in slightly different output of the nvme id-ctrl command.

igaw, 2025-02-20:
> I don't know if this still works, since the thread is marked "closed". Please advise if I should open a new issue.

I close tickets which are not bugs pretty fast. No need to open a new one.

> NVMe status: Invalid Format: The LBA Format specified is not supported(0x410a)

I suppose the disk doesn't like to the format all namespaces command. The more expensive enterprise disk usually support also namespace operations. So it might just be this consumer disk is not implementing all the fancy namespace operations. Did you try to format namespace 0? 

```shell
# nvme id-ns /dev/nvme0n1
[...]
lbaf  0 : ms:0   lbads:9  rp:0 (in use)
lbaf  1 : ms:0   lbads:12 rp:0

# nvme format /dev/nvme0 -n 1 -s 1 -l 0
```

berndspeiser, 2025-02-20:
Unfortunately, this does not help.
I get output from the id-ns command which seems identical to what you show (here I reproduce the full response for reference):

```
> sudo nvme id-ns /dev/nvme0n1
NVME Identify Namespace 1:
nsze    : 0x773bd2b0
ncap    : 0x773bd2b0
nuse    : 0x773bd2b0
nsfeat  : 0
nlbaf   : 1
flbas   : 0
mc      : 0
dpc     : 0
dps     : 0
nmic    : 0
rescap  : 0
fpi     : 0
dlfeat  : 0
nawun   : 0
nawupf  : 0
nacwu   : 0
nabsn   : 0
nabo    : 0
nabspf  : 0
noiob   : 0
nvmcap  : 0
mssrl   : 0
mcl     : 0
msrc    : 0
nulbaf  : 0
anagrpid: 0
nsattr  : 0
nvmsetid: 0
endgid  : 1
nguid   : ace42e00315ae4152ee4ac0000000001
eui64   : ace42e00315ae415
lbaf  0 : ms:0   lbads:9  rp:0 (in use)
lbaf  1 : ms:0   lbads:12 rp:0
```
but the format command results in the same error message:
```
> sudo nvme format /dev/nvme0 -n 1 -s 1 -l 0
You are about to format nvme0, namespace 0x1.
WARNING: Format may irrevocably delete this device's data.
You have 10 seconds to press Ctrl-C to cancel this operation.

Use the force [--force] option to suppress this warning.
Sending format operation ...
NVMe status: Invalid Format: The LBA Format specified is not supported(0x410a)
```

 Anything else that can be done?

igaw, 2025-02-21:
The only thing left is to check if there is something funny going on with the format command itself. Newer version of nvme-cli will print the exact commands issued.

> sudo nvme format /dev/nvme0 -n 1 -s 1 -l 0 -vvv

BTW, there was a period where the kernel didn't align the buffers on 4k bounderies for the commands. Some devices implemented their dma engine wrong and got really confused. We fixed this initially in userspace  but newer kernels have a fallback when the buffers are not aligned. In short, please test with latest versions.

berndspeiser, 2025-02-22:
Thanks for this suggestion, but unfortunately, I don't see any additional information if invoking verbosity by either using "-vvv" (as you suggested) nur with "-v" or "--verbose" (as advertised in the "nvme format help" man page. This is true for both my preliminary installation on disk and the Live version on USB stick.
I have already posted the nvme versions I have available in an earlier post. Can you see from this if these versions are maybe not recent enough?
As regards the kernel, I am somewhat limited because for the intended format operation, I will need a Live system. Can you advise as to which kernel versions would be appropriate for further testing. My present kernel versions have also already been posted earlier.
Another comment as regards vendors: as mentioned already I had contacted both the manufacturer of the SSD (SK Hynix) and that of the notebook (LG), but neither of them was able to provide any useful information. Maybe not taking the questions serious ...

igaw, 2025-02-24:
The current latest release is 2.11, obviously your using an older version since it doesn't support the verbose command line flag. 

If your distro doesn't ship a newer version, you could grap a test version if you don't want to build your own binary:

https://github.com/linux-nvme/nvme-cli?tab=readme-ov-file#testing

Sadly, many vendors do not take customers serious. The only option you have there is to return the hardware.

berndspeiser, 2025-02-24:
OK, the 2.11 version does give at least some additional information (see below).

with specified namespace:
```
> sudo ./nvme-cli-latest-x86_64.AppImage format /dev/nvme0 -s 1 -n 1 -l 0 -vvv
opcode       : 06
flags        : 00
rsvd1        : 0000
nsid         : 00000000
cdw2         : 00000000
cdw3         : 00000000
data_len     : 00001000
metadata_len : 00000000
addr         : 5587e4f53000
metadata     : 0
cdw10        : 00000001
cdw11        : 00000000
cdw12        : 00000000
cdw13        : 00000000
cdw14        : 00000000
cdw15        : 00000000
timeout_ms   : 00000000
result       : 00000000
err          : 0
latency      : 7961 us
opcode       : 06
flags        : 00
rsvd1        : 0000
nsid         : 00000001
cdw2         : 00000000
cdw3         : 00000000
data_len     : 00001000
metadata_len : 00000000
addr         : 5587e4f55000
metadata     : 0
cdw10        : 00000000
cdw11        : 00000000
cdw12        : 00000000
cdw13        : 00000000
cdw14        : 00000000
cdw15        : 00000000
timeout_ms   : 00000000
result       : 00000000
err          : 0
latency      : 250 us
You are about to format nvme0, namespace 0x1.
WARNING: Format may irrevocably delete this device's data.
You have 10 seconds to press Ctrl-C to cancel this operation.

Use the force [--force] option to suppress this warning.
Sending format operation ...
opcode       : 80
flags        : 00
rsvd1        : 0000
nsid         : 00000001
cdw2         : 00000000
cdw3         : 00000000
data_len     : 00000000
metadata_len : 00000000
addr         : 0
metadata     : 0
cdw10        : 00000200
cdw11        : 00000000
cdw12        : 00000000
cdw13        : 00000000
cdw14        : 00000000
cdw15        : 00000000
timeout_ms   : 000927c0
result       : 00000000
err          : 16650
latency      : 46756 us
        NVMe status: Invalid Format: The LBA Format specified is not supported(0x410a)
```
for all namespaces:
```
> sudo ./nvme-cli-latest-x86_64.AppImage format /dev/nvme0 -s 1 -n 0xffffffff -l 0 -vvv
opcode       : 06
flags        : 00
rsvd1        : 0000
nsid         : 00000000
cdw2         : 00000000
cdw3         : 00000000
data_len     : 00001000
metadata_len : 00000000
addr         : 558879d91000
metadata     : 0
cdw10        : 00000001
cdw11        : 00000000
cdw12        : 00000000
cdw13        : 00000000
cdw14        : 00000000
cdw15        : 00000000
timeout_ms   : 00000000
result       : 00000000
err          : 0
latency      : 7607 us
You are about to format nvme0, namespace 0xffffffff(ALL namespaces).
WARNING: Format may irrevocably delete this device's data.
You have 10 seconds to press Ctrl-C to cancel this operation.

Use the force [--force] option to suppress this warning.
Sending format operation ...
opcode       : 80
flags        : 00
rsvd1        : 0000
nsid         : ffffffff
cdw2         : 00000000
cdw3         : 00000000
data_len     : 00000000
metadata_len : 00000000
addr         : 0
metadata     : 0
cdw10        : 00000200
cdw11        : 00000000
cdw12        : 00000000
cdw13        : 00000000
cdw14        : 00000000
cdw15        : 00000000
timeout_ms   : 000927c0
result       : 00000000
err          : 16650
latency      : 55813 us
NVMe status: Invalid Format: The LBA Format specified is not supported(0x410a)
```
I also tried the sanitize command:
```
> sudo ./nvme-cli-latest-x86_64.AppImage sanitize /dev/nvme0 -a 2 -vvv
opcode       : 84
flags        : 00
rsvd1        : 0000
nsid         : 00000000
cdw2         : 00000000
cdw3         : 00000000
data_len     : 00000000
metadata_len : 00000000
addr         : 0
metadata     : 0
cdw10        : 00000002
cdw11        : 00000000
cdw12        : 00000000
cdw13        : 00000000
cdw14        : 00000000
cdw15        : 00000000
timeout_ms   : 00000000
result       : 00000000
err          : 16396
latency      : 21533 us
NVMe status: Command Sequence Error: The command was aborted due to a protocol violation in a multi- command sequence(0x400c)
```
Do you see any information pertinent to the problem?

igaw, 2025-02-24:
> Do you see any information pertinent to the problem?

The issued commands are correctly formatted (`5.1.10 Format NVM command` in the base spec 2.1)

This device is not supporting the format command. 

Just use encrypted filesystems/partition and don't rely on ssd firmwares. From a security point of view, all data at rest should be encrypted anyway. And if you overwrite the complete disk with zeros, there wont be much left to be useful on the drive.