linux-nvme/nvme-cli #2716: show-regs decoding is off

painbird7, 2025-02-18:
I am using the 2.8. Here is the show-regs output. However, it decodes fine on 2.4. Any thoughts on whether this is fixed in the latest release?
CAP.AMS associated bits are 18:17.

nvme show-regs /dev/nvme0 -H
cap     : 8f0c030140307ff
	Controller Ready With Media Support (CRWMS): Supported
	Controller Ready Independent of Media Support (CRIMS): Not Supported
	NVM Subsystem Shutdown Supported   (NSSS): Not Supported
	Controller Memory Buffer Supported (CMBS): The Controller Memory Buffer is Not Supported
	Persistent Memory Region Supported (PMRS): The Persistent Memory Region is Not Supported
	Memory Page Size Maximum         (MPSMAX): 134217728 bytes
	Memory Page Size Minimum         (MPSMIN): 4096 bytes
	Controller Power Scope              (CPS): NVM subsystem scope
	Boot Partition Support              (BPS): No
	Command Sets Supported              (CSS): NVM command set is Supported
	                                           One or more I/O Command Sets are Not Supported
	                                           I/O Command Set is Supported
	NVM Subsystem Reset Supported     (NSSRS): Yes
	Doorbell Stride                   (DSTRD): 4 bytes
	Timeout                              (TO): 10000 ms
	Arbitration Mechanism Supported     (AMS): Weighted Round Robin with Urgent Priority Class is Not supported
	Contiguous Queues Required          (CQR): Yes
	Maximum Queue Entries Supported    (MQES): 2048

igaw, 2025-02-19:
Please test with the newest version yourself. And if it doesn't work, please post the output of the working version and the output of the none working version.

ikegami-t, 2025-03-16:
Changed the output by the commit d70d4b100 but still investigating the root cause.

ikegami-t, 2025-03-16:
Very sorry the above comment was not correct so let comment again. The v2.8 show-regs AMS output is degraded by the commit 7a0647a34 changes but it was fixed by the above commit d70d4b1 then currently the newest nvme-cli version should be worked correctly.

igaw, 2025-07-25:
no feedback. closing.