linux-nvme/nvme-cli #2721: The LBA Format specified is not supported(0x410a) when creating last namespace

maxpain, 2025-02-23:
Hello. I have Samsung PM9A3 M.2 1.92TB and I want to create four 480G NVMe namespaces.

```bash
root@test:/home/maxpain# nvme create-ns /dev/nvme1 -S 480G -C 480G -flbas 1 -dps 0 -nmic 0
create-ns: Success, created nsid:1
root@test:/home/maxpain# nvme create-ns /dev/nvme1 -S 480G -C 480G -flbas 1 -dps 0 -nmic 0
create-ns: Success, created nsid:2
root@test:/home/maxpain# nvme create-ns /dev/nvme1 -S 480G -C 480G -flbas 1 -dps 0 -nmic 0
create-ns: Success, created nsid:3
root@test:/home/maxpain# nvme create-ns /dev/nvme1 -S 480G -C 480G -flbas 1 -dps 0 -nmic 0
NVMe status: Invalid Format: The LBA Format specified is not supported(0x410a)
```

I also tried `-s 117187500 -c 117187500` instead.

After this, `nvme id-ctrl /dev/nvme1` shows:
```
tnvmcap   : 1920383410176
unvmcap   : 480382377984
```

However, creating single namespace, occupying all the 4K blocks, works:
```bash
nvme create-ns /dev/nvme1 -s 468843606 -c 468843606 -flbas 1 -dps 0 -nmic 0
```

I don't understand what's how to properly and slice my NVMe disk for four even namespaces.

keithbusch, 2025-02-23:
480GiB is 515396075520 bytes, but you have 480382377984 available. I'm not sure the device's return status makes much sense, but your command shouldn't succeed.

maxpain, 2025-02-24:
> 480GiB is 515396075520 bytes, but you have 480382377984 available. I'm not sure the device's return status makes much sense, but your command shouldn't succeed.

`-S 480G -C 480G` is gigabytes, not gibibytes.

Anyway, as I said, I tried this as well:
```bash
# nvme create-ns /dev/nvme1 -s 117187500 -c 117187500 -flbas 1 -dps 0 -nmic 0
create-ns: Success, created nsid:1
# nvme create-ns /dev/nvme1 -s 117187500 -c 117187500 -flbas 1 -dps 0 -nmic 0
create-ns: Success, created nsid:2
# nvme create-ns /dev/nvme1 -s 117187500 -c 117187500 -flbas 1 -dps 0 -nmic 0
create-ns: Success, created nsid:3
# nvme create-ns /dev/nvme1 -s 117187500 -c 117187500 -flbas 1 -dps 0 -nmic 0
NVMe status: Invalid Format: The LBA Format specified is not supported(0x410a)
```

480 gigabytes = `480,000,000,000` bytes, which is `117,187,500` 4K logical blocks

`117,187,500 * 4` = `468,750,000` 4K logical blocks

tnvmcap is `1,920,383,410,176` bytes, which is `468,843,606` 4K logical blocks, so we even have `93,606` more block available.

igaw, 2025-02-24:
Stupid question, does the device support more than 3 namespaces? E.g. can you create 4 namespaces with 1G? I agree this would be weird but who knows.

maxpain, 2025-02-24:
> Stupid question, does the device support more than 3 namespaces? E.g. can you create 4 namespaces with 1G? I agree this would be weird but who knows.

Yes, it supports up to 32 namespaces.
Interestingly, I can create fourth namespace with `116,522,070` blocks, but not `116,522,071`.

```
# nvme create-ns /dev/nvme1 -s 116522071 -c 116522071 -flbas 1 -dps 0 -nmic 0
NVMe status: Invalid Format: The LBA Format specified is not supported(0x410a)
# nvme create-ns /dev/nvme1 -s 116522070 -c 116522070 -flbas 1 -dps 0 -nmic 0
create-ns: Success, created nsid:4
```

maxpain, 2025-02-24:
And it still has `unvmcap 3,109,011,456` (`759,036` blocks)

maxpain, 2025-02-24:
I even can create the fifth small namespace
[image]

maxpain, 2025-02-24:
I even can create the fifth small namespace

```bash
# nvme create-ns /dev/nvme1 -s 253013 -c 253013 -flbas 1 -dps 0 -nmic 0
NVMe status: Invalid Format: The LBA Format specified is not supported(0x410a)
# nvme create-ns /dev/nvme1 -s 253012 -c 253012 -flbas 1 -dps 0 -nmic 0
create-ns: Success, created nsid:5
# nvme attach-ns /dev/nvme1 -c 0x6 -n 5
attach-ns: Success, nsid:5
# nvme list
Node                  Generic               SN                   Model                                    Namespace  Usage                      Format           FW Rev  
--------------------- --------------------- -------------------- ---------------------------------------- ---------- -------------------------- ---------------- --------
/dev/nvme0n1          /dev/ng0n1            S7KGNJ0W930023H      Samsung SSD 990 PRO 4TB                  0x1         67.16  GB /   4.00  TB    512   B +  0 B   4B2QJXD7
/dev/nvme1n1          /dev/ng1n1            S666NN0X210193       SAMSUNG MZ1L21T9HCLS-00A07               0x1        480.00  GB / 480.00  GB      4 KiB +  0 B   GDC7502Q
/dev/nvme1n2          /dev/ng1n2            S666NN0X210193       SAMSUNG MZ1L21T9HCLS-00A07               0x2        480.00  GB / 480.00  GB      4 KiB +  0 B   GDC7502Q
/dev/nvme1n3          /dev/ng1n3            S666NN0X210193       SAMSUNG MZ1L21T9HCLS-00A07               0x3        480.00  GB / 480.00  GB      4 KiB +  0 B   GDC7502Q
/dev/nvme1n4          /dev/ng1n4            S666NN0X210193       SAMSUNG MZ1L21T9HCLS-00A07               0x5          1.04  GB /   1.04  GB      4 KiB +  0 B   GDC7502Q

maxpain, 2025-02-24:
It allows creating more small namespaces:

```bash
# nvme create-ns /dev/nvme1 -s 253012 -c 253012 -flbas 1 -dps 0 -nmic 0
create-ns: Success, created nsid:6
# nvme create-ns /dev/nvme1 -s 253012 -c 253012 -flbas 1 -dps 0 -nmic 0
create-ns: Success, created nsid:7
# nvme create-ns /dev/nvme1 -s 253012 -c 253012 -flbas 1 -dps 0 -nmic 0
NVMe status: Namespace Insufficient Capacity: Creating the namespace requires more free space than is currently available(0x2115)
# nvme attach-ns /dev/nvme1 -c 0x6 -n 6
attach-ns: Success, nsid:6
# nvme attach-ns /dev/nvme1 -c 0x6 -n 7
attach-ns: Success, nsid:7
# nvme list
Node                  Generic               SN                   Model                                    Namespace  Usage                      Format           FW Rev  
--------------------- --------------------- -------------------- ---------------------------------------- ---------- -------------------------- ---------------- --------
/dev/nvme0n1          /dev/ng0n1            S7KGNJ0W930023H      Samsung SSD 990 PRO 4TB                  0x1         67.16  GB /   4.00  TB    512   B +  0 B   4B2QJXD7
/dev/nvme1n1          /dev/ng1n1            S666NN0X210193       SAMSUNG MZ1L21T9HCLS-00A07               0x1        480.00  GB / 480.00  GB      4 KiB +  0 B   GDC7502Q
/dev/nvme1n2          /dev/ng1n2            S666NN0X210193       SAMSUNG MZ1L21T9HCLS-00A07               0x2        480.00  GB / 480.00  GB      4 KiB +  0 B   GDC7502Q
/dev/nvme1n3          /dev/ng1n3            S666NN0X210193       SAMSUNG MZ1L21T9HCLS-00A07               0x3        480.00  GB / 480.00  GB      4 KiB +  0 B   GDC7502Q
/dev/nvme1n4          /dev/ng1n4            S666NN0X210193       SAMSUNG MZ1L21T9HCLS-00A07               0x4        477.27  GB / 477.27  GB      4 KiB +  0 B   GDC7502Q
/dev/nvme1n5          /dev/ng1n5            S666NN0X210193       SAMSUNG MZ1L21T9HCLS-00A07               0x5          1.04  GB /   1.04  GB      4 KiB +  0 B   GDC7502Q
/dev/nvme1n6          /dev/ng1n6            S666NN0X210193       SAMSUNG MZ1L21T9HCLS-00A07               0x6          1.04  GB /   1.04  GB      4 KiB +  0 B   GDC7502Q
/dev/nvme1n7          /dev/ng1n7            S666NN0X210193       SAMSUNG MZ1L21T9HCLS-00A07               0x7          1.04  GB /   1.04  GB      4 KiB +  0 B   GDC7502Q
```

igaw, 2025-02-24:
Ah wait, I just saw you started of with the `-S`, `-C` variants. Too bad.

maxpain, 2025-02-24:
> that's why we introduced the `-S`, `-C` variants which should take care of it. Try to use these.

That's what I tried first (see my first message). It didn't work.

maxpain, 2025-02-24:
```bash
# nvme id-ctrl -H /dev/nvme1 | grep Granularity
  [7:7] : 0x1	Namespace Granularity Supported
```

maxpain, 2025-02-24:
```bash
# nvme create-ns /dev/nvme1 -s 117187500 -c 117187500 -flbas 1 -dps 0 -nmic 0 -vvv
opcode       : 06
flags        : 00
rsvd1        : 0000
nsid         : 00000000
cdw2         : 00000000
cdw3         : 00000000
data_len     : 00001000
metadata_len : 00000000
addr         : 5b7585efb000
metadata     : 0
cdw10        : 00000001
cdw11        : 00000000
cdw12        : 00000000
cdw13        : 00000000
cdw14        : 00000000
cdw15        : 00000000
timeout_ms   : 00000000
result       : 00000000
err          : 0
latency      : 204 us
opcode       : 06
flags        : 00
rsvd1        : 0000
nsid         : 00000000
cdw2         : 00000000
cdw3         : 00000000
data_len     : 00001000
metadata_len : 00000000
addr         : 5b7585efd000
metadata     : 0
cdw10        : 00000016
cdw11        : 00000000
cdw12        : 00000000
cdw13        : 00000000
cdw14        : 00000000
cdw15        : 00000000
timeout_ms   : 00000000
result       : 00000000
err          : 0
latency      : 98 us
opcode       : 0d
flags        : 00
rsvd1        : 0000
nsid         : 00000000
cdw2         : 00000000
cdw3         : 00000000
data_len     : 00001000
metadata_len : 00000000
addr         : 5b7585eff000
metadata     : 0
cdw10        : 00000000
cdw11        : 00000000
cdw12        : 00000000
cdw13        : 00000000
cdw14        : 00000000
cdw15        : 00000000
timeout_ms   : 0001d4c0
result       : 00000004
err          : 16650
latency      : 770 us
NVMe status: Invalid Format: The LBA Format specified is not supported(0x410a)
```

maxpain, 2025-02-24:
```bash
# nvme id-ns-granularity /dev/nvme1
Identify Namespace Granularity List:
   ATTR        : Namespace Granularity Attributes: 0
   NUMD        : Number of Descriptors           : 0

     Entry[ 0] :
................
     NSG       : Namespace Size Granularity     : 0x200000000
     NCG       : Namespace Capacity Granularity : 0x200000000
```

maxpain, 2025-02-24:
Does it mean that my NVMe have granularity of `8,589,934,592` bytes (8 GiB, gibibytes)?

maxpain, 2025-02-24:
Also interesting:

This works:
```bash
nvme create-ns /dev/nvme1 -s 176160768 -c 176160768 -flbas 1 -dps 0 -nmic 0 # 84 * 8GiB = 672 GiB
nvme create-ns /dev/nvme1 -s 176160768 -c 176160768 -flbas 1 -dps 0 -nmic 0 # 84 * 8GiB = 672 GiB
nvme create-ns /dev/nvme1 -s 116522070 -c 116522070 -flbas 1 -dps 0 -nmic 0 # All remaining blocks, not a multiple of 8 GiB
```

This doesn't:
```bash
nvme create-ns /dev/nvme1 -s 116522070 -c 116522070 -flbas 1 -dps 0 -nmic 0 # All remaining blocks, not a multiple of 8 GiB
nvme create-ns /dev/nvme1 -s 176160768 -c 176160768 -flbas 1 -dps 0 -nmic 0 # 84 * 8GiB = 672 GiB
nvme create-ns /dev/nvme1 -s 176160768 -c 176160768 -flbas 1 -dps 0 -nmic 0 # 84 * 8GiB = 672 GiB
```

igaw, 2025-02-24:
> Does it mean that my NVMe have granularity of `8,589,934,592` bytes (8 GiB/Gi**bi**Bytes)?

Yes.

> This doesn't:

I suppose because of the granularity requirements, start address needs to start at a `NSG` but the firmware doesn't require that the end address is on a `NSG`.

igaw, 2025-02-24:
The granularity logic in `create_ns` seems to depend on correctly identifying the `flabs`. Could you post the output from `nvme id-ns /dev/nvme0 -n 0xff`?

maxpain, 2025-02-24:
```bash
root@test:/home/maxpain# nvme create-ns /dev/nvme1 -S 480G -C 480G -flbas 1 -dps 0 -nmic 0 -vvv
opcode       : 06
flags        : 00
rsvd1        : 0000
nsid         : 00000000
cdw2         : 00000000
cdw3         : 00000000
data_len     : 00001000
metadata_len : 00000000
addr         : 645065b35000
metadata     : 0
cdw10        : 00000001
cdw11        : 00000000
cdw12        : 00000000
cdw13        : 00000000
cdw14        : 00000000
cdw15        : 00000000
timeout_ms   : 00000000
result       : 00000000
err          : 0
latency      : 756 us
opcode       : 06
flags        : 00
rsvd1        : 0000
nsid         : 00000000
cdw2         : 00000000
cdw3         : 00000000
data_len     : 00001000
metadata_len : 00000000
addr         : 645065b37000
metadata     : 0
cdw10        : 00000016
cdw11        : 00000000
cdw12        : 00000000
cdw13        : 00000000
cdw14        : 00000000
cdw15        : 00000000
timeout_ms   : 00000000
result       : 00000000
err          : 0
latency      : 136 us
opcode       : 06
flags        : 00
rsvd1        : 0000
nsid         : 00000000
cdw2         : 00000000
cdw3         : 00000000
data_len     : 00001000
metadata_len : 00000000
addr         : 645065b39000
metadata     : 0
cdw10        : 00000001
cdw11        : 00000000
cdw12        : 00000000
cdw13        : 00000000
cdw14        : 00000000
cdw15        : 00000000
timeout_ms   : 00000000
result       : 00000000
err          : 0
latency      : 158 us
opcode       : 06
flags        : 00
rsvd1        : 0000
nsid         : ffffffff
cdw2         : 00000000
cdw3         : 00000000
data_len     : 00001000
metadata_len : 00000000
[... 194 lines cut ...]
result       : 00000000
err          : 0
latency      : 118 us
opcode       : 0d
flags        : 00
rsvd1        : 0000
nsid         : 00000000
cdw2         : 00000000
cdw3         : 00000000
data_len     : 00001000
metadata_len : 00000000
addr         : 604680737000
metadata     : 0
cdw10        : 00000000
cdw11        : 00000000
cdw12        : 00000000
cdw13        : 00000000
cdw14        : 00000000
cdw15        : 00000000
timeout_ms   : 0001d4c0
result       : 00000004
err          : 16650
latency      : 767 us
NVMe status: Invalid Format: The LBA Format specified is not supported(0x410a)
```

igaw, 2025-02-24:
```shell
     Entry[ 0] :
................
     NSG       : Namespace Size Granularity     : 0x200000000
     NCG       : Namespace Capacity Granularity : 0x200000000
```

```c
	__u32 align_nsze = 1 << 20; /* Default 1 MiB */
	__u32 align_ncap = align_nsze;

			if (desc->nszegran && desc->nszegran < align_nsze)
				align_nsze = desc->nszegran;
			if (desc->ncapgran && desc->ncapgran < align_ncap)
				align_ncap = desc->ncapgran;
```

Ah, so we only use `nszegran` and `ncapgran` when it's below the magic value of `1 MiB`.  This is obviously not working in this case.

maxpain, 2025-02-24:
I also noticed that after creating a namespace, the `Usage` (`nuse`) field shows 100% utilization.

[image]

Is it a bug?

igaw, 2025-02-24:
3d1243e1bf22 ("create-ns: align the namespaces to 1Mib boundaries when using SI suffixes") 

@maurizio-lombardi I think we need to change this logic a slightly, do you agree with my reasoning?

maxpain, 2025-02-24:
> > I also noticed that after creating a namespace, the `Usage` (`nuse`) field shows 100% utilization.
> > Is it a bug?
> 
> No, this is not filesystem utilization.

When I received this NVMe disk from the factory, it had zero utilization before I used `nvme-cli`. That's the question. Just trying to figure out how it works.

[image]

maurizio-lombardi, 2025-02-24:
@igaw Maybe this should be modified to be like this:

`if (desc->nszegran)
       align_nsze = desc->nszegran;
  if (desc->ncapgran)
       align_ncap = desc->ncapgran;
`

But this would mean that, in this specific case, doing "nvme create-ns /dev/nvme1 -S 1G -C 1G" would create a 8Gb namespace...

igaw, 2025-02-24:
Yep this what I had in mind. Isn't this what the drive asked for, no? At least the experiments by @maxpain show the drive wants 8G alignment. We could still make it possible to overwrite from the command line.

igaw, 2025-02-24:
From the amount of users asking how to use nvme-cli, I am inclined to add a big fat warnings everywhere :)

PR #2722 (2025-02-26): nvme: enforce ns granularity on ns create
When the device supports the namespace granularity feature, update the alignment requirements accordingly. Because this is likely to change the capacity of the namespace also issue a info what is going to happen.

Fixes: #2721
closed 2025-02-26 by igaw