linux-nvme/nvme-cli #2734: set-feature 0xc6 (198) relaxed ordering "Unknown"

stannous, 2025-02-28:
Running Ubuntu noble  6.8.0-53-generic with nvme-cli 2.8-1ubuntu0.1 amd64 and started getting an "Unknown" for setting feature relaxed ordering (198 == 0xc6).  Is this a known "Unknown" issue?   (apologies to Donald Rumsfleld ;-)

$ sudo nvme list
Node                  Generic               SN                   Model                                    Namespace  Usage                      Format           FW Rev  
--------------------- --------------------- -------------------- ---------------------------------------- ---------- -------------------------- ---------------- --------
/dev/nvme0n1          /dev/ng0n1            S4YPNE0MC01678       SAMSUNG MZWLJ3T8HBLS-00007               0x1          3.84  TB /   3.84  TB    512   B +  0 B   EPK9GB5Q
/dev/nvme1n1          /dev/ng1n1            S4YPNE0N702988       SAMSUNG MZWLJ3T8HBLS-00007               0x1          3.84  TB /   3.84  TB    512   B +  0 B   EPK9GB5Q

$ sudo nvme set-feature -v  /dev/nvme0 -f 198 -v 1 -s
opcode       : 09
flags        : 00
rsvd1        : 0000
nsid         : ffffffff
cdw2         : 00000000
cdw3         : 00000000
data_len     : 00000000
metadata_len : 00000000
addr         : 0
metadata     : 0
cdw10        : 800000c6
cdw11        : 00000000
cdw12        : 00000000
cdw13        : 00000000
cdw14        : 00000000
cdw15        : 00000000
timeout_ms   : 00000000
result       : 00000000
err          : 0
latency      : 148 us
set-feature:0xc6 (Unknown), value:00000000, cdw12:00000000, save:0x1

keithbusch, 2025-02-28:
unfortunately the "-v"  doesn't mean what it used to, you need to use the long options name, `--value=1`. As far as value "c6" being unknown, it's a vendor specific value that the generic tool doesn't necessarily know how to decode.

stannous, 2025-03-01:
Yep, thanks for the tip.  -V 1 works like a charm.  I'll close this bug.

closed 2025-03-01 by stannous

igaw, 2025-03-04:
Yep, sorry for the renaming the options.