linux-nvme/nvme-cli #2736: nvme cli not working when installed nvme-cli ;

Nitheesh1609, 2025-03-05:
Tried to install nvme-cli latest head on nvme-cli and libnvme

1. nvme --version was 2.10 before
2. git clone
3. meson setup --force-fallback-for=libnvme .build -> SUCCESS
4. meson compile -C .build ->compile success with warnings on "%s"
5.  meson install -C .build -> install complete without issue
6. nvme --version
nvme: /lib64/libnvme.so.1: version `LIBNVME_1.11' not found (required by nvme)
nvme: /lib64/libnvme.so.1: version `LIBNVME_1.12' not found (required by nvme)

Nitheesh1609, 2025-03-05:
export LD_LIBRARY_PATH=$LD_LIBRARY@Path:/usr/local/lib64/
fixed the isssue.

closed 2025-03-05 by Nitheesh1609

igaw, 2025-03-05:
I thought meson takes care when installing the program with fallback libraries with `rpath`. Though it looks like you have already a libnvme installed and `ld` picks up the installed version. In this case the `LD_LIBRARY_PATH` seems like the correct fix. Though I am really no expert on these topics.