linux-nvme/nvme-cli #2737: Create namespace command not parser endurance group id correctly

yalanliufio, 2025-03-05:
I am using nvme version 2.9.1. When I try to issue command like this:

nvme delete-ns /dev/nvme0 -n -1
delete-ns: Success, deleted nsid:-1
nvme create-ns /dev/nvme0 -s 1000000 -c 1000000 -f 0 -e 2 -i 0
create-ns: Success, created nsid:1

nvme delete-ns /dev/nvme0 -n -1
delete-ns: Success, deleted nsid:-1
nvme create-ns /dev/nvme0 -s 1000000 -c 1000000 -f 0 -e 2
NVMe status: Invalid Field in Command: A reserved coded value or an unsupported value in a defined field(0x2)

I checked nvme cli code, seems like the above two create-ns commands doing the same thing as when -i option is not provided, it is defaulting to value 0. In create-ns command, -i option is used set the NVM Set Identifier (NVMSETID).  -e option is used to set endurance group id. Don't understand why above two commands have different results.

The drive I am using is only support 1 endurance group, I expect the above two commands should fail.  Check with our firmware engineers, they said no matter what value of endurance group id is given, they always receive the value of 0.

igaw, 2025-03-06:
About `-i`: if you don't specicify it is set to 'all namespaces' (-1). This works for delete operation but not for the create namespace command. There you need to specify in a valid id for the new namespace. 

About `-e`: The command line value is used to set the `endgid` member in the 'Namespace Management – Host Software Specified Fields' data structure. From a quick check, the offsets seems to be correct. 

[image]

```
struct nvme_ns_mgmt_host_sw_specified {
        __le64                     nsze;                 /*     0     8 */
        __le64                     ncap;                 /*     8     8 */
        __u8                       rsvd16[10];           /*    16    10 */
        __u8                       flbas;                /*    26     1 */
        __u8                       rsvd27[2];            /*    27     2 */
        __u8                       dps;                  /*    29     1 */
        __u8                       nmic;                 /*    30     1 */
        __u8                       rsvd31[61];           /*    31    61 */
        /* --- cacheline 1 boundary (64 bytes) was 28 bytes ago --- */
        __le32                     anagrpid;             /*    92     4 */
        __u8                       rsvd96[4];            /*    96     4 */
        __le16                     nvmsetid;             /*   100     2 */
        __le16                     endgid;               /*   102     2 */
        __u8                       rsvd104[280];         /*   104   280 */
        /* --- cacheline 6 boundary (384 bytes) --- */
        __le64                     lbstm;                /*   384     8 */
        __le16                     nphndls;              /*   392     2 */
        __u8                       rsvd394[105];         /*   394   105 */
        /* --- cacheline 7 boundary (448 bytes) was 51 bytes ago --- */
        union {
                __u8               rsvd499[13];          /*   499    13 */
                struct {
                        __u8       znsco;                /*   499     1 */
                        __le32     rar;                  /*   500     4 */
                        __le32     ror;                  /*   504     4 */
                        __le32     rnumzrwa;             /*   508     4 */
                } __attribute__((__packed__)) zns;       /*   499    13 */
        };                                               /*   499    13 */
        /* --- cacheline 8 boundary (512 bytes) --- */
        __le16                     phndl[128];           /*   512   256 */
        /* --- cacheline 12 boundary (768 bytes) --- */
        __u8                       rsvd768[3328];        /*   768  3328 */

        /* size: 4096, cachelines: 64, members: 19 */
} __attribute__((__packed__));
```

Can you eleaborate what you expecting and which value is not set?

yalanliufio, 2025-03-06:
Thanks for the explanation of -i option, understand in the create-ns command, the -i is the NVMSetID
  [  --nvmset-id=<NUM>, -i <NUM> ]      --- NVM Set Identifier (NVMSETID)

1. our drive only support 1 endurance group, this can be checked by id-ctrl command:
nvme id-ctrl /dev/nvme2 | grep endgidmax
endgidmax : 1

2. when issue below create-ns command, I expect it fail, but it passed
nvme delete-ns /dev/nvme2 -n -1
delete-ns: Success, deleted nsid:-1
nvme create-ns /dev/nvme2 -s 1000000 -c 1000000 -f 0 -e 2 -i 0
create-ns: Success, created nsid:1

3. however, if I use the admin-passthru with the endgid=2, command did fail as expected:
Here is the passthru command:
[root@sm-121h-517 test-scripts]# nvme admin-passthru /dev/nvme2 -O 0x0d -l 4096 -n 0 -i logs/ns_data.bin -w
NVMe status: Invalid Field in Command: A reserved coded value or an unsupported value in a defined field(0x2)

I generate the ns-file by this:
                    nsze = 1024*1024*1024
                    ncap = 1024*1024*1024
                    endgid = 2
                    ns_data = bytearray(4096)
                    ns_data[0:7] = nsze.to_bytes(8, "little")
                    ns_data[8:15] = ncap.to_bytes(8, "little")
                    ns_data[102:103] = endgid.to_bytes(2, "little")

                    ns_file = os.path.join(log_utils.get_test_log_dir(), "ns_data.bin")
                    with open(ns_file, "wb") as f:
                        f.write(ns_data)

yalanliufio, 2025-03-10:
Hello igaw,

the ns_data[102:104] = endgid.to_bytes(2, "little") is correct because the slice operation exclusive of the end index.
For example, when using [102:103], the slice will only include the element at index 102 but not at index 103.

The library I am using is 1.9
nvme --version
nvme version 2.9.1 (git 2.9.1)
libnvme version 1.9 (git 1.9)

I will try the latest release 2.11 and capture a trace to check again, thanks!

igaw, 2025-03-11:
D'oh I forgot how the slices work in Python. Anyway, I've reproduces it locally. The arg parser is not working, though I don't know why yet.

igaw, 2025-03-11:
Thanks for the report. I've merged a fix.

closed 2025-03-11 by igaw

yalanliufio, 2025-03-11:
Great, thank you so much for the fix!

yalanliufio, 2025-04-04:
Confirm this issue is fixed in release 2.12, thank you!

PR #2741 (2025-03-11): nvme: use correct parser type for nvmsetid and endgid option
nvmsetid and endgid are both shorts, thus the OPT_SHRT parser type needs to be used. Otherwise the result is written to the wrong memory location.

Fixes: 972effdf884e ("nvme.c: create-ns arguments are added")
Fixes: #2737