linux-nvme/nvme-cli #2743: Error when cross compiling with musl (OpenWrt build system)

graysky2, 2025-03-15:
When building this package with the OpenWrt build system, the following error breaks compilation:
```
../nvme-print.c: In function 'is_temperature_fahrenheit':
../nvme-print.c:805:19: error: 'LC_MEASUREMENT' undeclared (first use in this function)
  805 |         setlocale(LC_MEASUREMENT, "");
      |                   ^~~~~~~~~~~~~~
../nvme-print.c:805:19: note: each undeclared identifier is reported only once for each function it appears in
```

I think it due to musl not supporting locale detection like glibc does. I came up with [this hacky work-around](https://github.com/openwrt/packages/pull/26142/files#diff-33ed2a75bf22b88e04f422661385300117762523c9d5d190293e163a49989d0e) but wanted to bring this up. Maybe you have a more elegant solution.

ikegami-t, 2025-03-16:
Looks the issue fixed by the commit 9444ce03a. Could you please confirm the build with the commit changes?

graysky2, 2025-03-16:
Hi, I applied that commit to the tagged release 2.11 but it did not fix the build. See below.
If I add the following, the build works as expected.
```
--- a/plugins/netapp/netapp-nvme.c
+++ b/plugins/netapp/netapp-nvme.c
@@ -22,6 +22,7 @@
 #include <unistd.h>
 #include <errno.h>
 #include <string.h>
+#include <libgen.h>
 
 #include "common.h"
 #include "nvme.h"
```

Example of build failing without the above patch:
```
[27/105] Compiling C object nvme.p/plugins_netapp_netapp-nvme.c.o
FAILED: nvme.p/plugins_netapp_netapp-nvme.c.o 
x86_64-openwrt-linux-musl-gcc -Invme.p -I. -I.. -Iccan -I../ccan -I/scratch/union/staging_dir/target-x86_64_musl/usr/include -I/scratch/union/staging_dir/target-x86_64_musl/usr/include/json-c -I/scratch/union/staging_dir/toolchain-x86_64_gcc-14.2.0_musl/usr/include -I/scratch/union/staging_dir/toolchain-x86_64_gcc-14.2.0_musl/include -I/scratch/union/staging_dir/toolchain-x86_64_gcc-14.2.0_musl/include/fortify -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=gnu99 -fomit-frame-pointer -D_GNU_SOURCE -include config.h -march=x86-64-v3 -O3 -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/scratch/union/build_dir/target-x86_64_musl/nvme-cli-2.11=nvme-cli-2.11 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -MD -MQ nvme.p/plugins_netapp_netapp-nvme.c.o -MF nvme.p/plugins_netapp_netapp-nvme.c.o.d -o nvme.p/plugins_netapp_netapp-nvme.c.o -c ../plugins/netapp/netapp-nvme.c
../plugins/netapp/netapp-nvme.c: In function 'netapp_smdevices_print_regular':
../plugins/netapp/netapp-nvme.c:284:49: error: implicit declaration of function 'basename' [-Wimplicit-function-declaration]
  284 |                 if (devname && !strcmp(devname, basename(devices[i].dev))) {
      |                                                 ^~~~~~~~
../plugins/netapp/netapp-nvme.c:284:49: error: passing argument 2 of 'strcmp' makes pointer from integer without a cast [-Wint-conversion]
  284 |                 if (devname && !strcmp(devname, basename(devices[i].dev))) {
      |                                                 ^~~~~~~~~~~~~~~~~~~~~~~~
      |                                                 |
      |                                                 int
In file included from /scratch/union/staging_dir/toolchain-x86_64_gcc-14.2.0_musl/include/fortify/string.h:22,
                 from ../plugins/netapp/netapp-nvme.c:24:
/scratch/union/staging_dir/toolchain-x86_64_gcc-14.2.0_musl/include/string.h:39:27: note: expected 'const char *' but argument is of type 'int'
   39 | int strcmp (const char *, const char *);
      |                           ^~~~~~~~~~~~
../plugins/netapp/netapp-nvme.c: In function 'netapp_smdevices_print_json':
../plugins/netapp/netapp-nvme.c:352:49: error: passing argument 2 of 'strcmp' makes pointer from integer without a cast [-Wint-conversion]
  352 |                 if (devname && !strcmp(devname, basename(devices[i].dev))) {
      |                                                 ^~~~~~~~~~~~~~~~~~~~~~~~
      |                                                 |
      |                                                 int
/scratch/union/staging_dir/toolchain-x86_64_gcc-14.2.0_musl/include/string.h:39:27: note: expected 'const char *' but argument is of type 'int'
   39 | int strcmp (const char *, const char *);
      |                           ^~~~~~~~~~~~
../plugins/netapp/netapp-nvme.c: In function 'netapp_ontapdevices_print_regular':
../plugins/netapp/netapp-nvme.c:442:49: error: passing argument 2 of 'strcmp' makes pointer from integer without a cast [-Wint-conversion]
  442 |                 if (devname && !strcmp(devname, basename(devices[i].dev))) {
      |                                                 ^~~~~~~~~~~~~~~~~~~~~~~~
      |                                                 |
      |                                                 int
/scratch/union/staging_dir/toolchain-x86_64_gcc-14.2.0_musl/include/string.h:39:27: note: expected 'const char *' but argument is of type 'int'
   39 | int strcmp (const char *, const char *);
      |                           ^~~~~~~~~~~~
../plugins/netapp/netapp-nvme.c: In function 'netapp_ontapdevices_print_json':
../plugins/netapp/netapp-nvme.c:483:49: error: passing argument 2 of 'strcmp' makes pointer from integer without a cast [-Wint-conversion]
  483 |                 if (devname && !strcmp(devname, basename(devices[i].dev))) {
      |                                                 ^~~~~~~~~~~~~~~~~~~~~~~~
      |                                                 |
      |                                                 int
/scratch/union/staging_dir/toolchain-x86_64_gcc-14.2.0_musl/include/string.h:39:27: note: expected 'const char *' but argument is of type 'int'
   39 | int strcmp (const char *, const char *);
      |                           ^~~~~~~~~~~~
../plugins/netapp/netapp-nvme.c: In function 'netapp_smdevices':
../plugins/netapp/netapp-nvme.c:716:25: error: assignment to 'char *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
  716 |                 devname = basename(argv[optind++]);
      |                         ^
../plugins/netapp/netapp-nvme.c: In function 'netapp_ontapdevices':
../plugins/netapp/netapp-nvme.c:801:25: error: assignment to 'char *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
  801 |                 devname = basename(argv[optind++]);
      |                         ^
[60/105] Compiling C object nvme.p/nvme.c.o
ninja: build stopped: subcommand failed.
```

graysky2, 2025-03-16:
Closing this issue.

closed 2025-03-16 by graysky2