linux-nvme/nvme-cli #2749: JSON output change for 'list' in 2.11+ breaks automation

markozagar, 2025-03-26:
It seems the output change in Commit 929f461 significantly alters the output of `nvme list -o json`. As I understand it, the only JSON option right now is the output of (what was previously) `nvme list -o json -v`, with no option to get the original output back.

Our automation is using something like `nvme list --output-format=json | jq -r '.Devices[]|.DevicePath'`, which no longer produces a list of device paths of NVME drives on the system, as of 2.11 (slow package update process means we didn't see this earlier).

The alternative right now is to use `nvme list | awk '/^\/dev {print $1}'`, which is parsing human-readable output in favour of JSON, which just feels wrong.

Interestingly, our system does not populate `Paths` for devices, so the cleanest new option of listing devices would be something like `nvme list -o json | jq -r '.Devices[] | .Subsystems[] | .Controllers[] | .Namespaces[] | "/dev/" + .NameSpace'`. I'm also worried that some strange configuration or distribution might not use `/dev/nvmeXn1` paths, in which case the 'detection script' will have to get more complex when it happens. I have not seen this in the wild yet, so maybe this worry is unfounded.

Is this the intended path forward, or is this potentially an oversight? What is your advice on a (relatively stable) way to retrieve of a list of NVME drives in scripts?

Output pre-2.11:

```
{
  "Devices":[
    {
      "NameSpace":1,
      "DevicePath":"/dev/nvme0n1",
      "GenericPath":"/dev/ng0n1",
      "Firmware":"5B2QGXA7",
      "ModelNumber":"Samsung SSD 980 PRO 2TB",
      "SerialNumber":"S6B0NL0T924753T",
      "UsedBytes":185560145920,
      "MaximumLBA":3907029168,
      "PhysicalSize":2000398934016,
      "SectorSize":512
    },
    ...
```

Output since 2.11:

```
{
  "Devices":[
    {
      "HostNQN":"nqn.2014-08.org.nvmexpress:uuid:4c4c4544-0037-4410-8032-b1c04f505433",
      "HostID":"579f640c-2adb-44a2-a43c-006ec4c2bba6",
      "Subsystems":[
        {
          "Subsystem":"nvme-subsys0",
          "SubsystemNQN":"nqn.1994-11.com.samsung:nvme:980PRO:M.2:S6B0NL0T924753T",
          "Controllers":[
            {
              "Controller":"nvme0",
              "Cntlid":"6",
              "SerialNumber":"S6B0NL0T924753T",
              "ModelNumber":"Samsung SSD 980 PRO 2TB",
              "Firmware":"5B2QGXA7",
              "Transport":"pcie",
              "Address":"0000:01:00.0",
              "Slot":"",
              "Namespaces":[
                {
                  "NameSpace":"nvme0n1",
                  "Generic":"ng0n1",
                  "NSID":1,
                  "UsedBytes":185566969856,
                  "MaximumLBA":3907029168,
                  "PhysicalSize":2000398934016,
                  "SectorSize":512
                }
              ],
              "Paths":[]
            }
          ],
          "Namespaces":[]
        },
        ...
```
```

ikegami-t, 2025-03-29:
@martin-gpy Just FYI.

martin-gpy, 2025-04-02:
Sigh. The JSON verbose output is usually a superset of the JSON regular output, but that does not seem to be the case here for ```nvme list```. As suggested by @igaw , something similar to [6a73c83](https://github.com/linux-nvme/nvme-cli/commit/6a73c83b3c07c8a03b55cee86ff7457722037c79) would be ideal.

PR #2762 (2025-04-03): Revert "nvme-print-json: display only verbose output"
This reverts commit 929f4619a9aa9c900247384480fe60ff96931b07.

The output format chagned and breaks existing automation.

Fixes: #2749
closed 2025-04-03 by igaw