linux-nvme/nvme-cli #2759: Returning zeroes

kleshas, 2025-03-30:
arch, sway. fully updated

sudo nvme smart-log /dev/nvme0n1 returns zero for all the fields (except -273C for termperature).

Tried downgrading nvme-cli to last version and the one before that.  Reinstalled.  Used the Node and Generic fields for the devices as listed in nvme list.  Nothing helping.   No errors in journalctl.  This worked for years up to a few days ago.

igaw, 2025-04-01:
> This worked for years up to a few days ago.

What has changed? Did you update nvme-cli, kernel or firmware?

> sudo nvme smart-log /dev/nvme0n1 returns zero for all the fields (except -273C for termperature).

The temp is mapped from Kelvin to Celsius, so the value is also 0. As far I remember (and a quick glance into the commit history) we didn't touch this area for a while. 

Could you past the output of

```shell
nvme get-log -i 0x2 -l 0x100 -vv /dev/nvme0
```

kleshas, 2025-04-01:
```
sudo nvme get-log -i 0x2 -l 0x100 -vv /dev/nvme0
Device:nvme0 log-id:2 namespace-id:0xffffffff
       0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
0000: 00 38 01 64 0a 01 00 00 00 00 00 00 00 00 00 00 ".8.d............"
0010: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0020: 3a bc 28 08 00 00 00 00 00 00 00 00 00 00 00 00 ":.(............."
0030: c4 fc 4c 03 00 00 00 00 00 00 00 00 00 00 00 00 "..L............."
0040: 2a 46 ac 1c 00 00 00 00 00 00 00 00 00 00 00 00 "*F.............."
0050: d9 e5 8b 2f 00 00 00 00 00 00 00 00 00 00 00 00 ".../............"
0060: d7 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0070: d8 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
0080: 93 44 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ".D.............."
0090: 5c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "\..............."
00a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
00b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
00c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
00d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
00e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
00f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
```

igaw, 2025-04-01:
Try  ```nvme smart-log /dev/nvme0``` 

The documentation claims it supposed to work for both char device and block devices. Let's see if the doc is right.

kleshas, 2025-04-01:
`sudo nvme smart-log /dev/nvme0
Smart Log for NVME device:nvme0 namespace-id:ffffffff
critical_warning			: 0
temperature				: 40 °C (313 K)
available_spare				: 100%
available_spare_threshold		: 10%
percentage_used				: 1%
endurance group critical warning summary: 0
Data Units Read				: 136888546 (70.09 TB)
Data Units Written			: 55378473 (28.35 TB)
host_read_commands			: 481073498
host_write_commands			: 797775711
controller_busy_time			: 1239
power_cycles				: 1240
power_on_hours				: 17558
unsafe_shutdowns			: 92
media_errors				: 0
num_err_log_entries			: 0
Warning Temperature Time		: 0
Critical Composite Temperature Time	: 0
Thermal Management T1 Trans Count	: 0
Thermal Management T2 Trans Count	: 0
Thermal Management T1 Total Time	: 0
Thermal Management T2 Total Time	: 0`

kleshas, 2025-04-01:
OK, figured out the issue.  Seems like partition number isn't required/wanted/needed.

Something changed I guess as this worked before, but I should've caught this.  Sorry for wasted time.

closed 2025-04-01 by kleshas

igaw, 2025-04-02:
I've checked the documentation: 

[image]

So it's supposed to work when your device has the `SMART/Health Log Page per NS Supported bit set. In my case it isn't:

```shell
# nvme id-ctrl /dev/nvme0 -H
[...]
lpa       : 0xf
  [6:6] : 0     Telemetry Log Data Area 4 Not Supported
  [5:5] : 0     LID 0x0, Scope of each command in LID 0x5, 0x12, 0x13 Not Supported
  [4:4] : 0     Persistent Event log Not Supported
  [3:3] : 0x1   Telemetry host/controller initiated log page Supported
  [2:2] : 0x1   Extended data for Get Log Page Supported
  [1:1] : 0x1   Command Effects Log Page Supported
  [0:0] : 0x1   SMART/Health Log Page per NS Supported
[...]
```