linux-nvme/nvme-cli #2761: The reference tag is parsed incorrectly, causing the PI test case to fail when the format is 512+8, with metadata transferred as a separate buffer

yalanliufio, 2025-03-31:
We recently upgraded our nvme cli version from 2.2.1 to 2.9.1, then encountered this problem.

Release 2.2.1 works correctly, but release 2.9.1 and the latest release 2.12, both have this issue.

When SSD drive is format as 512+8, and metadata is transferred as separate buffer, this is the id-ns output:

[root@localhost test-scripts]# nvme id-ns /dev/nvme1n1 
NVME Identify Namespace 1:
nsze    : 0x1bf1f72b0
ncap    : 0x1bf1f72b0
nuse    : 0x10
nsfeat  : 0x10
nlbaf   : 4
flbas   : 0x1
mc      : 0x3
dpc     : 0x13
dps     : 0x1
nmic    : 0
lbaf  0 : ms:0   lbads:9  rp:0 
lbaf  1 : ms:8   lbads:9  rp:0 (in use)
lbaf  2 : ms:0   lbads:12 rp:0 
lbaf  3 : ms:8   lbads:12 rp:0 
lbaf  4 : ms:64  lbads:12 rp:0 
[root@localhost test-scripts]#

When using nvme version 2.2.1, 
[root@localhost test-scripts]# nvme --version
nvme version 2.2.1 (git 2.2.1)
libnvme version 1.2 (git 1.2)

issue a write command and also enable the kernel trace:, this is the output
nvme write /dev/nvme1n1 -s 0x02000000 -c 2 -z 1536 -y 24 -r 0x02000000 -d write_D1D0027T.data -M write_meta_D1D0027T_slba02000000_nlb2.meta -p 0 -a 0
            nvme-29400   [004] ....... 235741.995845: nvme_setup_cmd: nvme1: disk=nvme1n1, qid=5, cmdid=768, nsid=1, flags=0x0, meta=0x1, cmd=(nvme_cmd_write slba=33554432, len=2, ctrl=0x0, dsmgmt=0, reftag=33554432)
          <idle>-0       [004] d..h1.. 235741.995853: nvme_sq: nvme1: disk=nvme1n1, qid=5, head=1, tail=1
          <idle>-0       [004] d..h1.. 235741.995854: nvme_complete_rq: nvme1: disk=nvme1n1, qid=5, cmdid=768, res=0x0, retries=0, flags=0x2, status=0x0

In this case, reference tag is parsed correctly, command finished successfully

However, in both nvme release 2.9.1 and 2.12, repeat about command, get this output:
[root@localhost test-scripts]# nvme --version
nvme version 2.9.1 (git 2.9.1)
libnvme version 1.9 (git 1.9)

nvme write /dev/nvme1n1 -s 0x02000000 -c 2 -z 1536 -y 24 -r 0x02000000 -d write_D1D0027T.data -M write_meta_D1D0027T_slba02000000_nlb2.meta -p 0 -a 0
            nvme-21954   [016] ....... 234476.695797: nvme_setup_cmd: nvme1: disk=nvme1n1, qid=17, cmdid=64, nsid=1, flags=0x0, meta=0x1, cmd=(nvme_cmd_write slba=33554432, len=2, ctrl=0x0, dsmgmt=0, reftag=0)
          <idle>-0       [016] d..h1.. 234476.695844: nvme_sq: nvme1: disk=nvme1n1, qid=17, head=1, tail=1
          <idle>-0       [016] d..h1.. 234476.695844: nvme_complete_rq: nvme1: disk=nvme1n1, qid=17, cmdid=64, res=0x0, retries=0, flags=0x2, status=0x181

The command failed, this is due to the reference tag is parsed as reftag=0 but the value given by command is 0x2000000, decimal=33554432

This is the command dry-run output, but not very helpful as it is not 16 command dwords, reference tag is the command dword14, is there a way to show the 16 dwords of write command?

[root@localhost test-scripts]# nvme write /dev/nvme1n1 -s 0x02000000 -c 2 -z 1536 -y 24 -r 0x02000000 -d write_D1D0027T.data -M write_meta_D1D0027T_slba02000000_nlb2.meta -p 0 -a 0 -w
opcode       : 01
nsid         : 01
flags        : 00
control      : 0000
nblocks      : 0002
metadata     : 1177800
addr         : 1178000
slba         : 2000000
dsmgmt       : 00000000
reftag       : 2000000
apptag       : 0000
appmask      : 0000
storagetagcheck : 0000
storagetag      : 0
pif             : 00
sts             : 00
[root@localhost test-scripts]#

write_D1D0027T.data and write_meta_D1D0027T_slba02000000_nlb2.meta is attached, I changed it to txt file as data file can't be attached.

Other format with meta data probably has same problem, but I only tested this 512+8 format so far

[write_D1D0027T.data.txt](https://github.com/user-attachments/files/19536699/write_D1D0027T.data.txt)

[write_meta_D1D0027T_slba02000000_nlb2.meta.txt](https://github.com/user-attachments/files/19536700/write_meta_D1D0027T_slba02000000_nlb2.meta.txt)

yalanliufio, 2025-03-31:
io-passthru command passed with the same parameters

[root@localhost test-scripts]# nvme io-passthru /dev/nvme1 -n 1 -O 1 -l 1536 -m 24 -4 0x02000000  -6 1 -8 0x02000000 -9 0 -i write_D1D0027T.data -M write_meta_D1D0027T_slba02000000_nlb2.meta 
IO Command Write is Success and result: 0x00000000

[root@localhost test-scripts]# nvme write /dev/nvme1n1 -s 0x02000000 -c 2 -z 1536 -y 24 -r 0x02000000 -d write_D1D0027T.data -M write_meta_D1D0027T_slba02000000_nlb2.meta -p 0 -a 0
NVMe status: Invalid Protection Information: The command's Protection Information Field settings are invalid for the namespace's Protection Information format(0x181)
[root@localhost test-scripts]#

igaw, 2025-04-01:
Are you sure about the `-p 0` argument? This is for a 16 bit tag as far I understand, but the tag provided is clearly more than 16bits. 

Yes, extending `--dry-run` output would be useful. Someone has to do it, that's the problem.

ikegami-t, 2025-04-02:
You can use '-vv' option to dump dword values.

igaw, 2025-04-02:
For some reason I though the `pif` value is mapped to `-p` command line argument. 

There were many changes in the handling of the PI code between 2.2.1 to 2.9.1. @francispravin5 do you have any ideas?

ikegami-t, 2025-04-02:
Changed by the commit 113c80f6 and https://github.com/linux-nvme/libnvme/commit/9350e73e.
@brandon-paupore-sndk Currenty the nvme write comand uses only nvme_io_args `reftag_u64` but the libnvme function uses both `reftag` and `reftag_u64`. Do you know which is correct handling? (For me it seems better to delete the duplicated `reftag` value from the structure.)

ikegami-t, 2025-04-02:
@yalanliufio Could you please try to test the PR #2765 changes?

brandon-paupore-sndk, 2025-04-02:
> Changed by the commit [113c80f](https://github.com/linux-nvme/nvme-cli/commit/113c80f6e1508fc7fd2526314dcc860d9f5c6513) and [linux-nvme/libnvme@9350e73e](https://github.com/linux-nvme/libnvme/commit/9350e73e). [@brandon-paupore-sndk](https://github.com/brandon-paupore-sndk) Currenty the nvme write comand uses only nvme_io_args `reftag_u64` but the libnvme function uses both `reftag` and `reftag_u64`. Do you know which is correct handling? (For me it seems better to delete the duplicated `reftag` value from the structure.)

This _should_ be correct as only intended to be run with the updated v2 version of `nvme_io_args` for nvme-cli, but if the args->args_size checking isn't working then that'd explain the issue here. The idea was to avoid requiring updates if some function was using the older version of the structure but calling an updated LibNVMe.

In any case, those changes you put up seems like a good test to verify.

ikegami-t, 2025-04-02:
Thanks for your confirmation and comments.

yalanliufio, 2025-04-02:
@ikegami-t the '--vv' option doesn't work, it says: unrecognized option 'vv' like this:
nvme write /dev/nvme7n1 -s 0 -c 2 -z 1536 -y 24 -r 2 -d write_D1D0027T.data -M write_meta_D1D0027T_slba02000000_nlb2.meta -p 0 -a 0 --vv
write: unrecognized option '--vv'

yalanliufio, 2025-04-02:
@ikegami-t 
I would like to try this PR 2765 change, could you please tell me how to compile and build this code?

ikegami-t, 2025-04-02:
No not '--vv' but '-v'.

yalanliufio, 2025-04-02:
> You can use '-vv' option to dump dword values.

ikegami-t, 2025-04-02:
You can see readme documentation or wiki page below to build.
1. readme documentation 
https://github.com/linux-nvme/nvme-cli/blob/master/README.md
2. wiki page
https://github.com/linux-nvme/nvme-cli/wiki/Building-nvme%E2%80%90cli

yalanliufio, 2025-04-02:
> No not '--vv' but '-v'.

The -v doesn't show anything
 nvme write /dev/nvme7n1 -s 0 -c 2 -z 1536 -y 24 -r 2 -d write_D1D0027T.data -M write_meta_D1D0027T_slba02000000_nlb2.meta -p 0 -a 0 -v 
NVMe status: Invalid Protection Information: The command's Protection Information Field settings are invalid for the namespace's Protection Information format(0x181)

yalanliufio, 2025-04-02:
> Very sorry please try '-vv'.

Great, this option works, it shows the dwords, also it did show the reference tag is wrong:
Here is the command I issued:
nvme write /dev/nvme7n1 -s 0 -c 2 -z 1536 -y 24 -r 2 -d write_D1D0027T.data -M write_meta_D1D0027T_slba02000000_nlb2.meta -p 0 -a 0 -vv
here is the write output:
opcode       : 01
flags        : 00
rsvd1        : 0000
nsid         : 00000001
cdw2         : 00000000
cdw3         : 00000000
data_len     : 00000600
metadata_len : 00000018
addr         : c57000
metadata     : c523d0
cdw10        : 00000000
cdw11        : 00000000
cdw12        : 00000002
cdw13        : 00000000
cdw14        : 02000000
cdw15        : 00000000
timeout_ms   : 00000000
result       : 00000000
err          : 385
latency      : 53 us
NVMe status: Invalid Protection Information: The command's Protection Information Field settings are invalid for the namespace's Protection Information format(0x181)

as you can see, in the command -r is 2, but it shows cdw14(reftag)=0x02000000. If I issue this command again with -r 0x02000000, then cdw14=0x00000002, seems like the reftag value used the value from previous write command

ikegami-t, 2025-04-02:
You may use the appimage below instead of the build.
<https://github.com/linux-nvme/nvme-cli/actions/runs/14221991469>

yalanliufio, 2025-04-02:
> [@yalanliufio](https://github.com/yalanliufio) Could you please try to test the PR [#2765](https://github.com/linux-nvme/nvme-cli/pull/2765) changes?

I copied the nvme.c file to my nvme 2.12 directory (total 3 lines changes of .reftag		= cfg.ref_tag,) and rebuild it, it didn't solve the problem, the reftag value is still wrong, same as before.

ikegami-t, 2025-04-02:
Thank you. Noted it so let me investigate again later maybe tomorrow.

brandon-paupore-sndk, 2025-04-02:
> Great, this option works, it shows the dwords, also it did show the reference tag is wrong: Here is the command I issued: nvme write /dev/nvme7n1 -s 0 -c 2 -z 1536 -y 24 -r 2 -d write_D1D0027T.data [...]

The reference tag is successfully being mapped to effectively big-endian ordering. Based on [previous discussions](https://github.com/linux-nvme/libnvme/pull/587) including graphics from the spec from when this change was merged, it does seem to define the tag values in MSB->LSB bit ordering for these cdwords.

Which is to say that everything seems to be working as intended here. If there is some doubt about how the tags should be passed down then that'd be good to confirm on.

keithbusch, 2025-04-02:
The MSB->LSB representation is for the data within the metadata payload as this was essentially copied from SCSI T10. Within the NVMe SQE's dwords, though, these need to be little endian.

keithbusch, 2025-04-02:
Commits fa0f82ce0753553c67d7dc808b75ea9abfd966bd and b121a6fb9f60315029dd54f115f69355a16bb789 appear to be the problem. NVMe command dwords are always in little endian, this change should not have been accepted.

igaw, 2025-04-03:
> Commits fa0f82ce0753553c67d7dc808b75ea9abfd966bd and b121a6fb9f60315029dd54f115f69355a16bb789 appear to be the problem. NVMe command dwords are always in little endian, this change should not have been accepted.

Sorry about that. I keep forget to check for this type of error when doing reviews.

igaw, 2025-04-03:
@brandon-paupore-sndk could you send a fix for this? I can't test this stuff.

brandon-paupore-sndk, 2025-04-03:
@igaw I've put up changes [here ](https://github.com/linux-nvme/libnvme/pull/982)to revert the various commits at fault here, including the commit for copy commands and their source ranges. I haven't taken the time to manually verify that the bits end up in the expected order, but the update is fairly direct.

ikegami-t, 2025-04-03:
The CDW data converted by the kernel driver function to LE from CPU endian as below so seems no needed to convert to LE by the libnvme but just okay to handle the data as CPU endian.
```c
static int nvme_user_cmd(struct nvme_ctrl *ctrl, struct nvme_ns *ns,
		struct nvme_passthru_cmd __user *ucmd, unsigned int flags,
		bool open_for_write)
{
...
	c.common.cdw2[0] = cpu_to_le32(cmd.cdw2);
	c.common.cdw2[1] = cpu_to_le32(cmd.cdw3);
...
	c.common.cdw14 = cpu_to_le32(cmd.cdw14);
...
```
By the way CDW 2 and 3 described as seems CDW 2 used for the reftag but the libnvme function set only CDW3 as below. Is this really correct?
1. CDW 2 and 3 description
[image]
2. libnvme function
```c
static int nvme_set_var_size_tags(__u32 *cmd_dw2, __u32 *cmd_dw3, __u32 *cmd_dw14,
		__u8 pif, __u8 sts, __u64 reftag, __u64 storage_tag)
{
...
	case NVME_NVM_PIF_32B_GUARD:
		cdw14 = reftag & 0xffffffff;
		cdw3 = reftag >> 32;
		cdw14 |= ((storage_tag << (80 - sts)) & 0xffff0000);
		if (sts >= 48)
			cdw3 |= ((storage_tag >> (sts - 48)) & 0xffffffff);
		else
			cdw3 |= ((storage_tag << (48 - sts)) & 0xffffffff);
		cdw2 = (storage_tag >> (sts - 16)) & 0xffff;
		break;
	case NVME_NVM_PIF_64B_GUARD:
		cdw14 = reftag & 0xffffffff;
		cdw3 = (reftag >> 32) & 0xffff;
		cdw14 |= ((storage_tag << (48 - sts)) & 0xffffffff);
		if (sts >= 16)
			cdw3 |= ((storage_tag >> (sts - 16)) & 0xffff);
		else
			cdw3 |= ((storage_tag << (16 - sts)) & 0xffff);
		break;
```

brandon-paupore-sndk, 2025-04-03:
@ikegami-t  It's an 80-bit region overall, so even with the maximum size at 64 bits for a reference tag with the PIF=32b and STS=16 there should never be any of it in CDW2.

[image]

ikegami-t, 2025-04-03:
Thank you so much for your explanation. Understood very well.

igaw, 2025-04-03:
I've merged https://github.com/linux-nvme/libnvme/commit/89ac31f536baf15c1834f877b4c1fb9aca1cf94f  which should address this issue. Any chance to test this? There is an AppImage in case you don't want to build your own version: https://monom.org/linux-nvme/upload/AppImage/nvme-cli-latest-x86_64.AppImage

yalanliufio, 2025-04-03:
@igaw yes, I will test it, thank you so much!

igaw, 2025-04-04:
Oopsi, I need to update the libnvme dependency in order to get the fix added to the nvme-cli build. Let me update it and rebuild nvme-cli for the AppImage.

yalanliufio, 2025-04-04:
This version is great, and the write command passed

nvme write /dev/nvme6n1 -s 16909060 -c 2 -z 1536 -y 24 -r 16909060 -d write.data -M write.meta -p 0 -a 0 -vv
opcode       : 01
flags        : 00
rsvd1        : 0000
nsid         : 00000001
cdw2         : 00000000
cdw3         : 00000000
data_len     : 00000600
metadata_len : 00000018
addr         : 56095728d000
metadata     : 560957288490
cdw10        : 01020304
cdw11        : 00000000
cdw12        : 00000002
cdw13        : 00000000
cdw14        : 01020304
cdw15        : 00000000
timeout_ms   : 00000000
result       : 00000000
err          : 0
latency      : 16 us
write: Success

Thanks a lot for the great help! Looking forward to the new release 2.13 to integrate into our test automation.

closed 2025-04-05 by igaw