linux-nvme/nvme-cli #2778: R/W data transfer limited by block device driver (submit-io: Invalid argument)

FrastSerg, 2025-04-19:
The size of read/write operations is limited by the block device driver. The maximum size can be obtained with a command like:
`# blockdev --getioopt /dev/nvme0n3 # in bytes
`
Or by:
`# cat /sys/block/nvme0n3/queue/max_sectors_kb
`
Error reproduction sequence:
1) Create a file that is larger than the value displayed by the command:
```
# cat /sys/block/nvme0n3/queue/max_sectors_kb
128

# dd bs=1K count=129 if=/dev/random of=/tmp/test129K.data
129+0 records in
129+0 records out
 132096 bytes
```

2) Let's write data to an empty NVMe namespace device by nvme-cli utility:
```
#  nvme write -s50 -z132096 -d /tmp/test129K.data -v /dev/nvme0n3
opcode       : 01
nsid         : 03
flags        : 00
control      : 0000
nblocks      : 0000
metadata     : 0
addr         : 701b48574000
slba         : 32
dsmgmt       : 00000000
reftag       : 0
apptag       : 0000
appmask      : 0000
storagetagcheck : 0000
storagetag      : 0
pif             : 00
sts             : 00
submit-io: Invalid argument
```

3) Reduce the size of the data being written to the maximum value:
```
# echo $(( 128 * 1024))
131072

#  nvme write -s50 -z131072 -d /tmp/test129K.data -v /dev/nvme0n3
opcode       : 01
nsid         : 03
flags        : 00
control      : 0000
nblocks      : 0000
metadata     : 0
addr         : 7b6873c68000
slba         : 32
dsmgmt       : 00000000
reftag       : 0
apptag       : 0000
appmask      : 0000
storagetagcheck : 0000
storagetag      : 0
pif             : 00
sts             : 00
write: Success
```

The same problem exists for large read operations.

keithbusch, 2025-04-19:
Everything appears to be working as designed. What are expecting to be different?

FrastSerg, 2025-04-20:
In all likelihood, you are right. This is not a problem with nvme-cli, but with libnvme, which you are using. Apparently, these are such different projects (even though they are in the same "linux-nvme" account) that you cannot be held responsible for any shortcomings in the operation of any components included in your project.

I wanted to warn our colleagues about the existing problem, since they, first of all, encounter this utility, and not the library...

In any case, thanks for the utility! An excellent entry point into the technologies of direct work with NVMEe equipment and studying specifications!

igaw, 2025-04-22:
FWIW, `libnvme` and `nvme-cli` are maintained and developed by the same group of people. `nvme-cli` is the frontend, the library provides abstraction etc to the kernel interface. `nvme-cli` is the main consumer of `libnvme`. 

A good idea is to update the library on regular basis if you just using the library. Bug fixes go in all the time.

closed 2025-04-22 by igaw