linux-nvme/nvme-cli #2781: blktests nvme/030 fails when `70-nvmf-autoconnect.rules` is active

igaw, 2025-04-23:
blktests nvme/030 test is likely to fail if the 70-nvmf-autoconnect.rules is active. In this case two discovery are running in parallel and nvme-cli/libnvme gets out of sync and it looks as nvme-cli operates on the wrong device:

```
nvme/030 (tr=fc) (ensure the discovery generation counter is updated appropriately) [failed]
    runtime    ...  1.254s
    --- tests/nvme/030.out      2023-08-30 08:39:08.428409596 +0000
    +++ /tmp/blktests/nodev_tr_fc/nvme/030.out.bad      2025-04-23 13:15:40.722450950 +0000
    @@ -1,2 +1,6 @@
     Running nvme/030
    +Failed to open ctrl nvme1, errno 11
    +Failed to open ctrl nvme1, errno 11
    +failed to get discovery log: Bad file descriptor
    +tests/nvme/rc: line 489: ((: 3 <=  : syntax error: operand expected (error token is "<=  ")
     Test complete
```

```
    [ 1434.556136] run blktests nvme/030 at 2025-04-10 10:56:03
    [ 1434.779180] nvmet: adding nsid 1 to subsystem blktests-subsystem-1
    [ 1434.858814] nvme nvme1: NVME-FC{0}: create association : host wwpn 0x20001100aa000001  rport wwpn 0x20001100ab000001: NQN "nqn.2014-08.org.nvmexpress.discovery"
    [ 1434.860275] (NULL device *): {0:0} Association created
    [ 1434.861773] nvmet: Created discovery controller 1 for subsystem nqn.2014-08.org.nvmexpress.discovery for NQN nqn.2014-08.org.nvmexpress:uuid:a94a718e-09b2-4b5b-898a-116845846761.
    [ 1434.864704] nvme nvme1: NVME-FC{0}: controller connect complete
    [ 1434.865082] nvme nvme1: NVME-FC{0}: new ctrl: NQN "nqn.2014-08.org.nvmexpress.discovery", hostnqn: nqn.2014-08.org.nvmexpress:uuid:a94a718e-09b2-4b5b-898a-116845846761
    [ 1434.871357] nvme nvme1: Removing ctrl: NQN "nqn.2014-08.org.nvmexpress.discovery"
    [ 1434.953940] nvmet: adding nsid 1 to subsystem blktests-subsystem-2
    [ 1434.954581] (NULL device *): {0:0} Association deleted
    [ 1435.007936] (NULL device *): {0:0} Association freed
    [ 1435.009560] (NULL device *): Disconnect LS failed: No Association
    [ 1435.060507] nvme nvme2: NVME-FC{1}: create association : host wwpn 0x20001100aa000001  rport wwpn 0x20001100ab000001: NQN "nqn.2014-08.org.nvmexpress.discovery"
    [ 1435.061682] (NULL device *): {0:0} Association created
    [ 1435.062499] nvmet: Created discovery controller 1 for subsystem nqn.2014-08.org.nvmexpress.discovery for NQN nqn.2014-08.org.nvmexpress:uuid:0f01fb42-9f7f-4856-b0b3-51e60b8de349.
    [ 1435.064831] nvme nvme2: NVME-FC{1}: controller connect complete
    [ 1435.065244] nvme nvme2: NVME-FC{1}: new ctrl: NQN "nqn.2014-08.org.nvmexpress.discovery", hostnqn: nqn.2014-08.org.nvmexpress:uuid:0f01fb42-9f7f-4856-b0b3-51e60b8de349
    [ 1435.070407] nvme nvme2: Removing ctrl: NQN "nqn.2014-08.org.nvmexpress.discovery"
    [ 1435.143134] (NULL device *): {0:0} Association deleted
    [ 1435.181772] (NULL device *): {0:0} Association freed
    [ 1435.182360] (NULL device *): Disconnect LS failed: No Association
```

ikegami-t, 2025-04-27:
Seems related to the issue: #2760 (, the PRs: #2775 and https://github.com/linux-nvme/libnvme/pull/993) since the errno: 11 is EAGAIN and the issue: #2760 caused by the error: -4 (EINTR) as the both issues caused by the kernel error behaviors as regression.

igaw, 2025-04-28:
I haven't really dig into what is causing the blktests to break. Yes, could be related indeed.

igaw, 2025-05-05:
Indeed, this tests fails because the ioctl syscall returns EAGAIN.

igaw, 2025-07-25:
The fix is merged. Close this one.

closed 2025-07-25 by igaw